mne-tools/mne-python #2624: Error running make_forward_solution: Head<-->sensor transform not found

emilyps14, 2015-11-18:
From a thread on the [Mne_analysis] list: https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2015-November/002943.html

I'm getting an error when I run mne.make_forward_solution, but only if I
try to save the result.

That is, if I set `fname=None` then the whole thing runs fine, but if I set
`fname=<filename>`, it fails in the call to write_forward_solution with the
error:
    Head<-->sensor transform not found

Seems like write_forward_solution is expecting values in info under
`dev_head_t` or `ctf_head_t`, which never got set.

I think it may be related to the fact that I'm loading (just EEG) data from
a brainvision file using mne.io.read_raw_brainvision.

I've been struggling to figure out how to include the digitizer information
in the preprocessed data. My digitizer info is in a .fif file that
otherwise doesn't contain any measurement data (I can't even figure out how
to load it using the python MNE). So far the only thing I've done with that
file is to use it to update the channel locations with a call to the C
implementation:
```
    call(['mne_check_eeg_locations',
          '--file', fname,
          '--dig', op.join(subjects_dir,subject,subject + '.fif'),
          '--fix'])
```

agramfort, 2015-11-19:
thanks for the bug report.

I think you'll need to share the files if you want help

emilyps14, 2015-11-19:
Yep. I'm working on creating a simplified version of the code that still has the error, and I've managed to get an error somewhere else.

emilyps14, 2015-11-19:
OK.

I noticed that the default python distribution on the Martinos system was using MNE 0.8. Switched to the anaconda distribution which was using MNE 0.9.

After working through some bugs related to the version change, I no longer get the error I reported before. 

No more issue!

closed 2015-11-19 by emilyps14