mne-tools/mne-python #2639: v0.11

Eric89GXL, 2015-11-24:
Do we want to make a `maxwell_filter`-themed holiday release in a month? If so, I'll start pushing existing issues marked v0.11 to v0.12.

agramfort, 2015-11-25:
would be good.

what are the list of things we want to complete before?

dengemann, 2015-11-25:
Great idea.
On my side some more ICA edits, e.g. components  by rank and probably #2586.

Eric89GXL, 2015-11-25:
CTF reading and maybe raw test refactoring would be good. I'll get marking.

agramfort, 2015-11-25:
once we have CTF support we should update the SPM examples are work with
the native format. Not the converted fif files.

Eric89GXL, 2015-11-25:
Sure. You want me to do that as part of the PR? I suspect you're already avoiding reviewing it because of the number of lines changed, so maybe a subsequent PR is better :)

Also the FIFF constants PR should go in release, I've marked it.

closed 2015-11-25 by Eric89GXL