mne-tools/mne-python #2645: BUG?: resample + concatenation breaks

kingjr, 2015-11-26:
I'm regularly confronted to an error when I resample a set of epochs, and try to concatenate them.

```
import numpy as np
from mne.epochs import concatenate_epochs  # BTW, why is concatenate_epochs not accessible from mne directly
nepoch, nchan, ntime = 100, 30, 20
# make dummy epochs1 and epoch2, sfreq = 250 Hz
epochs = concatenate_epochs([epochs1, epochs2])
# works

epochs1.resample(epochs1.info['sfreq'] // 2)
epochs2.resample(epochs2.info['sfreq'] // 2)
epochs = concatenate_epochs([epochs1, epochs2])
# error
```

```
RuntimeError                              Traceback (most recent call last)
<ipython-input-6-28041ffc5880> in <module>()
      4 epochs1.resample(125)
      5 epochs2.resample(125)
----> 6 epochs = concatenate_epochs([epochs1, epochs2])

/home/jrking/mne-python/mne/epochs.pyc in concatenate_epochs(epochs_list)
   2602     .. versionadded:: 0.9.0
   2603     """
-> 2604     return _finish_concat(*_concatenate_epochs(epochs_list))

/home/jrking/mne-python/mne/epochs.pyc in _finish_concat(info, data, events, event_id, tmin, tmax, baseline, selection, drop_log, verbose)
   2580                       baseline=baseline, add_eeg_ref=False,
   2581                       selection=selection, drop_log=drop_log,
-> 2582                       proj=False, on_missing='ignore', verbose=verbose)
   2583     out.drop_bad_epochs()
   2584     return out

/home/jrking/mne-python/mne/epochs.pyc in __init__(***failed resolving arguments***)
    280             if data.ndim != 3 or data.shape[2] != \
    281                     round((tmax - tmin) * self.info['sfreq']) + 1:
--> 282                 raise RuntimeError('bad data shape')
    283             self.preload = True
    284             self._data = data

RuntimeError: bad data shape
```

I think we already mentioned this error with @Eric89GXL , but I haven't found the issue.

Eric89GXL, 2015-11-26:
Can you fill out your minimal example with e.g. EpochsArray? That way others can run a bit of code to confirm/test more easily.

kingjr, 2015-11-26:
Sure

```
import copy
import numpy as np
from mne.epochs import EpochsArray
from mne.io.meas_info import create_info
from mne.epochs import concatenate_epochs  # BTW, why is concatenate_epochs not accessible from mne directly

n_trial, n_chan, n_time, sfreq  = 100, 30, 20, 250
data = np.zeros((n_trial, n_chan, n_time ))
chan_names = ['meg_%02i' % chan for chan in range(n_chan)]
chan_types = ['mag'] * n_chan
events = np.zeros((n_trial, 3), int)
info = create_info(chan_names, sfreq, chan_types)
epochs1 = EpochsArray(copy.deepcopy(data), copy.deepcopy(info),
                      events=np.array(events, int), verbose=False)
epochs2 = EpochsArray(copy.deepcopy(data), copy.deepcopy(info),
                      events=np.array(events, int), verbose=False)

# make dummy epochs1 and epoch2, sfreq = 250 Hz
epochs = concatenate_epochs([epochs1, epochs2])
# works

epochs1.resample(epochs1.info['sfreq'] // 2)
epochs2.resample(epochs2.info['sfreq'] // 2)
epochs = concatenate_epochs([epochs1, epochs2])
# error

```

closed 2015-11-30 by Eric89GXL

kingjr, 2015-11-30:
> Closed #2645 <https://github.com/mne-tools/mne-python/issues/2645> via

Thanks!

Eric89GXL, 2015-11-30:
Thanks for the easy minimal example, you can see a very similar version ended up in our test suite :)

PR #2657 (2015-11-30): MRG: Cleaner use of tmin/tmax
Closes #2645.

The bug was due to improperly setting `tmax`. I've changed `epochs` to derive it from `.times` instead, which should be a cleaner approach.