mne-tools/mne-python #2648: TraitError: The 'input' trait of a QuadricDecimation instance is 'read only'. + 1 more failing test

yarikoptic, 2015-11-26:
```
======================================================================
ERROR: Test triangular surface decimation
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/utils.py", line 635, in dec
    return function(*args, **kwargs)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/tests/test_surface.py", line 158, in test_decimate_surface
    _, this_tris = decimate_surface(points, tris, n_tri)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/surface.py", line 761, in decimate_surface
    return _decimate_surface(points, triangles, reduction)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/surface.py", line 725, in _decimate_surface
    decimate = tvtk.QuadricDecimation(input=src, target_reduction=reduction)
  File "/usr/lib/python2.7/dist-packages/tvtk/tvtk_classes.zip/tvtk_classes/quadric_decimation.py", line 63, in __init__
    tvtk_base.TVTKBase.__init__(self, vtk.vtkQuadricDecimation, obj, update, **traits)
  File "/usr/lib/python2.7/dist-packages/tvtk/tvtk_base.py", line 307, in __init__
    super(TVTKBase, self).__init__(**traits)
  File "/usr/lib/python2.7/dist-packages/traits/trait_handlers.py", line 104, in _read_only
    name, class_of( object ) )
TraitError: The 'input' trait of a QuadricDecimation instance is 'read only'.
```
and another one which just reemerged
```
======================================================================
FAIL: Test CombineMarkersModel Traits Model
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/utils.py", line 635, in dec
    return function(*args, **kwargs)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/gui/tests/test_kit2fiff_gui.py", line 35, in test_kit2fiff_model
    assert_false(model.can_save)
AssertionError: <bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f6568c0fad0>> is not false
    '<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f6568c0fad0>> is not false' = self._formatMessage('<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f6568c0fad0>> is not false', "%s is not false" % safe_repr(expr))
>>  raise self.failureException('<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f6568c0fad0>> is not false')
    
```

I have 

```
$> acpolicy mayavi2   
mayavi2:
  Installed: 4.4.3-2
  Candidate: 4.4.3-2
  Version table:
 *** 4.4.3-2 0
        900 http://http.debian.net/debian/ stretch/main amd64 Packages
        600 http://http.debian.net/debian/ sid/main amd64 Packages
        100 /var/lib/dpkg/status
```

agramfort, 2015-11-26:
@jaeilepp can you replicate?

yarikoptic, 2015-11-26:
also I have missed two more failures

```
======================================================================
ERROR: Test CombineMarkersModel Traits Model
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/utils.py", line 635, in dec
    return function(*args, **kwargs)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/gui/tests/test_marker_gui.py", line 33, in test_combine_markers_model
    assert_false(model.mrk3.can_save)
AttributeError: 'function' object has no attribute 'can_save'

======================================================================
ERROR: Test that there are no tabs in our source files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/tests/test_docstring_parameters.py", line 153, in test_tabs
    __import__(modname)  # because we don't import e.g. mne.tests w/mne
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/gui/_coreg_gui.py", line 47, in <module>
    from ._fiducials_gui import MRIHeadWithFiducialsModel, FiducialsPanel
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/gui/_fiducials_gui.py", line 347, in <module>
    class FiducialsFrame(HasTraits):
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.10.1+dfsg/mne/gui/_fiducials_gui.py", line 392, in FiducialsFrame
    show_labels=False),
  File "/usr/lib/python2.7/dist-packages/traitsui/group.py", line 238, in __init__
    raise TypeError, "Unrecognized argument type: %s" % value
TypeError: Unrecognized argument type: <function <lambda> at 0x7f6568bc1e60>
-------------------- >> begin captured logging << --------------------
traits.has_traits: WARNING: DEPRECATED: traits.has_traits.wrapped_class, 'the 'implements' class advisor has been deprecated. Use the 'provides' class decorator.
apptools.preferences.preferences: DEBUG: loading preferences from </home/yoh/.enthought/mayavi_e3/preferences.ini>
apptools.preferences.preferences: DEBUG: loading preferences from <<open file '/usr/lib/python2.7/dist-packages/mayavi/preferences/preferences.ini', mode 'rb' at 0x7f6504eafd20>>
apptools.preferences.preferences: DEBUG: loading preferences from <<open file '/usr/lib/python2.7/dist-packages/tvtk/plugins/scene/preferences.ini', mode 'rb' at 0x7f6504983420>>
--------------------- >> end captured logging << ---------------------
```

jaeilepp, 2015-11-27:
> @jaeilepp can you replicate?

No success so far...

agramfort, 2015-11-27:
see also #2582

what version of traits is causing the problem?

jaeilepp, 2015-11-27:
I'm running traits 4.5.0 and cannot replicate the error.

jaeilepp, 2015-11-27:
I couldn't reproduce any of the traits errors with or without mayavi. What kind of environments are you running the tests on @yarikoptic ?

yarikoptic, 2015-11-27:
hm -- that is happening on my laptop in a "dirty" environment with mayavi2... when building in a clean debian unstable/sid with python-vtk instead -- doesn't happen.  Let me just finalize package as it is now and try building across debians/ubuntus

yarikoptic, 2015-11-27:
"Read only" issue does happen on debian testing (only) -- didn't check in detail... may be it is a bug in one of the dependencies which was fixed later since doesn't happen in unstable (sid).

yarikoptic, 2016-05-14:
FWIW -- tried to build 0.12 and it happened now in clean debian sid (unstable)environment:
```
.S..........................SSSSSSSSSSSS...S./build/python-mne-0.12+dfsg/mne/tests/test_proj.py:204: DeprecationWarning: The copy parameter is deprecated and will be removed in 0.14. In 0.13 the behavior will be to operate in-place (like copy=False). In 0.12 the default is copy=True. Use inst.copy() if necessary.
  raw = Raw(raw_fname).crop(0, raw_time)
....SSS..SS.SSS...SSS..S.SS.SS.SSSS.SSSSSSSSSSSS.S..SSSESS.....S.......................SS
======================================================================
ERROR: Test triangular surface decimation
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/build/python-mne-0.12+dfsg/mne/utils.py", line 730, in dec
    return function(*args, **kwargs)
  File "/build/python-mne-0.12+dfsg/mne/tests/test_surface.py", line 163, in test_decimate_surface
    _, this_tris = decimate_surface(points, tris, n_tri)
  File "/build/python-mne-0.12+dfsg/mne/surface.py", line 763, in decimate_surface
    return _decimate_surface(points, triangles, reduction)
  File "/build/python-mne-0.12+dfsg/mne/surface.py", line 727, in _decimate_surface
    decimate = tvtk.QuadricDecimation(input=src, target_reduction=reduction)
  File "/usr/lib/python2.7/dist-packages/tvtk/tvtk_classes.zip/tvtk_classes/quadric_decimation.py", line 63, in __init__
    tvtk_base.TVTKBase.__init__(self, vtk.vtkQuadricDecimation, obj, update, **traits)
  File "/usr/lib/python2.7/dist-packages/tvtk/tvtk_base.py", line 307, in __init__
    super(TVTKBase, self).__init__(**traits)
  File "/usr/lib/python2.7/dist-packages/traits/trait_handlers.py", line 104, in _read_only
    name, class_of( object ) )
TraitError: The 'input' trait of a QuadricDecimation instance is 'read only'.

----------------------------------------------------------------------
Ran 593 tests in 286.839s
```
I will try to look into it later

agramfort, 2016-05-15:
It's weird as we don't do anything fancy here:

https://github.com/mne-tools/mne-python/blob/master/mne/surface.py#L714

PR #3233 (2016-05-16): BF: use tvtk.configure_input instead of direct input=src
Trying to resolve error during testing on Debian sid, see
http://stackoverflow.com/a/35131392 for origin for the "fix".
Closes #2648 

I have little to no clue in tvtk etc, just nearly blindly following the recipe from stackoverflow  which seems to avoid error during testing at least on debian sid.  Will try now across releases
commit 154d7e5d (2016-05-16): BF: use tvtk.configure_input instead of direct input=src

Trying to resolve error during testing on Debian sid, see
http://stackoverflow.com/a/35131392 for origin for the "fix".
Closes #2648
closed 2016-05-16 by Eric89GXL