mne-tools/mne-python #2652: Saving raw object converts highpass and lowpass from None to nan and cannot handle it afterwards

Teekuningas, 2015-11-28:
Hi,

It seems you can do this:
```python
>> raw = mne.io.read_raw_egi('file-with-no-highpass-and-lowpass-information')
>> raw.info['highpass'] is None
True
>> raw.save('temp.fif')
>> raw = mne.io.Raw('temp.fif')
>> raw.info['highpass']
nan
>> raw.filter(l_freq=1, h_freq=40)
>> raw.info['highpass']
nan
```

So saving raw object load from egi with highpass and lowpass set to zero has highpass and lowpass converted to nan, and after reread for example filtering assumes that "no information" is coded as None, and is not willing to update highpass and lowpass information, if otherwise. As a workaround I am at the moment converting None's to 0 and sfreq/2 before saving, but it might not should be necessary.

agramfort, 2015-11-29:
see https://github.com/mne-tools/mne-python/pull/2655

Teekuningas, 2015-11-30:
Yes, that does the trick. Thanks.

closed 2015-11-30 by Teekuningas