mne-tools/mne-python #2661: Change 'seeg' to 'ieeg', or add an 'ecog' channel type

choldgraf, 2015-11-30:
I noticed that there is an 'seeg' electrode type allowed for many functions. I wonder if it was intentional to use 'seeg' to specifically mean depth electrodes, or if this is more of an umbrella term to refer to intracranial electrodes (which would include both depth electrodes and surface electrodes).

What are thoughts about either:

A. Changing 'seeg' to 'ieeg', so that it's a catchall term for intracranial recordings.
B. Keeping it that way but adding an 'ecog' type, so that it's explicit which one you're dealing with.

It's just not clear to me why stereo eeg would be added but not surface electrodes.

agramfort, 2015-12-01:
I would keep seeg and ecog separate as they
need different forward models (in the future)
and they are visualized differently.

Eric89GXL, 2015-12-01:
@jaeilepp we'll need to coordinate with the FIFF format mainters to add another channel type. I think they were already going to add SEEG so it might be a good time to bother them about also adding ECoG.

choldgraf, 2015-12-01:
I agree that an extra 'ecog' type would be the best as well, as it uses
different kinds of references and visualizations.

jaeilepp, 2015-12-01:
Okay, I'll send an e-mail tomorrow.

jaeilepp, 2015-12-02:
FYI Elekta was okay adding ECoG to their FIFF definitions.

agramfort, 2015-12-02:
let's go with this using the constant number they suggested.

choldgraf, 2015-12-02:
I am excited not to remember to disable `add_eeg_ref` every time I load in ecog data now :)

Actually, that reminds me, if `ecog` and `seeg` types are in there, a good addition would be to include some referencing algorithms that people use. For ecog, it's often a common-average reference, often in subsets of electrodes depending on their geometry. For seeg, it's often either a bipolar reference or a single reference electrode.

Eric89GXL, 2016-02-03:
We should try to get this for 0.12. @choldgraf do you have time to add it? I think if you grep `seeg` and `SEEG` you might find all the places you need to add something to get it working properly.

choldgraf, 2016-02-03:
When's the target date for this? I can't prioritize it now but if/when I resubmit my paper I can spend time doing development stuff

Eric89GXL, 2016-02-03:
Next release should be in April

choldgraf, 2016-02-04:
cool - I think I'll be able to get to this by then. To be clear, it sounds
like the plan would be to add another type, `ecog`. So there would be
*both* `seeg` and `ecog`.

agramfort, 2016-02-04:
> So there would be *both* `seeg` and `ecog`.
>

+1

Eric89GXL, 2016-03-17:
@jaeilepp in anticipation of this, can you work with the Elekta folks to add the constant, and update our `io.constants`?

jaeilepp, 2016-03-18:
Yeah, 902 is reserved for ecog. I could add it to ``io.constants`` in one of my PRs, but I'll leave updating the picking functions to someone who is more familiar with ecog data.

Eric89GXL, 2016-04-20:
@choldgraf any time to work on this in the next week? If not, I can tackle it.

choldgraf, 2016-04-20:
I might have some time, assuming this doesn't become one of those "seems
easy but breaks everything" kinds of changes :-)

Eric89GXL, 2016-04-20:
Hopefully `git grep seeg` would show you all the places that need to change or be replicated. If it's harder than that, then there must be some deeper problem we need to fix, and I can take over then.

Eric89GXL, 2016-04-20:
FYI the ECoG channel type is already there https://github.com/mne-tools/mne-python/blob/master/mne/io/constants.py#L180

choldgraf, 2016-04-20:
Cool, I'll give it a shot and hopefully it won't be too complicated (famous
last words)

choldgraf, 2016-04-20:
Hmm, so if we already have 'ecog' in there in constants, as well as `seeg`,
then what's the scope of this change? Is it just finding any reference to
`seeg` and then adding in another reference to `ecog` as well, so that we
support both? Any also looking for places where people say `seeg` but
really mean `ecog`? (though I think that last one is not common)

Eric89GXL, 2016-04-20:
Yeah that is what I have in mind. Hopefully it's in `mne.io.pick.py` functions and not many other places, but `git grep` should be informative.

choldgraf, 2016-04-20:
cool, I'll let ya know what's up

Eric89GXL, 2016-04-22:
@choldgraf I think I can do this while taking over #3053 actually

choldgraf, 2016-04-22:
ah, ok...well, let me know if you need help? happy to weigh in on the PR if
you want to ping me etc

PR #3157 (2016-04-28): MRG: ECoG channel type
Closes #2661.

Ready for review/merge from my end.

@choldgraf can you try it out a bit see if it works decently? I'm especially curious if the values chosen for plotting make sense.
closed 2016-04-28 by agramfort