mne-tools/mne-python #2664: Joint ER[P/F] and topo plot

jona-sassenhagen, 2015-12-02:
Once #2662 is merged, only one aspect is missing for my favorite way to plot.

[image]

Combined time series and topoplots, like ERPLAB has.

I'd like to implement this next, but I'm wondering about the API. A function `mne.viz.evoked.joint_plot`? Another arg for `evoked.plot`, or maybe `evoked.plot_topomap`?

agramfort, 2015-12-02:
I would add times param to evoked.plot to add topos from specific time
instants.

jona-sassenhagen, 2015-12-02:
I'd also want to expose most or all other params for plot_topoplot though, and also allow auto times (e.g. the new 'peaks'  functionality), how about a `list | dict` arg? (if list, treat it as times)

agramfort, 2015-12-02:
hum then can you make it work just by passing a fig or axis object that
contains the evoked.plot to a new function. Our plot functions have already
too many parameters.

jona-sassenhagen, 2015-12-02:
For the future: EEGLAB also has joint time-frequency/topo plots.
[image]

So if we also want that (IMO they're very nice), one should think about an API that also covers that case.

agramfort, 2015-12-02:
nilearn.plotting handles this by returning more than a fig and having an
object that allows to interact with plot and complement it.

Something along these lines:

fig = evoked.plot()
fig.add_topo(times=...)

jona-sassenhagen, 2015-12-02:
Hm ... but that's not really MNE style is it? Also I'm not sure I could program that ...
Both the topoplot and the ts plot need access to the original evoked object. And API wise, I think a method on evoked objects is always nicest.

Would you be okay with a new method (`evoked.jointplot`) like this:

```
def jointplot(evoked, title=None, ts_args=None, topo_args=None):
"""Plots joint top and time series.

    evoked : evoked
        instance of evoked
    title : str
        The title of the figure
    ts_args : none | dict
        dict of args that are passed to evoked.plot
    topo_args : none | dict
        dict of args that are passed to evoked.plot_topo

    returns : fig"""
```

So if called without args, it'd just return default ts and topo plots. But you could still do e.g.

```
evoked.jointplot(title="P3 plot", ts_args=dict=(spatial_colors=True), topo_args=dict(times=.3))
```
and e.g. more elaborate args to topoplot.

agramfort, 2015-12-02:
first let's make it work, test it on many datasets, then we'll make it nice
with a good API

jona-sassenhagen, 2015-12-02:
Ok!

jona-sassenhagen, 2016-01-16:
Closed by #2738

closed 2016-01-16 by jona-sassenhagen