mne-tools/mne-python #2665: semantics for events with duration

jasmainak, 2015-12-02:
This relates partly to reproducing the brainstorm results in https://github.com/mne-tools/mne-python/pull/2587. Currently, mne-python does not have the semantics to handle annotations with durations. One usecase is to visualize bad segments in the raw data using `raw.plot()`. I am opening the issue so that the API can be discussed.

cc @jaeilepp @agramfort @dengemann

agramfort, 2015-12-02:
as discussed one option to avoid introducing a new concept in MNE would be
use events.

e.g.

[[100 0 1],
 [200 1 0]]

would code for a event of type 1 from 100 to 200 milliseconds.

consequence is that it's not obvious if it's a bad segment or not, unless
for example we say that if the event id is negative it is bad segment.
it's a bit more cryptic than having some kind of event id and look for the
work
bad in it.

jasmainak, 2015-12-02:
can you give an example how it would look if there were more than one segment + a few experiment-related events? Would you allow for overlapping events?

jasmainak, 2015-12-02:
what happens when you have some events which have a duration and others which don't have a duration?

dengemann, 2015-12-02:
> what happens when you have some events which have a duration and others which don't have a duration?

we could have lists of arrays or 3D arrays

>  Would you allow for overlapping events?

open to the user, you can put in your array whatever you want

> consequence is that it's not obvious if it's a bad segment or not, unless
for example we say that if the event id is negative it is bad segment.

we could have a user defined bads tag or as you say negative

Eric89GXL, 2015-12-02:
I was thinking:
```
100 200 1
```
i.e., onset-offset-kind.

dengemann, 2015-12-02:
> i.e., onset-offset-kind.

but how do you then combine this with the exisiting approach?

jasmainak, 2015-12-02:
@Eric89GXL : the documentation currently says:

> The first column contains the event time in samples and the third column contains the event id. For output = 'onset' or 'step', the second column contains the value of the stim channel immediately before the the event/step.

The second column does not agree with what you propose

Eric89GXL, 2015-12-02:
How we save and load the annotations is to me a mostly separate issue from how we use them. I'd prefer to have an annotation format that is compact and simple to understand, ideally requiring no new IO functions (onset-offset-kind fits this), and then work on how to use that annotation in a nice way from an API standpoint.

But since you asked, I originally implemented a function that worked on `epochs` to eliminate any epochs that were considered "bad". I don't think that's the best approach, though. Instead we could "attach" them to raw objects (in memory only for now), then:

1. They can easily be plotted with `raw.plot`

2. When Epochs get constructed, they can be aware of the annotations attached to the `Raw` instance and drop any events (+/- tmin/tmax) that live in "bad" segments. Maybe the user needs to then pass `bad_id` in addition to `event_id` to mark which annotated segments are not to be used using `key, value` pairs, and we can mark in the drop log the right reason using the `bad_id` `value`.

This to me is pretty simple from a user perspective, and also shouldn't be too difficult to implement from an API perspective.

Eric89GXL, 2015-12-02:
The other nice thing about onset-offset-kind is that one can easily get to this format from other readers/annotation software, since it can be easily written out as plain test and read using our `read_events` function.

Eric89GXL, 2015-12-02:
@jasmainak I know. It wouldn't be an "event" or `-eve` file anymore but an "annotation" or `-ann` file or so. I don't see a problem with redefining what the columns mean here.

dengemann, 2015-12-02:
I like this proposal a lot @Eric89GXL

dengemann, 2015-12-02:
It would be terribly useful to generate new raw files from sleep annotations(!) and so on.

Eric89GXL, 2015-12-02:
Does anyone use annotation software, or work with anyone who does? It might be good to know what their saving formats look like. My guess is that it's onset, offset, and some form of text entry that allows the user to say what the event is. We'd be pretty close to that, only we'd force people to construct the `bad_id` dict, which I don't think is a bad thing.

dengemann, 2015-12-02:
why bad_id? why not annotation_id?

jasmainak, 2015-12-02:
@Eric89GXL , the problem is that you are going to break a lot of scripts if you redefine what it means

dengemann, 2015-12-02:
> @Eric89GXL , the problem is that you are going to break a lot of scripts if you redefine what it means

what would you break? we would introduce a separate thing as I understand it. We have events VS annotations. That would be the cleanest.

Eric89GXL, 2015-12-02:
@jasmainak I don't think so, nobody should use `-ann.txt` files when they want to use events...? It would only be redefined in the context of annotation use, not event use.

@dengemann the idea is that it use the annotation to define bad segments to exclude when making `Epochs`. But maybe we want a more complete representation, where the annotations stick around and could also be plotted in the `Epochs` object, too, instead of just disqualifying trials. In that case `annotation_id` would make sense, but we'd need to think more about the API.

FWIW my proposal for a saving format isn't substantially different from @agramfort's. Mine is slightly more compact, readable, and easier to construct from a user perspective. But it does change the meaning of the columns relative to `events`, which is a disadvantage. I +0.5 prefer my approach for readability. But given this has already caused some confusion for @jasmainak, I am a bit concerned. Maybe sufficient documentation would work?

jasmainak, 2015-12-02:
@Eric89GXL : are you suggesting introducing another attribute to `raw`?

dengemann, 2015-12-02:
> But maybe we want a more complete representation, where the annotations stick around and could also be plotted in the Epochs object, too, instead of just disqualifying trials. In that case annotation_id would make sense, but we'd need to think more about the API.

We should go that way I feel. It would be terrific for neurological data and for spontaneous brain activity. I have a clear vision now. We can with that API use them for rejection but also for other things such as plotting and re/fine-segmentation, etc. What you propose as data layout would be nice with a separate parameter / attribute.

jasmainak, 2015-12-02:
FWIW, I'll add my proposal as well. I feel `events` should be an object. It will be able to deal with complex annotations that way. The `get` property could return the numpy array that you get currently and it could be structured differently for annotations like @Eric89GXL suggests. An attribute to the object would tell you how it's structured. And `mne.viz.plot_events` could be refactored into this.

agramfort, 2015-12-02:
sounds like we need a hangout session to converge.

I need to run.

Just 1 word : I agree with Eric's proposal but the devil is in the API
detail.

jaeilepp, 2015-12-03:
I like Eric's proposal too. I'm just wondering how to handle cropped data. With Eric's system it doesn't matter if the end part of the segment gets cut off, but if the beginning gets cut off, the segmentation data is lost.

One option would be to use negative values for segments in combination with Alex's solution. This way it would be possible to infer the segments, but I'm not sure how it would behave with overlapping events without looking into the code.

Eric89GXL, 2015-12-03:
@jaeilepp I'm not sure I follow. My system and Alex's system contain the same information, but just represent it differently. So if you e.g. add an annotation and crop the raw data, I don't see any problem with having / converting to negative times in the representation, and thus preserving the information.

jaeilepp, 2015-12-03:
Yeah, that was a mistake on my part. Realized it myself too after writing the comment.

dengemann, 2015-12-03:
We should use a new parameter with a clean annotation semantics. Combining events and annotations is than easy as semantics are clear defined.  I'm afraid of a new objects here and the maintenance they bring with, let's stick with numpy arrays and simple structures. My 2 cents.

jaeilepp, 2015-12-10:
Yeah, I agree with Denis now. I noticed that when reading events from kit files (with files for testing), the second column has some data. So there are already semantics tied to it.

dgwakeman, 2015-12-11:
fwiw, I already do this on my neuromag system. I simply raise a trigger when a stimulus appears and don't lower it until the stimulus disappears. Thus the length of presentation is present in the file directly.

agramfort, 2015-12-11:
yes we considered it. But it does not work when annotations are overlapping
eg blink during a certain sleep stage.

dgwakeman, 2015-12-11:
It works fine in my cases. If my event is code 1 and the button response is code 255. I can still separate them using the "ignore" feature in the C code, or just deconvolving the signal into their binary values. Then I get the length of all.

jaeilepp, 2015-12-12:
The overlapping events can be masked out in MNE, so basically this is a question of where and how we want to separate the triggers. It would be nice to have separate annotations for events and segments, so it would be easier to deal with them separately (visualization, rejection etc.)

Eric89GXL, 2016-02-03:
Closing, we have this now. Thanks @jaeilepp

closed 2016-02-03 by Eric89GXL

choldgraf, 2016-02-04:
Hey all - I know this issue has been closed and it looks like people's needs have been met here. That said, I just wanted to share a code snippet for an `Events` class that I've been using to keep track of metadata, trial lengths, etc for my trials. I've found it useful for quickly generating MNE events arrays using different kinds of trial types etc. Maybe somebody will find it useful :)

https://github.com/choldgraf/ecogtools/blob/master/mneutils.py#L35

If you guys think there's any place in the MNE repo for this, then I'm happy to contribute, but it sounds like people are satisfied already.