mne-tools/mne-python #2672: reading eeglab set files

mmagnuski, 2015-12-05:
I was wondering if you have any tools for reading eeglab `.set` files (along with `.fdt`) and converting to mne-python representation?
It would have seemed easy to do if not for the fact that sometimes (always?) `scipy.io.loadmat` fails in reading eeglab `.set` files (eeglab `.set` files are just `.mat` files). See my issue on `scipy` repo [here](https://github.com/scipy/scipy/issues/5516).

jasmainak, 2015-12-05:
Not at the moment, but I think this would be a useful addition definitely.

mmagnuski, 2015-12-06:
I would be happy to dig into this, but I would have to use something else than `scipy.io.loadmat` to open the `.set` files. I'll try using `h5py` or `h5io`...

jasmainak, 2015-12-06:
I tried the sample data from EEGLAB and it seems to be loading fine using scipy. Let us know what solution you find for your file.

mmagnuski, 2015-12-06:
Did you try the file I link to in the issue [here](https://github.com/scipy/scipy/issues/5516)?

mmagnuski, 2015-12-06:
I managed to open datasets from a different study with `loadmat`. I'll investigate this in the next couple of days.

jasmainak, 2015-12-06:
I just tried and both files work fine for me.

jasmainak, 2015-12-06:
... but I'm on python2.7

jona-sassenhagen, 2015-12-06:
This would be cool to have. Would you also do EEGLAB ICA files? :)
As a workaround, I've previously used the BVA export function of EEGLAB.

mmagnuski, 2015-12-06:
What do you mean by EEGLAB ICA files? 
eeglab keeps everything packed in the EEG structure (the `.set` file; the eeg signal is usually kept as float32 in a separate `.fdt` file). ICA weights are kept in `EEG.icaweights` and `EEG.icasphere`. I was planning to extract these too.

jona-sassenhagen, 2015-12-06:
Yes, I meant writing out the EEGLAB ICA structure to mne/fif ICA objects.

jasmainak, 2015-12-06:
I'm on it. I'll make a PR soon. @jona-sassenhagen you can lend a helping hand if you want ;) I'm not super-familiar with EEGLAB.

closed 2015-12-23 by agramfort

mmagnuski, 2016-06-10:
Just BTW - I still can't import that file (or any other file for that study) I linked at the [scipy issue](https://github.com/scipy/scipy/issues/5516) through `read_epochs_eeglab` :cry: (on python 3 at least)

jona-sassenhagen, 2016-06-10:
A practical workaround might be downsampling in EEGLAB.

mmagnuski, 2016-06-10:
The samling rate was rather low for that file (250) but I could try downsampling. The funny thing is that `MAT.jl` (Julia lib for reading/writing mat files) handles the file without problems.

agramfort, 2016-06-10:
can you share the file?

mmagnuski, 2016-06-10:
Right, I forgot that the previous dropbox link doesn't work anymore.
[Here is a working link](https://www.dropbox.com/sh/ek2dbhskfiurklx/AACEIgA_ywA_SW4MsQt7NgMJa?dl=0) - just one `.set` and corresponding`.fdt`.

agramfort, 2016-06-10:
`io.loadmat('faces_e_22.set')` works fine with me with

In [4]: scipy.__version__
Out[4]: '0.17.0'

jasmainak, 2016-06-11:
I can confirm that the issue can be reproduced in python 3.5

jasmainak, 2016-06-11:
@mmagnuski it seems more like a scipy issue to me than an mne-python one. You can get it working by switching to python 2.7

mmagnuski, 2016-06-11:
Thanks for checking! 
Yes, I have scipy issue opened for this but it was inactive for quite long. I've already switched to Julia + PyCall for this project, but I will try things out in python 2.7 conda env.

agramfort, 2016-06-12:
pb solved:

https://github.com/scipy/scipy/issues/5516#issuecomment-225413189

use:

io.loadmat('faces_e_22.set', uint16_codec='latin1')

we should expose the uint16_codec parameter in the eeglab reader.

@mmagnuski can you give it a try?

mmagnuski, 2016-06-12:
Thanks, works very well!

agramfort, 2016-06-12:
cool

can you do the PR to expose the codec param in eeglab reader?

mmagnuski, 2016-06-12:
Sure, I can do that tomorrow.

KirstensGitHub, 2016-09-07:
Hi, I am working in python MNE with a .set file from EEGlab(matlab). I have imported the data using:

x=mne.io.read_raw_eeglab('Subject1.set')

but am trying to follow some tutorials and always run into the following error:
"IOError: [Errno 2] No such file or directory: u'Subject1.fdt'"

One line which gives this error is ln[8] of the following tutorial:
https://plot.ly/ipython-notebooks/mne-tutorial/

The error arises basically any time the word "time" appears

Help?! Thank you!! :)

jasmainak, 2016-09-07:
Hi @KirstensGitHub could you share your `Subject1.set` file? Usually, the EEGLAB reader tries to find the `.fdt` file if that's pointed to in the `.set` file. Is your data also in the `.set` file? The reason you get your error later is probably because `preload=False`.

KirstensGitHub, 2016-09-07:
I see! 
location: **KirstensGitHub/Mich_collab/Subject1.set**

I had seen somewhere after loading that the fdt file did exist, but the computer seemed to not find/recognize it.  

I just tried **x=mne.io.read_raw_eeglab('Subject1.set', preload=True)** but got the same error..

jasmainak, 2016-09-07:
what is the location of the '.fdt' file? It is typically in the data field of your mat file structure. Can you share both files (*.set and *.fdt) on dropbox or privately by email so we can diagnose further?

KirstensGitHub, 2016-09-08:
Ah, I feel so embarrassed. I just had the .fdt file saved in the wrong place... *blush*
All working now!

Thanks so much for your quick replies! I'm relatively new to GitHub and have never seen an "issues" page/discussion like this before- very cool!

KirstensGitHub, 2016-09-25:
another file-type-related question: is there a way to use mne.read_evokeds on the .fdt and .set files?

jasmainak, 2016-09-25:
@KirstensGitHub this is not possible as of now. I suppose we could theoretically support it. But I don't know what the other developers have to say. I'm sure @agramfort has an opinion on this. @KirstensGitHub isn't it possible for you to save the epochs and then average it in MNE?

jona-sassenhagen, 2016-09-25:
Is the EEGLAB format intended to store evoked data? I think it's intended for epochs and continuous.

EEGLAB is really not about evoked stuff, but about single trial stuff.

jasmainak, 2016-09-25:
This is indeed what I thought ...

zhang470, 2017-01-14:
On my Windows 10 system, the mne.io.read_raw_eeglab reader always tried to append ".mat" to the .set file name with the error No such file or directory: 'xxxx.set.mat'. I tried reading the same EEG data in edf format , and mne.io.read_raw_edf() worked just fine.  Not sure why .mat is automatically appended to the .set file name...

agramfort, 2017-01-14:
what version of scipy are you using?

zhang470, 2017-01-14:
scipy version: 0.18.1  

platform : win-64
          conda version : 4.2.15
       conda is private : False
      conda-env version : 4.2.15
    conda-build version : 2.0.2
         python version : 2.7.13.final.0
       requests version : 2.12.4

jasmainak, 2017-01-14:
can you paste the error and also what exactly you are passing as `input_fname`?

mmagnuski, 2017-01-14:
The error @KirstensGitHub had (didn't notice that discussion before) reminds me of another problem when reading set files - eeglab, when reading set + fdt files checks the filename in EEG.data but also the name of the set file (mne python only checks EEG.data). In some cases these names are not exactly the same (ie EEG.data.replace('.fdt', '.set') != set_filename) so might be good not to rely (at least not only) on EEG.data field.

zhang470, 2017-01-14:
I just realize I need to do "import scipy" first. Now mne.io.read_raw_eeglab () can read raw EEG .set files just fine. But I could not get the mne.read_epochs_eeglab () to read in the epoched datasets in EEGLAB to work. The reader always showed AttributeError. 

    epochs = mne.read_epochs_eeglab(fname)
  File "C:\Users\Yang\Anaconda3\lib\site-packages\mne\io\eeglab\eeglab.py", line 241, in read_epochs_eeglab
    uint16_codec=uint16_codec)
  File "<string>", line 2, in __init__
  File "C:\Users\Yang\Anaconda3\lib\site-packages\mne\utils.py", line 708, in verbose
    return function(*args, **kwargs)
  File "C:\Users\Yang\Anaconda3\lib\site-packages\mne\io\eeglab\eeglab.py", line 484, in __init__
    event_type = '/'.join(ep.eventtype.tolist())
AttributeError: 'int' object has no attribute 'tolist'

agramfort, 2017-01-15:
please share the epochs EEGLAB dataset

zhang470, 2017-01-17:
Here are the epoched files from EEGLAB. I tested  several other epoched files from different EEG experiments using different versions of EEGLAB. They all showed the same error. Thanks. 
http://zhanglab.wikidot.com/local--files/method/KH1_TONE1K_IGN1_epoch.fdt
http://zhanglab.wikidot.com/local--files/method/KH1_TONE1K_IGN1_epoch.set

barisakin98, 2019-12-25:
Hi, 
I am working on a .fdt file. While ı am trying to read it with python scipy-loadmat, I am taking this error:
ValueError: Unknown mat file type, version 147, 65

what should I do in this situation, thanks for your helps!

jasmainak, 2019-12-25:
Can you please provide the full traceback? Also for questions, the right place to ask is the mailing list. You should post on github only if you are reasonably sure you have a bug

agramfort, 2019-12-25:
Fdt is not matlab file. It’s a companion file to a .set file that is actually a .mat file