mne-tools/mne-python #2674: MultiTaskLasso and GAT

kingjr, 2015-12-06:
@jona-sassenhagen @agramfort @dengemann 

On my computer, the `MultiTaskLasso` from sklearn seems to be unable to run parallel predictions (fit is fine though). I haven't run an investigation yet... Do you encounter the same issue?

```
import numpy as np
import mne
from mne.datasets import sample
from mne.decoding import GeneralizationAcrossTime
from sklearn.preprocessing import StandardScaler
from sklearn.pipeline import make_pipeline
from sklearn.linear_model import MultiTaskLasso
from sklearn.cross_validation import KFold

# preproc
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
events_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
raw = mne.io.Raw(raw_fname, preload=True)
picks = mne.pick_types(raw.info, meg=True, exclude='bads')  # Pick MEG channels
events = mne.read_events(events_fname)
event_id = {'AudL': 1, 'AudR': 2, 'VisL': 3, 'VisR': 4}
epochs = mne.Epochs(raw, events, event_id, -0.050, 0.600, proj=True,
                    picks=picks, baseline=None, preload=True,
                    reject=dict(mag=5e-12), verbose=False)

# Fit multiple conditions at once
events_side = epochs.events[:, 2] % 2
events_modality = epochs.events[:, 2] > 2
Y = np.vstack((events_side, events_modality)).T

# GAT
clf = make_pipeline(StandardScaler(), MultiTaskLasso(alpha=.01))
gat = GeneralizationAcrossTime(clf=clf, cv=KFold(len(Y)), n_jobs=-1)
gat.fit(epochs, y=Y)
gat.predict(epochs)  # seems to run a single core, and thus takes for ever
```

jona-sassenhagen, 2015-12-06:
Works fine in parallel for me.

jona-sassenhagen, 2015-12-06:
```
gat.predict(epochs)  # seems to run a single core, and thus takes for ever
[Parallel(n_jobs=48)]: Done   1 out of   1 | elapsed:  1.2min remaining:    0.0s
[Parallel(n_jobs=48)]: Done   1 out of   1 | elapsed:  1.2min finished
```

agramfort, 2015-12-06:
no idea

kingjr, 2015-12-06:
I updated to sklearn v.17 and it works now. 

Thanks!

kingjr, 2015-12-06:
Arg, updating it through conda made the problem reappear.. And removing or reinstalling sklearn through pip doesn't change anything anymore. I'll re-open it until I find the fix.

kingjr, 2016-02-17:
https://github.com/mne-tools/mne-python/pull/2856 fixes it. I suspect this is the list of list __repr__ and/or the nested lists concatenation that caused the issue. No problem with np.arrays anymore.

closed 2016-02-17 by kingjr