mne-tools/mne-python #2685: no inv for LCMV

yuval-harpaz, 2015-12-09:
Dear MNE team
I am trying to compute LCMV(g2) for epilepsy data. for this I tried using lcmv_raw. I want to compute the beamformer weights on raw data (500sec) and then apply them to short data epochs (0.25sec) , compute g2 (estimator for kurtosis) and save the sum of that as source strength. I can't do it because if I use the whole data I get memory error of course (the stc is too big), and computing LCMV for short data is not the way to go. I suggest doing LCMV in two stages, giving the inv first, then applying it. in the meantime, is there a way for me to get the weights while doing LCMV?
thanks

agramfort, 2015-12-09:
the solution only depends on noise_cov and data_cov which are parameters.
You can therefore call lcmv_raw on smaller slices of data. It will be
equivalent although slightly slower.

would that be good enough?

yuval-harpaz, 2015-12-10:
right, sorry, should have thought of that.
in any case, it would be more 'open source' like if the weights are
accessible. In SAM beamforming you always get the weights (but not the
forward model and other stuff), and in fieldtrip you get them if you
specify cfg.keepfilter='yes'. but this would be lower priority
thanks

agramfort, 2015-12-10:
write us how you would like to us it (2-3 lines code snippet) and we'll see
what we can do.

thx

yuval-harpaz, 2015-12-10:
I would like to have a nsourse x nchan np.ndarray so I could then multiply
it by the data to get raw source timecourse (no depth normalization). I
suggest to give it as a second output variable for lcmv_raw, but do which
ever cheepest way for you ...
thanks

dengemann, 2015-12-10:
It would be a trivial edit to the code, but an API change. We could of course habe some return_weights parameter that equals to False by default. Option2: maintain a copy of the code privately and extract what you wish for debugging purposes. Option 3: private function that returns the weights. The you call that one for debugging your analysis.

yuval-harpaz, 2015-12-10:
all works for me. As is, I am analysing 10min recording and it is going to
take the whole day

agramfort, 2015-12-10:
>
> all works for me. As is, I am analysing 10min recording and it is going to
> take the whole day
>

lcmv_raw for 10min of data processed by chunks takes 1 day???

yuval-harpaz, 2015-12-10:
well, I got to segment 600 of 1819 (quarter seconds segments) and it is
11:30 here, it will take a few hours more
I do it with this loop

Kur=np.zeros((fwd['nsource']))
for sampi in range(0,len(samps)):
    start=samps[sampi]
    stop=start+dur
    stc = mne.beamformer.lcmv_raw(raw, fwd, empty_cov,
data_cov,start=start, stop=stop)
    kur=g2(stc.data)
    kur[kur<0]=0
    Kur=Kur+kur
    print("DONE "+str(sampi+1)+" of "+str(len(samps)))

agramfort, 2015-12-10:
one option would be to add support for lazy stc as we do for minimum_norm
... it means passing a tuple (kernel, data) to SourceEstimate.

agramfort, 2017-07-22:
this is done is https://github.com/mne-tools/mne-python/pull/4389

closed 2017-07-22 by agramfort