mne-tools/mne-python #2687: Sklearn warnings for cov

Eric89GXL, 2015-12-09:
I'm running latest `sklearn` and I get this when doing `mne.compute_covariance`:
```
/home/larsoner/.local/lib/python2.7/site-packages/sklearn/cross_validation.py:42: DeprecationWarning: This module has been deprecated in favor of the model_selection module into which all the refactored classes and functions are moved. Also note that the interface of the new CV iterators are different from that of this module. This module will be removed in 0.20.
  "This module will be removed in 0.20.", DeprecationWarning)
/home/larsoner/.local/lib/python2.7/site-packages/sklearn/grid_search.py:43: DeprecationWarning: This module has been deprecated in favor of the model_selection module into which all the refactored classes and functions are moved. This module will be removed in 0.20.
  DeprecationWarning)
```
@agramfort @dengemann any time to look into how to fix this before 0.11 release (in the next week probably), or should we push to 0.12?

agramfort, 2015-12-10:
git grep cross_validation

then rename cross_validation to model_selection

and make sure tests pass and updated examples also

however we should support both versions for some time but just in the code
(not in examples)

Eric89GXL, 2015-12-11:
Not that simple unfortunately, just spent half an hour trying to fix `time_gen.py` but it's not trivial. There are API changes from `cross_validation` to `model_selection` in terms of how iteration is done, and when different parameters are passed. So @dengemann or @kingjr (most recent authors in that vicinity) will probably have to take a look. Moving to 0.12.

agramfort, 2015-12-11:
do you have a WIP PR I could take over during we?

Eric89GXL, 2015-12-11:
No, I only had one or two good lines and a bunch of not-working junk after that so I `reset --hard`. Your idea was a good start, though. You can run `mne/tests/test_cov.py` and `mne/decoding/tests` to see the warnings.

PR #2739 (2016-01-26): WIP : add support for new version of cross-val objects
try to switch to new sklearn cross-val objects.

I made it work for cv but fore time gen it's way more convoluted.

@kingjr and maybe @dengemann if you have the time to take over this PR to make it work with time gen it would be awesome.

Thank you

closes #2687
closed 2016-01-26 by agramfort