mne-tools/mne-python #2688: Issue reading in FIF files from Fieldtrip

choldgraf, 2015-12-09:
I'm trying to read in a FIF file that was created with the `fildtrip2fif` function in Fieldtrip. When I try to change channel types with the loaded Raw file, I am getting a KeyError:

```python
KeyError                                  Traceback (most recent call last)
<ipython-input-6-3edc023a40c6> in <module>()
----> 1 data.set_channel_types({'EEG 1': u'eeg'})

/home/knight/holdgraf/anaconda/lib/python2.7/site-packages/mne/channels/channels.pyc in set_channel_types(self, mapping)
    276                 warnings.warn("The unit for Channel %s has changed "
    277                               "from %s to %s." % (ch_name,
--> 278                                                   unit2human[unit_old],
    279                                                   unit2human[unit_new]))
    280             self.info['chs'][c_ind]['unit'] = human2unit[ch_type]

KeyError: 0
```

It looks like the units of the FIF file say channels are of type `0`, corresponding to "None" I assume, but that MNE assumes that "None" is mapped to `-1` instead, thus the conflict. I'm not sure if this is a mistake of MNE, or a mistake of Fieldtrip. Any thoughts?

choldgraf, 2015-12-09:
Pinging @yvofon because she had this problem too

choldgraf, 2015-12-09:
Maybe @robertoostenveld has thoughts as well?

Eric89GXL, 2015-12-09:
We do have it as -1:

https://github.com/mne-tools/mne-python/blob/master/mne/io/constants.py#L705

Which comes from `fiff_types.h`:
```
#define FIFF_UNIT_NONE -1
```
But maybe this is what FieldTrip uses, which is for multipliers?
```
#define FIFF_UNITM_NONE 0
```

robertoostenveld, 2015-12-09:
the code used in fieldtrip mainly consists of a call to mne-matlab, see https://github.com/fieldtrip/fieldtrip/blob/master/fieldtrip2fiff.m

I wonder whether it might be going wrong in the subfunction sens2fiff (part of fieldtrip2fiff) or in the fiff_define_constants function (part of mne-matlab).

robertoostenveld, 2015-12-09:
line 272 in fieldtrip2fiff states

    chs(1,k).unit_mul     = 0;

and "mul" is multiplier?

Eric89GXL, 2015-12-09:
At our end it looks like the error comes from the `unit`, though, and not `unit_mul`. I wonder if it's this line:

chs(1,k).unit         = NaN;

And somehow it gets translated to 0 since it's stored as `int` and not `double` when written out to FIF format?

choldgraf, 2015-12-09:
from my matlab I get:

```matlab
>> int16(nan)

ans =

      0

>> double(nan)

ans =

   NaN
```

Eric89GXL, 2015-12-09:
@choldgraf can you do e.g. `import pdb; pdb.pm()` and get us the values for ch['kind'], ch['coil_type'], ch['unit'], ch['unit_mul'], ch['coord_frame'], and ch['loc']? If these are all zero it suggests that conversion from NaN to zero on writing is the culprit (which I suspect is the case based on what you wrote above), and I think the correct fix is for FieldTrip to set the value to -1 (Unknown) instead of NaN (which gets translated to zero).

Eric89GXL, 2015-12-09:
Or I guess mne-matlab (less explicitly) could also issue a fix to convert `NaN` channel type to -1...

Eric89GXL, 2015-12-09:
@choldgraf if you feel like trying the post-mortem but have never done it, you'll need to use command `p` for print like:
```
import pdb
pdb.pm()
(pdb) p self.info['chs'][c_ind]['kind']
...
```

choldgraf, 2015-12-09:
Here you go (doing this with the notebook %debug magic which I think does the same thing basically:

```python
for i in ['kind', 'coil_type', 'unit', 'unit_mul', 'coord_frame', 'loc']: print('{1}: {0}\n'.format(self.info['chs'][c_ind][i], i))
kind: 2

coil_type: 0

unit: 0

unit_mul: 0

coord_frame: 0

loc: [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
```

Eric89GXL, 2015-12-09:
Thanks. Looks like it's indeed the NaN values being converted to 0 that
sets the wrong value

agramfort, 2015-12-10:
can you check that a fix at:

https://github.com/mne-tools/mne-matlab/blob/master/matlab/fiff_write_ch_info.m#L125

fixes your pb? Basically write -1 if ch.unit == NaN

choldgraf, 2015-12-10:
I'll see if this works w/ the original dataset today.

choldgraf, 2015-12-11:
Hey - we gave this shot today and ran into errors with mne-matlab. Nobody in the lab has any experience using this so we might just be doing something stupid. 

```matlab
old_data = './old_data.fif';
raw = fiff_setup_read_raw(old_data, false);

%% Write the chs
newfile = './new_data.fif';
raw.info.chs(1).unit = nan;
raw.info.chs(1).unit_mul = nan;
f = fopen(newfile);
fiff_write_ch_info(f, raw.info.chs(1));
fclose(f);

%% Read the chs
rawnew = fiff_setup_read_raw(newfile, false);
```

And it spits out this error:

```matlab
Opening raw data file ./new_data.fif...
Error using fiff_open (line 56)
file does not start with a file id tag

Error in fiff_setup_read_raw (line 71)
[ fid, tree ] = fiff_open(fname);
```

Note this is without me modifying any of the MNE matlab files yet...

agramfort, 2015-12-11:
you cannot write with fiff_write_ch_info outside of a measurement info
block.

choldgraf, 2015-12-12:
OK - not sure why but the fix doesn't seem to change the problem. Here's the script I'm running:

I added these lines to fiff_write_ch_info.m:

```matlab
if isnan(ch.unit)
    disp('Changed unit...')
    ch.unit = -1;
end
count = fwrite(fid,int32(ch.unit),'int32');
if count ~= 1
    error(me,'write failed');
end

if isnan(ch.unit_mul)
    disp('Changed unit_mul...')
    ch.unit_mul = -1;
end
count = fwrite(fid,int32(ch.unit_mul),'int32');
if count ~= 1
    error(me,'write failed');
end
```
And then ran this:

```matlab
% Read in the old data structure to edit
old_data = './old_data.fif';
raw = fiff_setup_read_raw(old_data, false);
raw.info.chs(1).unit = nan;
raw.info.chs(1).unit_mul = nan;
raw.info.chs(1)

%% Write the chs
newfile = './new_data.fif';
f = fopen(newfile, 'a');
fiff_write_ch_info(f, raw.info.chs(1));
fclose(f);

%% Read the chs
rawnew = fiff_setup_read_raw(newfile, false);
rawnew.info.chs(1)
```

Which resulted in this output:
```matlab
Opening raw data file ./old_data.fif...
	Range : 0 ... 1372387  =      0.000 ...   899.407 secs
Ready.

ans = 

         scanno: 1
          logno: 1
           kind: 0
          range: 1
            cal: 1
      coil_type: 0
            loc: [12x1 double]
     coil_trans: []
        eeg_loc: []
    coord_frame: 0
           unit: NaN
       unit_mul: NaN
        ch_name: 'EEG 1'

Changed unit...
Changed unit_mul...
Opening raw data file ./new_data.fif...
	Range : 0 ... 1372387  =      0.000 ...   899.407 secs
Ready.

ans = 

         scanno: 1
          logno: 1
           kind: 0
          range: 1
            cal: 1
      coil_type: 0
            loc: [12x1 double]
     coil_trans: []
        eeg_loc: []
    coord_frame: 0
           unit: 0
       unit_mul: 0
        ch_name: 'EEG 1'
```
Note that the unit and unit_mul fields are back to zero

Eric89GXL, 2015-12-18:
Can you try reading the data with `mne-python` to see if the `unit` is also `0` and not `-1` there? That might help tell you if it's a problem with writing (which it seems like you've fixed probably) or reading (which we haven't looked at yet).

choldgraf, 2015-12-18:
I think that I'm doing something wrong in the writing, because I tried changing the channel names manually and they didn't change either. Can someone confirm that I'm using the fiff_write_ch_info function correctly? I've never used mne-matlab before...

choldgraf, 2015-12-18:
(and to answer @Eric89GXL the problem was there in mne-python as well, which is why I think that it's an issue with me saving properly)

dgwakeman, 2015-12-18:
check how you do it. There is a nasty feature in the matlab code which copies the header if you have an active file id, so it can be very difficult to change certain fields. I remember having to delete a few strange fields to get it to write a brand new header/info fields. (sorry I'm not more specific I am reaching back ~6-7 years now in memory)

choldgraf, 2015-12-18:
hmmm - well the only active `fid` that I've created is the one I had to create in order to write to the new file. AFAIK that's the way to change the header of a file, right? You have to open it and then use `fiff_write_ch_info` for each channel? It doesn't seem like the call to read the header from the original `fif` file kept the `fid` open, as raw.fid returns -1...

choldgraf, 2015-12-18:
It must be happening in the writing process

Making this modification to the write_ch_info.... function:
```matlab
disp(ch.unit)
count = fwrite(fid,int32(ch.unit),'int32');
disp(int32(ch.unit))
```
will produce (correctly) -1 two times. But reading back in the file it writes to does not work. I must be doing something wrong here. Maybe somewhere in here?

```matlab
f = fopen(newfile, 'a');
for i = ix_chans;
    fiff_write_ch_info(f, chs(i));
end
fclose(f);
```
It returned an error if I had to write the whole `chs` structure at once, which is why I'm doing it channel by channel, but something tells me this is wrong.

agramfort, 2015-12-19:
please share a complete gist and an up to date branch of mne-matlab so we
can replicate the pb.

Eric89GXL, 2016-02-03:
@choldgraf any progress here? It would be nice to get this working properly.

choldgraf, 2016-02-03:
I haven't worked on this since the issue was opened. The problem is it's not my issue, but one that a labmate was having. So I haven't had much time to work on it...

robertoostenveld, 2016-02-04:
please do let me know if there is anything that should be done on the FieldTrip side to resolve this issue. FieldTrip development just switched over completely from subversion to git/github, so you can simply do a pull request on https://github.com/fieldtrip/fieldtrip

choldgraf, 2016-02-04:
Sounds good - I'll get to this eventually. I think that my friend fixed it somehow (maybe by specifying a channel type, as this only happened when that was missing). Will keep folks updated if I work on it.

dengemann, 2016-02-04:
@robertoostenveld great news! I hope we'll have a moment during the upcoming Paris BrainHack to take  look at interoperability ideas.

choldgraf, 2016-02-04:
yeah +1 to that as well! Is github the proper place to open issues etc from now on then? Thus far I've just been bugging Arjen :)

robertoostenveld, 2016-02-05:
Right now I am regularly cross-linking the two, which works ok. 

On http://bugzilla.fieldtriptoolbox.org/ we have more users and (old time) developers, so it is still easier to involve others in the discussion. On github there are at the moment only a few that will notice the issue. I will look into setting it up such that the usual suspects will also receive github notifications, as they now do when something gets posted on bugzilla. 

Of course if if is not only an issue, but also the resolution to the issue (i.e. pull request) then you are more than welcome to use github!

choldgraf, 2016-04-27:
hey all - I'm not sure if this ever got fixed or not. I didn't make any changes myself and I think my friend just found a workaround for loading in their data from Fieldtrip. Did anybody discuss this offline?

jona-sassenhagen, 2016-04-27:
At the Frankfurt workshop, we hacked together a function for reading in Fieldtrip epochs created with a specific MATLAB version if anybody cares ...

robertoostenveld, 2016-04-28:
please note that I welcome any suggestions to improve the fif exporter code in http://github.com/fieldtrip/fieldtrip

kingjr, 2016-04-28:
@jona-sassenhagen I do, also see https://github.com/mne-tools/mne-python/issues/2476

choldgraf, 2016-04-28:
@robertoostenveld I'll definitely keep this in mind if I talk to other folks in the lab that are using Fieldtrip. I pretty much stick with Python but there are more and more folks that are doing both python and Matlab, so maybe this will come up again.

Eric89GXL, 2016-11-18:
I'll close this since it is probably fixed (?) but feel free to reopen if it comes up again

closed 2016-11-18 by Eric89GXL