mne-tools/mne-python #2691: ENH: Make all readers read in ~100 MB blocks

Eric89GXL, 2015-12-10:
Currently our raw readers are not quite consistent in how they read data in a memory efficient way.

1. `ctf`, `edf`, and `fiff` have data stored in "blocks" of `N` samples, so should already be fairly memory efficient since we read them block-wise.

2. `nicolet` and the soon-to-be-added `eeglab` readers each use a reasonable read size (100 MB chunks), but should be refactored to use the same scheme/functions for setting limits (DRY).

3. `kit`, `brainvision`, `bti`, `egi`, and `kit` read in all of the data at once, and should be refactored to use the same read-size and functions as `nicolet` and `eeglab`.

This should help make our readers again more consistent, and improve memory consumption.

Eric89GXL, 2015-12-10:
Any volunteer for this one? If not I can do it.

Eric89GXL, 2015-12-10:
@jasmainak if you want to move toward this in #2676 by refactoring `nicolet` and your `eeglab` reader to use the same function for choosing how many samples to read in (based on the number of channels and bytes per sample) and the resulting read and data bounds it would already be helpful.

jasmainak, 2015-12-10:
yeah, I already discussed this refactoring with @jaeilepp . That would be a good start.

jaeilepp, 2015-12-11:
Yeah, I think I can take this one.

closed 2016-01-07 by agramfort