mne-tools/mne-python #2692: mne watershed

ImmanuelSamuel, 2015-12-11:
Hey guys, Is it possible to get a BEM for a t1 template? I tried to used watershed on mni template and it didn't work. It works on a standard t1.

agramfort, 2015-12-11:
I am sure their already exist good BEM surfaces for MNI template if it is
what you have in mind

dgwakeman, 2015-12-11:
you can rip the one out of SPM, I generated that based on the colin 27 scan. it isn't good the scans on most templates don't cover the whole head, and thus don't work well for BEMs. Generally, I just discourage people from using templates for source recon

ImmanuelSamuel, 2015-12-11:
I see .. Unfortunately I don't have the T1 for my participants. I was thinking of doing something like eLORETA. But then if I can get MNE to work then I can try multiple source recon methods and DICS. How about the 3T 18yo Schwartz ReactN32 that is in freesurfer?

dgwakeman, 2015-12-11:
I would just use the one from SPM, it is in their code somewhere. The weakness of the shape is unlikely to cause bigger problems than mapping data from one person to another.

ImmanuelSamuel, 2015-12-11:
@dgwakeman if you have time and since you already extracted the surfaces ... can you point me where to look in SPM and how to integrate with MNE. I following the tutorial and am at the point where I need the .surf files. Currently searching the SPM folder for clues.

ImmanuelSamuel, 2015-12-11:
OK I think I got it.

agramfort, 2015-12-11:
please share any insights here in case someone else has the goal

ImmanuelSamuel, 2015-12-11:
The surf files are in spmX/canonical/. I am testing them now. Freesurfer also has the same BEM data but they are in mat format. Dont know the surf format to convert time. Apparently they have a better estimate of the data by correcting for anti-aliasing.

ImmanuelSamuel, 2015-12-11:
However they are in .gii format need to convert them to surf. @dgwakeman Any ideas?

ImmanuelSamuel, 2015-12-11:
mris_convert does not support converting to .surf format.

agramfort, 2015-12-11:
nibabel can read gii and mne can save .surf files

ImmanuelSamuel, 2015-12-11:
```mris_convert <SPMfile> <surf file>

closed 2015-12-12 by ImmanuelSamuel