mne-tools/mne-python #2693: ENH: improve evoked.plot_image with channel layouts

jona-sassenhagen, 2015-12-11:
Evoked images (heatmap, channels on y axis) are often plotted grouped by hemisphere, e.g.
[image]

This is IMO nicer to interpret than the current MNE function. I have homegrown routines, is anyone interested in me trying to integrate them in MNE *eventually*? API would probably be adding a `grouping=` keyword to `evoked.plot_image`.

jasmainak, 2015-12-11:
why don't you simply use `picks` and pass the axis to `evoked.plot_image`?

jona-sassenhagen, 2015-12-11:
It's a convenience function. Having to do it manually - including automatically assigning channels to subplots - just clutters up my notebooks. There's also redundant axis labelling etc.

You could just use plt.imshow(evoked.data) in the first place after all, right?

But it's not a key feature, I'm just wondering if anybody cares.

jona-sassenhagen, 2015-12-21:
Closing this because there seems to be little interest and implementing it for non-10/20 channel names would be annoying.

dengemann, 2015-12-21:
@jona-sassenhagen I think this is cool. An example would also do it to me. Then more people might be attracted, which in turn might boost collective motivations to write an API for this.

jona-sassenhagen, 2015-12-21:
Okay, how about a joint example for this + the permutation test function? Cause figuring out a way to easily set up connectivity matrices for 10/20 data wasn't trivial for me.

dengemann, 2015-12-21:
> Cause figuring out a way to easily set up connectivity matrices for 10/20 data wasn't trivial for me. 

didn't we have these in the FT definitions? Do we have some suited example data for this? the sensor permuation clustering test example is already a bit long. Maybe We can have an example that is more devoted to 10-20 data?

jona-sassenhagen, 2015-12-21:
>didn't we have these in the FT definitions?

Where? :) Yes, I meant a new example, specifically for 10/20 data. (Which probably means I need to finally submit one of my own files, as I've been promising for ages.)

jona-sassenhagen, 2016-03-05:
@dengemann 

http://nbviewer.jupyter.org/gist/jona-sassenhagen/2f69c50f1cdd1fd0c9b4

dengemann, 2016-03-05:
I'm wondering where to fit this in. If you want it in MNE the example is probably too complex as it is. What would be your vision? The API would be EEG only or mono-channel type? The viz function could take a parameter for a grouping function, then the example could consist of the code seen minus the second  function definition + passing the first.

agramfort, 2016-03-05:
we have the notion of selection for defining group of channels. If we do
it, we should cook something using selection, eventually having a function
to find the channels based on selection name.

kingjr, 2016-03-05:
The visual left right dataset should be good for this.

I would personally prefer having a single box where channels are sorted by
x or y coordinates, add a rainbow on the y axis and add the corresponding
rainbow topo on the top left of the im plot.

My 2cc

jona-sassenhagen, 2016-04-07:
@kingjr do people do that in publications? I think I've never seen that before. Not saying it sounds bad ...

kingjr, 2016-04-07:
On the top of my mind something like this:
[image]

Wyart & Summerfield, Neuron 2012

Adding the option to use the y axis as the fronto-anterior, or left-right would be useful when the components of interest are visual/motor respectively.

jona-sassenhagen, 2016-04-24:
Hm ... we still don't have a clear idea of where this is heading. I think eventually, I'd like to extend evoked.plot_image. @agramfort @dengemann would you in principle be open for adding a few params to evoked.plot_image? Including what @kingjr has suggested?
I'd then think about a proper API.

I think this is particularly useful to show masked images, ie., for statistical tests.

agramfort, 2016-04-25:
no objection

dengemann, 2016-04-25:
Also fine with me.

jona-sassenhagen, 2016-04-25:
Okay I'll get on it. (Eventually.)

jona-sassenhagen, 2016-05-12:
Working on this, I've come upon the question for why viz._plot_evoked exists, handling the code for both images and butterfly. Wouldn't it make more sense to refactor these two apart? Or is there a strong rationale to keeping them in one function, depending on the plot_type parameter?

spatial_colors has made type=='butterfly' a lot more complicated, and if I finish this one, 'image' will also be more complicated.

Eric89GXL, 2016-05-12:
Without looking deeply into the code I don't know for sure, but it could be historical, or it could be that it's more DRY to keep them together. If you want to split them, maybe make a first PR to do only that to see if it makes the code cleaner, or at least as clean as before. That would be a useful test.

jona-sassenhagen, 2016-06-30:
Eventually, I will do this.

With contours ...
[image]

@mmagnuski maybe you're interested.

dengemann, 2016-06-30:
Yes it is historical, we have no strong commitments to maintaining private
functions. Anything that deconvolutes the private codes would be good.
Afterall we should facilitate viz contribs from newcomers. Readable code is
a start.

mmagnuski, 2016-06-30:
@jona-sassenhagen 
Looks good!

I often like masking the insignificant (or otherwise uninteresting) values with transparency. I actually have lots of matlab code for this, one example:
[image]

The channels are grouped differently here (frontal, central, posterior and each group has channels from left (top) to right (bottom)) - and it would be nice if both ways were possible here (in mne). 
If you want to take a look at the matlab code it is [here](https://github.com/mmagnuski/braintools/blob/master/plt/maskitsweet.m) and [here](https://github.com/mmagnuski/braintools/blob/master/plt/plot_time_elec.m) for example (but it's messy).  BTW masking transparently with colors not from the current colormap is a nightmare in matlab (or maybe just used to be) but a pleasure in matplotlib :)

jona-sassenhagen, 2016-07-05:
> a pleasure in matplotlib

I see you volunteering!

Example looks great BTW. I'd love to eventually have a plot_x_joint method for everything.

mmagnuski, 2016-07-09:
@jona-sassenhagen 
I could add some code for setting transparancy mask and maybe the lines too (currently the way I do the lines in matlab is really dumb) next week. But I wouldn't be able to do much more. :( (maybe some time later...)

jona-sassenhagen, 2016-07-09:
I'd like to do it actually, I enjoy writing viz code. But if you wanna do it, you'd probably do it faster and better.

mmagnuski, 2016-07-09:
No, that's better if you do it. I won't have much time for writing it in near future but will happily comment on your PR. Also I doubt I would do it better or faster.
And I still have some properties-related stuff to add.

jona-sassenhagen, 2018-03-31:
I am thinking about doing an example on this later. With the merging of #5010, probably don't need main code support. Closing.

closed 2018-03-31 by jona-sassenhagen