mne-tools/mne-python #2694: BUG: something wrong with spatial_colors legend

jona-sassenhagen, 2015-12-11:
Something wrong with the legend in files that should be fairly regular - no bad channels, all EEG ... but the legend colors seem completely random. (The line colors are fine!)

The tests look fine and another experiment of mine looks fine, it's only for some layouts.

[image]

I'll see if I can find what it's about, but I'm also attaching an evoked file that has the issue: [bad_example-ave.fif.gz](https://github.com/mne-tools/mne-python/files/59396/bad_example-ave.fif.gz)
@jaeilepp

jona-sassenhagen, 2015-12-11:
Jakko, the problem is  that `names = np.intersect1d(layout.names, used_nm)` returns a sorted list, but we don't want them to be sorted, but in original order (10/20 EEG chans have nonalphabetic order, it only works for MEG and some EEG layouts due to alphabetic channel layouts) . I'm not sure I understand the exact purpose of the line, and what could happen for specific channel types (the ones we need `_clean_names` for). Can we do `names = [name for name in layout.names if name in used_nm]` instead, or does that break something? It fixes my problems.

agramfort, 2015-12-11:
sounds like the good fix !

A

PR #2695 (2015-12-11): MRG: Fix nonalphabetically sorted chan names for spatial_colors legend
Works on my data, works on example, let's see what Travis says.

Closes #2694
closed 2015-12-11 by jasmainak