mne-tools/mne-python #2700: Incorrect EEG lead field matrices due to missing reference electrode location

emilyps14, 2015-12-14:
The location of the EEG reference electrode is not stored in the raw data, leading to incorrect lead field matrices.

The EEG reference location is used by the forward solution in the BEM calculation, accounting for the fact that the reference is effectively subtracted out from the signal during acquisition. 

The forward solution code in both MNE C and MNE Python expects the reference electrode location to be stored in the second column of the position matrix (in python: info[chs][<#>][eeg_loc]). In the Python distribution, there is no way to populate this column.

For my data at least, the information is stored in the hpts montage file, so it would make sense to load the reference electrode location when the montage is loaded.

From the mne_analysis mailing list:
https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2015-December/003012.html

Code to replicate the mismatch:
https://drive.google.com/file/d/0B8PjVL8SDL7IdDVHWE40YTJSZnM/view?usp=drive_web

agramfort, 2015-12-15:
Emily you're on my todo list but I need to find the time to have a look.

I know that Eric recently changed the eeg_loc / loc field to avoid
duplicate information.

thank you for your patience.

Eric89GXL, 2015-12-16:
Okay I had a second to look. So just to be clear, what commands did you use in Python to set the reference? In your email you mentioned using these:
```
raw = mne.io.read_raw_fif(rawfile,preload=True)
mne.io.add_reference_channels(raw,'EEG000',copy=False)
raw.set_montage(montage)
mne.io.set_eeg_reference(raw,['EEG000'],copy=False)
```
And that these did not set the `eeg_loc` (now just values in `loc`) in mne-python. They probably need to, but we'll have to think about the API since you can really only have one reference channel (and those functions allow you to use more than one).

How did you accomplish adding an EEG reference channel in MNE-C?

Eric89GXL, 2015-12-16:
@agramfort I think the first-level fix would actually be to prevent forward solutions from being constructed if data have a `custom_ref` flag applied at all, unless we can transform our custom references to act as projection matrices instead. Otherwise I can't see an easy way to get a general case of having `N` reference channels to work with the forward code.

emilyps14, 2015-12-16:
In the file I sent, I didn't set the reference at all for the python dataset (propofol_raw), since I couldn't figure out how to do it. I played around with the custom referencing functions just in case they could have handled it, but since they didn't I dropped it. 

I set the montage using: 

    montage = mne.channels.read_montage(montage_hpts, transform=True, unit='mm')
    raw.set_montage(montage)

The montage file I used is this one:
https://drive.google.com/open?id=0B8PjVL8SDL7IQWxtaXR6VXlwal9mQ1pnQUdMOGJ3Z2JiY2hN
which is a modified version of this one:
https://drive.google.com/open?id=0B8PjVL8SDL7IQWxtaXR6VXlwal9mQ1pnQUdMOGJ3Z2JiY2hN
which was converted from a FIF:
https://drive.google.com/open?id=0B8PjVL8SDL7IMHFKeE5hZGpsWnpMNmJGd1JNcW03eVZCZnB3

It would be nice if there were a function that identifies the reference based on the montage file, since that's where the information is (in electrode 0). Seems like read_montage and set_montage could do that. 

(It would also be nice if there were a way to load a fif file as a montage, so I wouldn't have to do the conversion and modification. Seems like I'm the only person who has digitizer info in a fif file, though, so that may not be a priority)

In terms of the MNE-C data (andres_ev), these calls took eeganes07.fif as input (so could have set the reference):

    mne_brain_vision2fiff --header level2.vhdr --dig eeganes07.fif --out level2_raw.fif --split 585
    mne_check_eeg_locations --file level2_raw.fif --dig eeganes07.fif --fix

I'm currently using  `mne_check_eeg_locations` on my python files to set the reference, so I know that works. I wouldn't be surprised if `mne_brain_vision2fiff` also sets it.

Eric89GXL, 2015-12-16:
> It would be nice if there were a function that identifies the reference based on the montage file, since that's where the information is (in electrode 0). Seems like read_montage and set_montage could do that.

I'm not too familiar with montage files -- is it always the case that electrode 0 is the reference electrode (if there is one), or is there some way to tell from the file? If there is a way to tell from the file, we should be able to get the readers and montage-setting functions to set `loc` (formerly `eeg_loc`) properly.

Eric89GXL, 2015-12-16:
> (It would also be nice if there were a way to load a fif file as a montage, so I wouldn't have to do the conversion and modification. Seems like I'm the only person who has digitizer info in a fif file, though, so that may not be a priority)

I agree it would be nice to have, and I don't think it should require too many lines of code to get it to work. Can you make as small as possible a BrainVision file that could be used with a given FIF-format montage that we can use in our test suite? It would allow us to develop the FIF-montage reading and the (developed on BrainVision but probably general) `loc`-setting simultaneously.

Eric89GXL, 2015-12-16:
And actually now that I think about it, what we'll really want to do is take that BV file, put it through your working/validated MNE-C forward pipeline, and use that as a validation target for the Python code.

Eric89GXL, 2015-12-16:
Some incremental progress so far. The EEG channels from the .fif file can now be read in using #2714:
```Python
>>> import mne
>>> dirname = '/home/larsoner/Desktop/test_fwd/'
>>> raw = mne.io.Raw(dirname + 'openclosed-crop-0-None-raw.fif')
>>> montage = mne.channels.read_montage(dirname + 'eeganes07.fif', path='')
>>> print(montage)
<Montage | eeganes07 - 65 channels: EEG000, EEG001, EEG002 ...>
```
But it would be good to get a small BrainVision file as described above to continue.

One thing I'm not sure about is how to incorporate the non-EEG positions (e.g., fiducials or extra points). @emilyps14 do you have a way to do that in Python already? I'm not aware of a function we have to do it. I assume the C code does it with the `--dig` argument?

emilyps14, 2015-12-16:
According to the [hpts file format](http://martinos.org/mne/dev/manual/c_reference.html#cjadjebh), an EEG identifier of 0 always refers to the reference electrode. So it should be safe to include the reference for that type of file. Not sure about other formats.

Let me think about how difficult it would be to create a set of files for validation. I don't have a nice script that I'm using for the C pipeline, just a text file with relatively thorough notes about what Andres did, along with the processed files. Being new to the process myself, I would feel pretty bad if I created validation files that were wrong in some way.

I'll start with the brainvision file -- that shouldn't be too tough.

emilyps14, 2015-12-16:
Today I wrote up a quick function to do the fiducials and extra points using array inputs to mne.channels.read_dig_montage, but I haven't validated it at all yet. 

Which step of the pipeline depends on them? I've only noticed their absence recently when trying to simulate data from source (mne.simulation.simulate_raw).

Eric89GXL, 2015-12-16:
I guess you don't have to worry too much about replicating the C pipeline steps -- just having the output raw FIF and forward solution is probably enough there. For the Python development I think I mostly need the equivalent source BrainVision file to try to get our reading tools to the same place. Then eventually we'd want a very small BrainVision file and an associated digitization FIF to include in the testing dataset (with your permission). I guess `read_dig_montage` might be what I need to improve next to get the digitizer points going. Let me know if you track down that BV file.

One step that could depend on `dig` data is coreg. Most times you have a set of `extra` points that help with it. The raw simulator also uses it, but we can change it not to (hadn't hit that use case before) or at least throw a more useful error than whatever cryptic message you probably got :)

emilyps14, 2015-12-16:
OK this set of files has a small brainvision file, raw, and fwd:
https://drive.google.com/open?id=0B8PjVL8SDL7IMzVGX2pGbTloQTA

The MNE C commands are in the 'process' script.

The set of files is big mostly because of the bem solution.

emilyps14, 2015-12-17:
Regarding `coreg`, I skipped that step by using Andres's transformation file (in the manual it strongly recommends having the the person in charge of the study do that step, since it's so important), which he created using mne_analyze. But of course I do have access to the digitizer  data so it's no problem to include it in the raw files. 

I'm surprised though that it's required for the simulated data, over and above the transformation file. I would have thought knowing the transformation matrix would be enough.

Eric89GXL, 2015-12-17:
The simulation module uses the locations of the cardinal landmarks to optionally add some simple blink-like and ECG-like artifacts

Eric89GXL, 2015-12-17:
In any case, I think I fully understand the problem now, and know how to proceed.

@emilyps14 any chance you can get me a the original BrainVision file that was used to make the `openclosed-crop-0-None-raw.fif` file? That's what I need to continue with #2714.

emilyps14, 2015-12-17:
I can't upload it to google drive since it's too big (even when I zip it: ~4.5Gb). Other options?

Eric89GXL, 2015-12-17:
Ouch! Do you have any smaller BrainVision files that use the same montage (# of electrodes, etc.) that you could share instead? If it's easy to record one -- even if it's just noise -- for a few seconds and save it, that should work just as well in terms of verifying the forward calculation works.

Eric89GXL, 2015-12-17:
Unfortunately our current testing dataset only uses 32 channels so I can't use it...

Eric89GXL, 2015-12-17:
(yours has 64 + ref)

Eric89GXL, 2015-12-17:
Actually, thinking about it deeper I should be able to hack something together. It's still useful if you can get me a file, but it's not absolutely critical now.

emilyps14, 2015-12-17:
The one I sent yesterday uses the same montage, except that `openclosed-crop-0-None-raw.fif` had been saved after excluding bads (`['EEG024', 'EEG026', 'EEG030', 'EEG050', 'EEG061']`.

I did change the electrode names after loading it so that they would match the montage.

Eric89GXL, 2015-12-17:
Ahh somehow I missed that! Sorry for the noise. I'll see what I can do.

emilyps14, 2015-12-17:
Does that match your observation? I had several versions of the files -- it's possible I sent the wrong one again.

Eric89GXL, 2015-12-17:
Yeah there are .vhdr etc. files in there. I'll let you know if they don't work, though.

Eric89GXL, 2015-12-17:
@emilyps14 I'm curious how you get `.fif` digitizer data out as a native format -- what system are you using that produces it?

emilyps14, 2015-12-17:
It's a somewhat old dataset (~2009) so the person who collected it and the person who originally analyzed it aren't around. The data was collected originally with Polhemus FASTRAK 3D. I guess the fif file I have access to may have been converted from some other format, but I can't manage to track down an original file.

PR #2714 (2016-01-27): MRG+1: Add FIF montage reading
Closes #2700.

cc @emilyps14.
closed 2016-01-27 by agramfort