mne-tools/mne-python #2706: BUG: partially downloaded dataset

jasmainak, 2015-12-15:
I was trying to update the testing dataset but I already had a partially downloaded zip file in my mne/examples/ directory. This made the downloading hang. I tried

```py
from mne.datasets import testing
testing.data_path(force_update=True)
```

I am not sure if this happens with other datasets. And also I'm not sure if there's an easy way to replicate this because you would need to have a partially downloaded copy of the old dataset (before it was updated).

Eric89GXL, 2015-12-15:
Usually partial downloads work, but for `testing` they might not since the hashes are not the same download-to-download. So one solution is to add a parameter to our dataset downloaders that says whether it can be resumed, which would be True for all datasets except the testing one.

jasmainak, 2015-12-17:
If it cannot be resumed, then why not simply download the entire dataset again instead of giving an incorrect logging message? Do we even need a new parameter?
[image]
My screen has been stuck here for more than 2 minutes. First, it tells me (incorrectly) that there is only 1 byte to download and second, it is stuck at 100% ...

Eric89GXL, 2015-12-17:
Most datasets *can* be resumed. So in our private functions somewhere we
need to indicate that all but the testing set can be resumed, and modify
our code appropriately. I assume it would be done by adding a parameter
somewhere. That's what I meant by adding a parameter.

Github also does not report the file size correctly, so that's why it
appears to sit there. PR welcome if you can find a fix.

jasmainak, 2015-12-17:
okay, let's keep the issue open. I'll get around to it when I have a better internet connection (which is also the reason I'm complaining a lot -- don't have the luxury to wait for another failed download). For now, I'll try to get the testing data using the browser ...

Eric89GXL, 2015-12-17:
One thing we could do that would probably make it resumable is make "releases" of the testing dataset. This will require simultaneous updates of `mne-python` (to say which version we need) and `mne-testing-updates`, but I think that's okay.

Eric89GXL, 2015-12-17:
I can give it a shot now actually, if what I'm thinking works it should be pretty quick. And my internet connection is stable enough that it shouldn't be too painful.

PR #2717 (2015-12-18): MRG: Allow data size for testing dataset
@jasmainak hopefully this helps. For some reason the resuming does not work properly, though. It resumes at a reasonable place, but the hash won't match and the archive won't decompress. Not sure why it's happening. But this is probably already worth merging, since it at least allows the user to see progress properly, while only adding a minor inconvenience for PRs that require a testing dataset update.

Closes #2706.
closed 2015-12-18 by jasmainak