mne-tools/mne-python #2711: BUG?: KIT coils have units FIFF_UNIT_T, not FIFF_UNIT_T_M

Eric89GXL, 2015-12-16:
The `unit` for all channels, including `FIFFV_COIL_KIT_GRAD` channels (and `FIFFV_COIL_KIT_REF_MAG`, which makes sense), is `FIFF_UNIT_T`. Is this a typo? It seems like KIT grads should have units `FIFF_UNIT_T_M`. Not having the correct units is making the picking functions think they're magnetometers, and thus breaking Maxwell filtering, which needs to distinguish between grad and mag to exclude external components. I can make the change in the KIT code, but then it won't match what MNE-C's conversion routines give. @mshamalainen @teonlamont any ideas?

Eric89GXL, 2015-12-16:
From reading a bit about KIT sensors, they do talk about converting the data to Tesla units (not Tesla / m), so it seems correct. But the channels behave like gradiometers according to Maxwell filtering (first degree external components are noisy around zero). I'd still like to hear if you think UNIT_T is the right unit, but now I'm also curious that, if it is, should doing `pick_types(meg='grad')` also return these channels? I think it probably should, even if the units don't match other grad channels...

Eric89GXL, 2015-12-16:
Scratch that, I was reading about CTF sensors...

agramfort, 2015-12-16:
I think that KIT are like for CTF axial gradiometers; The second
gradiometers are far from the head making the units of the gradiometer very
close to a magnetometer.

@mshamalainen I learnt my lesson :)

kingjr, 2015-12-16:
> The second gradiometers

"magnetometers". 

> I learnt my lesson :)

Close..

Eric89GXL, 2015-12-16:
So it's correct for the picking functions to treat these as mags? Any other
coil types that have this "feature" that you're aware of?

agramfort, 2015-12-16:
yes as the scales for plots etc should use mag defaults.... I was puzzled
too a few months ago...

Eric89GXL, 2015-12-16:
Thanks @agramfort

closed 2015-12-16 by Eric89GXL