mne-tools/mne-python #2715: [BUG?] Write ICA to file: n_samples_ not serializable

jona-sassenhagen, 2015-12-17:
With some ICA files, when I use `ica.save(...)`, I get
```TypeError: 59947 is not JSON serializable```
where 59947 is `ica.n_samples_`.
I haven't done anything special with the files.

When I do `ica.n_samples_ = int(ica.n_samples_)`, I can save the file. The type of `ica.n_samples_` in these cases is `numpy.int64`.

Is this a bug? Should I make a PR to enforce `n_samples_` always being `int`? Or am I doing something wrong?

agramfort, 2015-12-17:
the serialization should be more robust. that's the proper fix.

dengemann, 2015-12-17:
@jona-sassenhagen are you using Python 3? Just out of curiousity. Numpy types are more strictly handled there.

jona-sassenhagen, 2015-12-17:
You've corrected a lot of my py3 specific syntax in my mne code @dengemann :) yes, always py3. I'll submit a pr once I'm at a computer.

dengemann, 2015-12-17:
You basically need to convert to python ints. I think ther'es already an example for this in the serialization code, I had fixed this already for other cases.

jona-sassenhagen, 2015-12-17:
closed for #2716

closed 2015-12-17 by jona-sassenhagen

Eric89GXL, 2015-12-17:
FYI if you put `Closes #XXX` in your PR, it will auto-close the related issue with GitHub magic so you don't have to manually close.

jona-sassenhagen, 2015-12-17:
Oh that's nice.