mne-tools/mne-python #2719: dataset updates with partially downloaded dataset

jasmainak, 2015-12-18:
As discussed in https://github.com/mne-tools/mne-python/pull/2717

If a dataset is updated and a part zip file exists from a previous version of the dataset, then the download won't resume. Partially downloaded datasets are probably not as uncommon because our datasets are pretty huge.

agramfort, 2015-12-18:
can we detect that? using hash?

Eric89GXL, 2015-12-18:
Hash the partially downloaded version?

agramfort, 2015-12-18:
>
> Hash the partially downloaded version?

euh no. that would not work. hash something? :)
or use the hash on the tmp file name?

jasmainak, 2015-12-18:
I thought for a bit about this. You could have a file which stores hashes for 1000 chunks of the data (so 0.1% of the data). First this file is downloaded (and hopefully this would be accurate because it's a small file ~53 kB as I calculated). Then you can verify hashes for these chunks. If download stops, start from the chunk where the hashes don't match. Wdyt?

agramfort, 2015-12-18:
we could fix it yes but maybe we have more urgent priorities?

Eric89GXL, 2015-12-18:
To me that additional maintenance burden (more steps for dataset update)
isn't worth the tradeoff given what I assume is the low incidence of this
problem.

We come update the download filename with each update, and delete the old
version. Then at least we wouldn't get any errant restarts.

I suspect this is a problem that has minimal impact on most people (what is
the probability others have hit it?). Mainak I know the issue has affected
you, but as an experienced dev you are both more likely to hit the issue
due to needing updates, and also better equipped to deal with it manually
each time. So my vote is to keep things simple if we can.

jasmainak, 2015-12-18:
sure, I am not pressing for this. By the way, do we have to to update the hash every time we update the dataset?

Eric89GXL, 2015-12-18:
> do we have to to update the hash every time we update the dataset?

Yeah

jasmainak, 2015-12-18:
okay cool. closing.

closed 2015-12-18 by jasmainak