mne-tools/mne-python #2720: Question: Datatypes and semantics for preload in _BaseRaw

jasmainak, 2015-12-18:
In the `RawArray` class, I noticed this line: https://github.com/mne-tools/mne-python/blob/master/mne/io/array/array.py#L47 which passes `data` as the second argument. This led me to this: https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L236.

I have two questions:
1/ Why don't we have a separate `data` argument? It took me a while to realize that `preload` could actually be an `ndarray` too -- which is different from what the public API for `preload` is (bool or str). It can confuse a developer looking at the code for the first time.
2/ What is the reason for this: https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L238 ? We do allow 'single' format. In fact, that is what we strongly recommend when saving our files: https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L1183. 
3/ Finally, here is another weird thing I discovered:
```py
from mne.datasets import sample
from mne import io

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = io.Raw(raw_fname)
print('Actual data type of data: %s' % raw[0][0].dtype)
print('Data type stored in raw object: %s' % raw.orig_format)
```
Does this mean that all the processing is done with float64 but the data is saved in float32 format? Is this by design or a bug? Why are they not the same precision? I am a bit confused ...

agramfort, 2015-12-18:
> In the RawArray class, I noticed this line: https://github.com/mne-tools/mne-python/blob/master/mne/io/array/array.py#L47 which passes data as the second argument. This led me to this: [...]

hum... that's indeed convoluted.

basically if preload is an array we should set the _data attribute and
keep preload as True (ie str or bool).

eric ok?

> 2/ What is the reason for this: https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L238 ? We do allow 'single' format. In fact, that is what we strongly recommend when saving our files: https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L1183.
>
> 3/ Finally, here is another weird thing I discovered:
>
> from mne.datasets import sample
> from mne import io
>
> data_path = sample.data_path()
> raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
> raw = io.Raw(raw_fname)
> print('Actual data type of data: %s' % raw[0][0].dtype)
> print('Data type stored in raw object: %s' % raw.orig_format)
>
> Does this mean that all the processing is done with float64 but the data is saved in float32 format? Is this by design or a bug? Why are they not the same precision? I am a bit confused ...

it's a waste of space to save as float64. Using float32 with cals is
good enough in terms of precision, but you don't want to do numerics
(linalg or ffts) with float32. Or at least errors can be noticeable.

jasmainak, 2015-12-18:
> it's a waste of space to save as float64. Using float32 with cals is good enough in terms of precision, but you don't want to do numerics (linalg or ffts) with float32. Or at least errors can be noticeable.

okay, but why don't you convert to float64 when doing these operations and then back to float32? Otherwise, you are wasting RAM. At least that's how the image processing community handles arrays AFAIK (see example: http://in.mathworks.com/help/images/ref/imfilter.html). If the input to the function is float32, it returns float32 and so on ...

jasmainak, 2015-12-18:
... and here is the other problem that happens due to the upcasting:

```py
from mne.datasets import sample
from mne import io
data_path = sample.data_path() 
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif' 
raw = io.Raw(raw_fname, preload=True)
raw._data += 1e-60 # and some other operations that the user does

# now just save and load the data back
raw.save('test.fif', overwrite=True)
raw2 = io.Raw('test.fif', preload=True)

# I should have the same data now
from numpy.testing import assert_array_equal
assert_array_equal(raw._data, raw2._data)
```

agramfort, 2015-12-18:
we cannot change everything.

the _data attribute should always be float64 or complex128 and by default
we should follow what everybody does which is to store data in single
precision.

saving as float64 if raw._data is float64 while blow up the disks

jasmainak, 2015-12-18:
but do you agree that the simple operation of file write and read should preserve the data? At the very least, the documentation should warn the user if there is an upcasting going on while reading the data. Otherwise, it may result in nasty surprises.

jasmainak, 2015-12-18:
I spent close to 2 days trying to figure what the problem was ...

agramfort, 2015-12-18:
>
> I spent close to 2 days trying to figure what the problem was ...
>
data are the same up to numerical precision. All tests of IO use relative
differences because of this.

Eric89GXL, 2015-12-18:
> 1/ Why don't we have a separate data argument? It took me a while to
realize that preload could actually be an ndarray too -- which is different
from what the public API for preload is (bool or str). It can confuse a
developer looking at the code for the first time.

IIRC it would end up being redundant. Maybe we should call it
"preload_result" or something.

Eric89GXL, 2015-12-18:
> basically if preload is an array we should set the _data attribute and
> keep preload as True (ie str or bool).
>
> eric ok?

I think that's already what happens eventually...?

My recommendation is to improve docs and variable names of the base class.
Unless you have a lot of free time and a masochist steak it's probably
better to avoid chasing the base class.

In particular I think a lot of these weird choices were motivated by the
complex set of possibilities that exist across different readers
(especially FIF) and preload options. Unless you're prepared to try to
understand it all -- it took me a day or two last time I needed to -- it's
probably better not to mess with it.

Eric89GXL, 2015-12-18:
> okay, but why don't you convert to float64 when doing these operations
and then back to float32? Otherwise, you are wasting RAM. At least that's
how the image processing community handles arrays AFAIK (see example:
http://in.mathworks.com/help/images/ref/imfilter.html). If the input to the
function is float32, it returns float32 and so on ...

You will lose some precision bouncing back and forth compared to staying in
float64 the whole time.

Eric89GXL, 2015-12-18:
> ... and here is the other problem that happens due to the upcasting:

Floating point arithmetic is not exact, and expecting it to be will lead to
pain. So this example is expected behavior in my mind.

Staying in float32 will lead to more cumulative error compared to float64
when multiple operations (filtering, ICA, SSP, maxwell filtering, ...) are
carried out.  This can negatively impact data, and it's impact should be
worse than the example you posted.

Eric89GXL, 2015-12-18:
> > I spent close to 2 days trying to figure what the problem was ...
> >
> data are the same up to numerical precision. All tests of IO use relative
> differences because of this.

It is a painful lesson to learn that floating point operations should never
be expected to be exact, but worth learning. I agree we probably could have
saved you some time with better docs, though, so feel free to mention the
upconversion somewhere.

jasmainak, 2015-12-18:
> Staying in float32 will lead to more cumulative error compared to float64 when multiple operations (filtering, ICA, SSP, maxwell filtering, ...) are carried out. This can negatively impact data, and it's impact should be worse than the example you posted.

I am not saying that you shouldn't ICA etc. in float64. But if your data is originally in float32, upcasting it creates the _false impression_ that you have more precision than you actually have. The final result should be float32, no matter how you do the intermediate steps. Significance arithmetic tells you that you simply can't do better than the precision of the data you started off with. I can think of a few fixes:

- Use raw.orig_format to downcast your data in get_data() and similar functions
- Throw an error if you try to save data in an upcasted format. raw.orig_format will tell you that

etc.

Eric89GXL, 2015-12-18:
> I am not saying that you shouldn't ICA etc. in float64. But if your data
is originally in float32, upcasting it creates the false impression that
you have more precision than you actually have. The final result should be
float32, no matter how you do the intermediate steps. Significance
arithmetic tells you that you simply can't do better than the precision of
the data you started off with.

But if you do a sequence of operations on the same data, e.g. Maxwell
filtering, filtering, then SSP computation, the cumulative error should be
lower by upcasting, operating, and downcasting, than staying in float32 the
whole time. Do you agree?

If so, one problem with get_data always giving  (at best) float32 is that
the user might want to insert some step above in between e.g. filtering and
SSP computation, and put the data back.

Eric89GXL, 2015-12-18:
I think the best way to fight this false impression is with better
documentation

agramfort, 2015-12-18:
>
> I think the best way to fight this false impression is with better
> documentation

+1

jasmainak, 2015-12-18:
> But if you do a sequence of operations on the same data, e.g. Maxwell filtering, filtering, then SSP computation, the cumulative error should be lower by upcasting, operating, and downcasting, than staying in float32 the whole time. Do you agree?

I agree.

> If so, one problem with get_data always giving (at best) float32 is that the user might want to insert some step above in between e.g. filtering and SSP computation, and put the data back.

well ... the user could save the data in between and load it back -- and this will have the same issue. In fact, this is very often the mode of operation for most users. They maxfilter the data and save a new file. Then they have another script, which loads the maxfiltered data and epochs it. Then, they save this as epochs (actually, now I'm not sure if `epochs.save` uses float32 or float64 as default to save). Then there is another script which loads this and runs some decoding. This is a very common mode of analysis. One rarely runs all the steps in one go when you have multiple subjects. The only problem is that, right now -- all this is happening silently.

> I think the best way to fight this false impression is with better documentation

I agree this is the least we should do. Maybe in the "pitfalls" section which is discussed in this issue: https://github.com/mne-tools/mne-python/issues/1495. I have tagged this issue as DOC.

Eric89GXL, 2015-12-18:
> This is a very common mode of analysis. One rarely runs all the steps in one go when you have multiple subjects. The only problem is that, right now -- all this is happening silently.

Yeah that's true. FWIW the pipelines we use at UW partially operate in this way -- not all the steps, but as many as we can. We have to do Maxwell filtering separately because we haven't transitioned to mne-python's version yet (and we'll probably still save the intermediate result because MF with movement compensation takes forever). But our filtering, resampling, and SSP computations are all done in memory. So such chaining does happen in practice.

> I agree this is the least we should do. Maybe in the "pitfalls" section which is discussed in this issue: #1495. I have tagged this issue as DOC.

I've added it to #1495, so I'm going to go ahead and close this issue.

closed 2015-12-18 by Eric89GXL

jasmainak, 2015-12-18:
Thanks, sounds like a plan