mne-tools/mne-python #2726: Can ICA components and epoch data still be viewed in browse_raw?

mattjerhart, 2015-12-19:
I see there was a viz purge and I can no longer do epochs.plot(Trellis=False). Is it still possible to do ICA and trial rejection in browse_raw? How is it done now if so?

jasmainak, 2015-12-19:
@mattjerhart you can do trial rejection in browse_raw. Just click the epoch and it will turn red. Same with ICA, just click the component.

By the way, this question should go to the MNE mailing list, not here. Github is meant mostly for bug reports or if you want to contribute :)

dengemann, 2015-12-19:
@mattjerhart if you do epochs.plot you will see markers for the epochs boundaries, if you click inside one of them it will turn red. Also take a look at the help button, it will give you a table of features and related short cuts. I'm sorry if the purge was creating any inconveniences for you. I hope I can motivate you to play 5 minutes with it. I would predict that you will not miss a single feature of the previous plot.

jaeilepp, 2015-12-20:
We deprecated the trellis param. Now epochs.plot() should do the same thing.

mattjerhart, 2015-12-21:
Thanks everyone, based on the comments, I switched to the dev version and epoch.plot() is doing what I hoped it would.

agramfort, 2015-12-21:
great !

closed 2015-12-21 by agramfort