mne-tools/mne-python #2727: average refs

jasmainak, 2015-12-19:
As discussed in https://github.com/mne-tools/mne-python/pull/2676, we are not consistent in our readers with the argument `add_eeg_ref`. We need the following fixes:

1/ All readers should include this argument.
2/ make `set_channel_types` functions smart enough to check to see if any of the channel types being changed belong to an existing projector, and raise an error.

Eric89GXL, 2015-12-19:
I realized just now we actually cannot do 1 without deprecation cycle
because it's API change. So we should settle on it before release if
possible.

2 we should also do before release. I can do it on Monday.

jasmainak, 2015-12-19:
yeah, but we may have to do some deprecations anyway. I realized that there is a lot of inconsistency across readers. We should think of some way to enforce more consistency. Maybe this could have been one way to go: https://github.com/mne-tools/mne-python/pull/1930 ? but there might be other ways to do it.

Eric89GXL, 2015-12-19:
Can you mention the other inconsistencies here?

I say we target unification/consistency (1) for 0.12. We'll want a good
round of internal use before springing the API changes on users.

jasmainak, 2015-12-20:
Pretty much all the arguments. You have `ecg`, `emg` in nicolet reader but no such argument in EGI reader. EDF reader has `stim_channel` but nicolet and EGI reader don't have that. Also, `montage` is optional in EDF reader but not in EGI/nicolet reader. `proj` is `False` in `RawFIF` but the other readers don't have the argument. I think we could try to enforce this through the `_test_raw_reader` which should test all readers ... But having too many mandatory arguments will raise the barrier to add a new reader. I'm not sure what's a good way to go.

jasmainak, 2015-12-20:
Looks like `add_eeg_ref` doesn't work for our `read_epochs` function even though it's present :\

```py
import mne
from mne import io
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
event_id, tmin, tmax = 1, -0.2, 0.5

raw = io.Raw(raw_fname)
raw.add_proj([], remove_existing=True)

events = mne.read_events(event_fname)
raw.info['bads'] = ['MEG 2443', 'EEG 053']
picks = mne.pick_types(raw.info, meg=False, eeg=True, stim=True, eog=False,
                       exclude='bads')
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    picks=picks, baseline=(None, 0), preload=True,
                    reject=None, add_eeg_ref=False)
epochs.pick_channels(['EEG 001', 'EEG 002'])
epochs.save('test.fif')
from mne import read_epochs
epochs = read_epochs('test.fif', add_eeg_ref=True, proj=True)
x = epochs._data.mean(axis=1)
print(x)
```
`x` is expected to be 0 right?

agramfort, 2015-12-20:
a common test is good.

regarding proj param we have it for fif because it's the only format that
can contain projections vectors. Also we have proj param in Epochs because
it's constructed from Raw and not read from disk. So there is some
inconsistencies but not as maybe as you point out.

I would start by writing a common test that checks for example function
signatures of all readers.

agramfort, 2015-12-20:
indeed it feels like a nasty bug... no time to look now :(

Eric89GXL, 2015-12-20:
> regarding proj param we have it for fif because it's the only format that
> can contain projections vectors.

It's the only format that contains it on disk, but we can calculate and add
projectors to all our raw types except for one (EDF I think), and then we
only support writing the result to FIF. So I don't see that as a reason to
only do it for FIF raw reading.

agramfort, 2015-12-20:
I meant on disk yes. We don't need a proj param for reading files that
don't contain proj. Unless we add them automatically on read.

What I don't like much is to do things automatically on data so that saving
back to disk does not save the same data.

maybe setting proj=True by default in read_epochs was a mistake... let's
avoid this for other readers.

Eric89GXL, 2015-12-20:
> I meant on disk yes. We don't need a proj param for reading files that
don't contain proj. Unless we add them automatically on read.

I'm not talking about the `proj` param, though, but rather the `add_eeg_ref` param. Adding an average EEG ref can happen for any Raw type, not just RawFIF.

> What I don't like much is to do things automatically on data so that saving
back to disk does not save the same data.

Right now if you read data using RawFIF and save it back to disk, it will have an avg ref added by default.

For other readers, you can read them, save them to disk and have the same thing (for most of them I think). But then if you read them (as FIF) again and save them, they'll have the EEG ref added. Thus the inconsistency...

> maybe setting proj=True by default in read_epochs was a mistake... let's
avoid this for other readers.

Keep in mind it's not just `read_epochs` that we're dealing with, but also `RawFIF` and `read_raw_fif`. I think we did this to mimic what's done in MNE-C. Any time a FIF file is read that has EEG data, it has the average ref added when processed.

I'm actually fine with dropping this altogether and deprecating the `add_eeg_ref` parameter. We can instead have a method `_BaseRaw.add_avg_ref` or so, same with `_BaseEpochs`. This to me would be the cleanest approach, since it unifies everything and eliminates a constructor arg. WDYT? What I *don't* want to do is think of reading and processing FIF files as somehow being a special or exceptional case. I think what we do for FIF we should do for all our other readers, and vice-versa.

Eric89GXL, 2015-12-20:
> For other readers, you can read them, save them to disk and have the same thing (for most of them I think). But then if you read them (as FIF) again and save them, they'll have the EEG ref added. Thus the inconsistency...

And keep in mind also that historically, *all* formats in MNE-C would have had the avg ref added, because MNE-C can only process FIF. In other words, other formats would be converted to FIF, and then as soon as the data actually get processed (e.g. in `mne_process_raw`) they would have the average ref added. So the behavior of MNE-C is effectively (eventually) to add the ref, regardless of format.

Eric89GXL, 2015-12-20:
> Looks like add_eeg_ref doesn't work for our read_epochs function even though it's present :\

It's only for some combinations of `add_eeg_ref` and `proj`. Same code as above, but do this at the end and you get zeros (up to precision / atol=1e-19 or so):
```
epochs = read_epochs('test.fif', add_eeg_ref=True, proj=False)
epochs.apply_proj()
```
So I think the problem might be that the original Epochs object had `proj=True`, so got saved with `proj=True`. Then when it's read in, the new projection operator is probably never actually being applied to the data. So we probably need to be smarter with the combination of `add_eeg_ref=True` and `proj=True`, where we need to re-project our data if there wasn't an avg ref before. Make sense?

Eric89GXL, 2015-12-20:
I can fix that `Epochs` problem tomorrow unless you are keen to work on it @jasmainak

Eric89GXL, 2015-12-20:
Removed the "EASY" tag -- I don't think anything about working with our raw readers is suitable for beginners :)

jasmainak, 2015-12-20:
yeah, it turned out to be more complicated than I initially thought ;)

agramfort, 2015-12-20:
let's avoid breaking API in a rush. Let's first fix the problem.

Also we should check in all inverse methods that there is an EEG
average reference proj that is available and that no custom ref is applied.

I would introduce a

_check_inverse_eeg_ref

function for this.

AFAIK all projections present will be applied during inverse modeling
as the make_projector is called with include_active=True.
So if an EEG avg ref is present it will be used.

Eric89GXL, 2015-12-20:
Agreed about the fixes, but we should ideally also decide what we should do
long term so that nicolet and EEGLAB readers can be finalized for this
release while avoiding immediate deprecation next cycle.

I think my vote is not to have add kwarg, and instead add a method,
eventually deprecating for FIF.

agramfort, 2015-12-21:
>
> Agreed about the fixes, but we should ideally also decide what we should do
> long term so that nicolet and EEGLAB readers can be finalized for this
> release while avoiding immediate deprecation next cycle.
>

agreed.

> I think my vote is not to have add kwarg, and instead add a method,
> eventually deprecating for FIF.

I may need to think more but to me it's fine to add the eeg avg ref proj on
read
but then we should not apply it automatically. By having it it means we can
later
use the data in the inverse model without extract step.

that would probably mean removing proj param in read_epochs and read_evoked.
Or at least set it to false by default.

Eric89GXL, 2016-04-20:
Pushing to 0.13 unless someone is motivated to do this

PR #3545 (2016-09-15): [MRG+1]: Deprecate add_eeg_ref parameters
This PR basically acts as an API proposal for #3538 and #2727. @agramfort @dengemann and others feel free to have a look.

The one API thing that's missing is making `set_eeg_reference` a method of `Raw`, `Epochs`, and `Evoked`, which I think we should do. I'm okay with not doing the bipolar or `add_reference_channels` since they are used less often, and we can link to them in `See Also`. Is that okay with you @wmvanvliet?

I still need to update all examples along the lines of the one I did so far, as well as all tests that load raw data (ouch).

Clos [... cut]
closed 2016-09-15 by agramfort