mne-tools/mne-python #2730: epochs drop log plot % unclear

agramfort, 2015-12-20:
a student made me realize that the percentages we display in plot drop log is not very understandable. I could not explain it myself :)

here is plot 

[image]

obtained with the script below. Why do I see 21% rejected but only channels responsible for less than 10% of the bad epochs?

@Eric89GXL can you make it easier to understand? How about adding to the title the number of rejected epochs (not just the %)?

```
import mne
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_raw-eve.fif'
tmin, tmax = -0.2, 0.5

# Select events to extract epochs from.
event_id = {'Auditory/Left': 1, 'Auditory/Right': 2}

#   Setup for reading the raw data
raw = mne.io.Raw(raw_fname)
events = mne.read_events(event_fname)

#   Set up pick list: EEG + STI 014 - bad channels (modify to your needs)
include = []  # or stim channels ['STI 014']
raw.info['bads'] = ['MEG 2443', 'EEG 053']  # set bads

# pick EEG and MEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=True, stim=False, eog=True,
                       include=include, exclude='bads')
# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0),
                    reject=dict(grad=4000e-13, mag=4e-12, eog=150e-6),
                    preload=True)

epochs.drop_bad_epochs()
epochs.plot_drop_log(subject='sample')
```

Eric89GXL, 2015-12-20:
That is confusing, will look tomorrow.

I hope you are prepared for a reviewing spree :)

agramfort, 2015-12-21:
>
> That is confusing, will look tomorrow.
>
> I hope you are prepared for a reviewing spree :)

always ...

Eric89GXL, 2015-12-21:
Result from #2733:

[image]

PR #2733 (2015-12-21): MRG: Fix drop log plot
Closes #2730.

@agramfort works for me, ready for review/merge. Didn't add a test because it's for a `viz` function.
closed 2015-12-21 by jasmainak