mne-tools/mne-python #2731: FIFF constant conflict

jaeilepp, 2015-12-21:
Just noticed with new addition of seeg channels to ``raw.plot`` that one of the hpi channels has a kind of 702, which was our old seeg constant:
```
In [17]: raw.info['chs'][332]
Out[17]: 
{'cal': 9.999999747378752e-05,
 'ch_name': u'CHPI002',
 'coil_type': 0,
 'coord_frame': 0,
 'kind': 702,
 'loc': array([ 1632., -2303.,   -66.,  2318., -1998.,  -926.,  2555., -1272.,
        -1621.,  2284.,  -321., -1840.]),
 'logno': 2,
 'range': 1.0,
 'scanno': 333,
 'unit': -1,
 'unit_mul': -1}
```
So ``raw.plot`` causes:
```RuntimeError: Some channels not classified, please report this problem```
Not sure what's the best way to fix this. Was the old seeg constant (702) used in the previous release already, or can we just remove the check for it in channel type checking completely?

jaeilepp, 2015-12-21:
Just checked, and it was indeed used in 0.10.

jaeilepp, 2015-12-21:
And just to clarify, the error happens with addition of seeg channel to ``raw.plot`` in https://github.com/mne-tools/mne-python/pull/2676.

agramfort, 2015-12-21:
we have a conflict on ch types between HPI and SEEG? even with new constants?

jaeilepp, 2015-12-21:
No, the new constants seem to work. We left the check for old constants so that it is compatible with the old ones too. So basically you can check ``mne.io.pick`` where the old constant is also checked. Maybe we should just remove the check there. That would mean that old files saved with the wrong constant wouldn't work anymore.

agramfort, 2015-12-21:
> No, the new constants seem to work. We left the check for old constants so
> that it is compatible with the old ones too. So basically you can check
> mne.io.pick where the old constant is also checked. [...]
or do a hack for 1 release cycle as HPI channels have std names very
different from sEEG channels.?

closed 2015-12-21 by Eric89GXL