mne-tools/mne-python #2740: bem model and source estimate

ImmanuelSamuel, 2015-12-22:
Hey guys so I was able to get sLORETA working for evoked data for now. But I have an issue with the BEM model and the source estimate.

[image]

If you see here the occipital regions dont have sources. @dgwakeman when you were working on the colin27 t1 did you see this issue?

ImmanuelSamuel, 2015-12-22:
Also is there an alternative to pysurfer for py3 for now to visualize surface maps?

agramfort, 2015-12-23:
mayavi should soon work with py3k. In the mean time the answer is no unfortunately.

dgwakeman, 2015-12-26:
I don't fully understand. While showing evoked data might be how you detect the problem, it is best to work step by step to trace it. You are trying to view data on a T1 slice. Have you looked at the alignment between the T1 and your source space surfaces? (next move on down the analysis stream checking alignments etc.

hth
d

ImmanuelSamuel, 2015-12-28:
I see that the plot above is not very informative. I was trying to show you the alignment of the colin t1 and the sources. The following should be more clear.
[image]

dgwakeman, 2015-12-28:
What does the alignment look like between the T1 and the source space surfaces ?

ImmanuelSamuel, 2015-12-31:
I am sorry, I do not know how to visualize the T1 and the source space surfaces together. Can you guide me? Thank you

agramfort, 2015-12-31:
use freeview

you'll need to write/use a surface to see.

Eric89GXL, 2016-07-18:
Closing due to non response, @ImmanuelSamuel feel free to reopen if you have other updates

ImmanuelSamuel, 2018-10-05:
Finally got back to this. I checked the alignment between the T1 and the BEM model and they are perfectly aligned in freeview.

The following the code I used for analysis:
```
# source
src = mne.setup_volume_source_space(mri='path/to/colin27.nii')
model = mne.make_bem_model('spm12_canonical_folder',subjects_dir='path/to/spm12_canonical_folder')
bem_sol = mne.make_bem_solution(model) 
fwd = mne.make_forward_solution(info, None, src, bem_sol)
cov = mne.compute_covariance(epochs, method='auto')
inv = mne.minimum_norm.make_inverse_operator(info, fwd, cov, loose=1)
stc = mne.minimum_norm.apply_inverse(evoked, inv, lambda2=1. / 9.,  method='dSPM')
volume = stc.as_volume(fwd['src'], dest='mri')
```
This is what I get:
data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAkgAAADeCAYAAADGvv/YAAAABHNCSVQICAgIfAhkiAAAAAlwSFlzAAALEgAACxIB0t1+/AAAADl0RVh0U29mdHdhcmUAbWF0cGxvdGxpYiB2ZXJzaW9uIDIuMi4zLCBodHRwOi8vbWF0cGxvdGxpYi5vcmcvIxREBQAAIABJREFUeJzsvWl0XNWVNvzUPKo0y5ZkPGGDMeC4wdgM8UsIEKbQToAO/kLSyQrpBaHpdEPIR7KSYDvNokOHTmfiByEJTdLwQoYvwSQxJkwGAgm2IcTGNth4lCzbkiWVpCqVVNP343rv2vfcK90qDZZUOs9aWqo7n3vvuefs4dl7uwDkoaGhoaGhoaGhwXBPdAM0NDQ0NDQ0NCYbtICkoaGhoaGhoaFAC0gaGhoaGhoaGgq0gKShoaGhoaGhoUALSBoaGhoaGhoaCrSApKGhoaGhoaGhQAtIGhoaGhoaGhoKvBPdgNGira0NM2fOnOhmaIwRjhw5gsbGxoluhoaGhobGNIcLUzxRZD4/pZuvYQOXyzXRTdDQ0NDQmObQLjYNDQ0NDQ0NDQVaQNLQ0NDQ0NDQUKAFJA0NDQ0NDQ0NBWUrIK1duxYul4v/mpqacP311+P9998HANO2of5eeuklvPTSS3C5XNi+ffuQ15o7dy7uuusuxza1trYiGo1i7969AIBjx45h7dq12L9//5jc844dO3DppZciHA6jqakJ99xzD7LZ7LDHvPPOO7jyyivR1NSEQCCA2bNn4/Of/zza2trGpE0S3/72t3HppZeO+Xk1NDQ0NDTGGlM+im04VFZW4plnngEA7N27F9/4xjdw6aWX4p133sHrr7/O+/X39+PDH/4wvv71r+Oaa67h9YsXL8abb745Zu259957ce2112L+/PkADAFp3bp1+NCHPoS5c+eO6txdXV247LLLsHjxYjz11FN4//338aUvfQm5XA733nvvkMfF43HMmzcP//iP/4impibs27cP69atw9atW7F582Z4vWPXRW699Vbcd999eOmll/ChD31ozM6roaGhoaExHshP5b+hsGbNmnxtba1p3SuvvJIHkP/FL35hWt/b25sHkH/kkUcs53nxxRfzAPLbtm0b8lpz5szJf+lLXxpyez6fz8fj8XwoFMo/++yzvG7btm15APkXX3xx2GOLwX333ZevqqrKx+NxXnf//ffnQ6GQaV0xePbZZ/MA8lu3bh11u1TcfPPN+euuu27YfSa6T+k//af/9J/+039l62Kzw7nnngsAY+bSKgW/+MUvEAqF8OEPf5jbcPbZZwMALrnkEnbrjRQbNmzAFVdcgVgsxutWr16N/v5+bNq0qaRz1dbWAgAGBweH3e8//uM/sGDBAgSDQcyYMQNXXnkljhw5Muwx119/PX73u9+hs7OzpDZpaGhoaEx+XHHFFdi1axd2796Nu+++27J95cqV2Lp1K9LpNK6//nrTtm9961vYtm0btm3bhk984hMnq8lDoqxdbCpIMJqIxJLPP/88li9fDo/HAwBobGzEY489hptuugkPPvggzjnnHNP+mUzG8Zwej4eFql27drHwRZg9ezbC4TB27dqFa6+9dthz5XI5ZDIZ7Nu3D1/5yldw3nnnYfny5UPu/7Of/Qz33Xcf7r//fpx55pk4fvw4XnjhBSQSiWGvc+GFFyKdTuOVV17BqlWrHO9RQ0NDYyiQQkjjoMfj4TGW/nu9XqYK5EXePPpNPM1sNotcLgfAoD9olA63240HH3wQl19+OVpaWrB582asX78eO3fu5H0OHjyIz372sxbe7tVXX41zzjkHS5cuRSAQwKZNm7Bhwwb09vae7NtglL2ARILG3r17cdttt6GiogKXXXbZSW/H1q1bTQJBIBDAkiVLABhcp/PPP9+0v8/nczznI488gs9+9rMADA5SVVWVZZ/q6mp0dXU5nuvqq6/Gxo0bARiWtj/84Q9wu4c2ML7xxhv4yEc+gttuu43XXXfddY7XqaysxOzZs/HGG29oAUlDQ2NI1NTUIBwOAygIQn6/n7e73W4WaAj5fN4xMAUwhCaaG0h4yuVy/LupqYmvIc9Hv1OpFAAgmUzi8OHDpd9cmWL58uXYs2cP9u3bBwB44oknsGrVKpOAdODAAQCwvLvFixdj06ZNyGazSCaTePvtt3HllVfil7/85cm7AQVlLSAdP37cJGjMnj0bTz755ISUsjhy5Ajq6uqK3n/z5s2O+8ybN8+0bOeiy+fzRbnufvCDH6CzsxO7d+/Gvffei6uuugp/+tOfEAwGbfdfunQpfvKTn2DNmjW45pprcO6557LG5oS6ujpHV5yGhkb5IxqNAgCCwSAqKytN64CClYcEo0AgYBKSaDuNceFwGJFIBABYwfP7/TwZ03zQ2dmJmpoaAEBPTw+vI2uFFJDoGv39/RZrRjQaRUNDA4CC0BSPx8clCngqoLm5GYcOHeLllpYWrFixoqhj3377baxZswbf+c53EA6Hcckll2DHjh3j1dSiUNYCUmVlJZ577jm4XC7MnDkTTU1NE1bGIpVKIRAIFL3/0qVLHfeRAkl1dTW6u7st+8TjcVvLkoqFCxcCAFasWIGVK1di3rx5ePzxx/G5z33Odv/Pfe5z6O3txY9+9CN885vfRG1tLb7whS9g7dq1joJSIBDgwURDQ0NDozwwlJJeDP74xz/ivPPOw2uvvYb29na8/vrrRVFNrrjiCnR0dAy7z9atW4tqg4qyFpC8Xi+WLVs20c0AYJiL7QSYoVCqi23RokXYtWuXafuhQ4eQSCSwaNGikto6Z84c1NTUcL4mO7jdbtxxxx244447cOjQITz22GP42te+hubmZtx6663Dnr+7u5u1Nw0NjemFyspKthJVVFQAMKw8NLmSm6umpoYDRkKhEADDLUMWn1QqZXJ1AYbSSFYeaTWiY+gabrcbra2tAAo0jGAwyJM5cZD8fj9bompra3HKKacAKLjaent70d7ebjpPfX09j2/UloMHD478gU0htLS08DMCgFmzZpXkgrzvvvtw3333AQAee+wx7N692/GYjo4ObNmyZdh9RmoYKWsBaTLh9NNPZ78sgUzFdtaUUl1sV111Fb797W+jt7eXB50nn3wSoVAIF198cUltfffdd3H8+HGLC28onHLKKfjKV76CRx55xNEkmsvlcPDgQZx22mkltUlDQ0NDY3Jj8+bNWLhwIebOnYvW1lasXr0an/zkJ4s61u12o6qqCp2dnTj77LOxZMkSPPvss0UcmQcwPh4JLSAViT/+8Y8WC83ixYuxePFiAM4S6kUXXYT169eb1s2ePRuhUAiPPvooKisr4fP52OJVquXr1ltvxfe//31cd911uPvuu7F3716sXbsWd955pyn0f8GCBbj44ovxk5/8BABw1113wev1YsWKFaiqqsLOnTvxn//5nzj11FOxevXqIa93yy23oKamBueffz4qKyvx4osvYvfu3bj//vt5H6/Xi3vuuQf33HMPr3v33XfR19eHiy66qKT709DQmHoIh8NMLaiuruZ1aqRZTU2NhaOZSqVw9OhRAAWeUHd3NwYGBgAAAwMDbPGREWn0myxIQ5G26dpkIXK5XLyvXVRcOBxm5ZMsWhUVFViwYIHpmGPHjjHHko6NxWKsCJMnwcktNBWRzWZx++23Y+PGjfB4PPjpT3+KHTt2YN26ddiyZQuefvppLFu2DL/5zW9QXV2Na6+9FuvWrcNZZ50Fn8+HV155BYDxvj/1qU8VRbgHctAC0gTjzjvvtKxbs2YN1q5di/7+fhNx0A7XXXcdvvWtb+HgwYOYPXs2AMOk+/DDD2PdunW4+OKLkU6ni/bXqqiursbzzz+P22+/Hddeey2qqqpwxx13YO3atab9MpmMqdMtW7YMP/jBD/CjH/0IqVQKs2fPxvXXX4+vfvWrTHa0wwUXXICHH34YDz30EFKpFBYsWICHH34YH/vYx3gfGTZLeOaZZzBv3jz83d/93YjuU0NDY/KDlLKamhoLBzIcDrMwRILG4OAgR9tKAeL48eMAgHQ6DcAQQsiVJZVSOW7SenXskZCh/+Tuy+VylohfyYFJJpOcv43G+1gsxm2jgBafz8cWchKQjh8/zq44isybOXMmu/LKKa3Ahg0bsGHDBtO6NWvW8O8tW7aY3HCEgYEBnHnmmSO44vgJSC4Y9qkpi5EKFGOFeDyO2tpaPPjgg7jllluG3Xfp0qW46aab8OUvf/kktW7y4YILLsA111yDr3/960PuM1FEeg0NjbHBSAQk4hGdLAGJhJxoNIrPfe5zyOfzeOCBB4a9LxkZR/c5a9YsAGYBie7ZTkCiBLy5XK4sBaSTjXPPXYItWzYOu4/LNbLIdW1BGiEGBgbw2muv4aGHHoLH48HVV1/teMzXvvY1fPnLX8Ydd9wxpjXOpgr+8pe/YNeuXRbtQkNDY+qDBISKigoOffd4PGyhqa+vB2BYZeLxOABzaDyFh/f19QEwLNCqiyWfz/PYmclkLGH+QEEwkoKSqnTJY6PRKObNm4d8Ps/HyvPK33QecvMdO3aMrUpEzJ45cyaTs+UzoXB3EpD27dvHglZVVRULUMXkrdOQGD8O0rQqNTKWaGtrw5VXXoldu3bh6aeftjUZqrjhhhvwhS98gaMnphs6Ozvx6KOPFpV2QEND4+Rg3759uPTSSye6GRolYt++fUgmk+jt7UVbWxseeeSRYWkR5YscgAGHv5Fh+pkxxghz585lLaJYu [... cut ...] 82a0tbXJ/1tbWyXFw4HATGyGYRiGYRxwIpEI1q1bhwsuuABbtmzBypUrcfXVV3vSUOxPzMRmGIZhGMYBp1Kp4Oabb8ZLL72ESCSCxx577IAJR4BpkAzDMAzDMHyYD5JhGIZhGIaDCUiGYRiGYRgOJiAZhmEYhmE4mIBkGIZhGIbhYAKSYRiGYRiGgwlIhmEYhmEYDiYgGYZhGIZhOJiAZBiGYRiG4WACkmEYhmEYhoMJSIZhGIZhGA4mIBmGYRiGYTiYgGQYhmEYhuFgApJhGIZhGIaDCUiGYRiGYRgOJiAZhmEYhmE4mIBkGIZhGIbhYAKSYRiGYRiGgwlIhmEYhmEYDiYgGYZhGIZhOJiAZBiGYRiG4WACkmEYhmEYhoMJSIZhGIZhGA4mIBmGYRiGYTiYgGQYhmEYhuFgApJhGIZhGIaDCUiGYRiGYRgOJiAZhmEYhmE4mIBkGIZhGIbh8P8B/NOWRTIZQWIAAAAASUVORK5CYII=

ImmanuelSamuel, 2018-10-05:
Finally got back to this. I checked the alignment between the T1 and the BEM model and they are perfectly aligned in freeview.

The following the code I used for analysis:
```
# source
src = mne.setup_volume_source_space(mri='path/to/colin27.nii')
model = mne.make_bem_model('spm12_canonical_folder',subjects_dir='path/to/spm12_canonical_folder')
bem_sol = mne.make_bem_solution(model) 
fwd = mne.make_forward_solution(info, None, src, bem_sol)
cov = mne.compute_covariance(epochs, method='auto')
inv = mne.minimum_norm.make_inverse_operator(info, fwd, cov, loose=1)
stc = mne.minimum_norm.apply_inverse(evoked, inv, lambda2=1. / 9.,  method='dSPM')
volume = stc.as_volume(fwd['src'], dest='mri')
```
This is what I get:
[image]

Instead of the last line if I use ...
```
volume = stc.as_volume(fwd['src'], dest='surf')
```
... I get the following
[image]

agramfort, 2018-10-05:
mri should point to the T1.mgz file from your freesurfer reconstruction. Could it the pb?

ImmanuelSamuel, 2018-10-05:
Yes in this case T1.mgz is the freesurfer reconstruction of the colin27 image. Also that and the original are identical right? They both will look the same when loaded in freeview.

What is pb?

agramfort, 2018-10-06:
I don't know what the problem is

can you try passing the T1.mgz ?

can show me what you get with plot_alignment function?

ImmanuelSamuel, 2018-10-06:
[image]

ImmanuelSamuel, 2018-10-08:
The freesurfer subject directory is too large for github. But I did try the plot_alignment and I get this 
[image]

ImmanuelSamuel, 2018-10-08:
OK so the problem is the alignment as you suspected.

mne.viz.plot_alignment shows:
[image]

However freesurfer can properly align it
[image]

ImmanuelSamuel, 2018-10-09:
Since this (using the spm canonical surfaces) had all these issues. I just went ahead and manually edited the BEM surfaces to match my T1. Now the surfaces are very much aligned
[image]

The sources are also much better. But there still seems to be some alignment issues especially in the anterior and posterior sources.
[image]

agramfort, 2018-10-09:
you should not have to align manually anything if you rely on a freesurfer
recon (unless we have a but which is possible).

can you share data via dropbox or google drive if it's too big for github ?

ImmanuelSamuel, 2018-10-09:
Sometimes the meshes are not correct with brain surfaces that are outside the skull. In those cases some surface tweaking is necessary. Will try to get the data to you soon.

ImmanuelSamuel, 2018-10-11:
@agramfort were you able to look into this?

agramfort, 2018-10-11:
did you share anything with me? I don't know how we can replicate the pb

ImmanuelSamuel, 2018-10-11:
Yes, in the last post I had posted the link with the data.

agramfort, 2018-10-13:
I have your files. thanks. Can you share a script working from the files so
I can easily replicate and investigate?

larsoner, 2019-08-20:
I'm pretty sure this was due to not properly including the surface information in the BEM files, which has been fixed. Feel free to reopen if I'm wrong

closed 2019-08-20 by larsoner