mne-tools/mne-python #2742: GFP in evoked.plot does not play along with ylim kwarg

jona-sassenhagen, 2015-12-23:
The GFP yaxis is set based on the auto limits, but if a user defines y lims themselves with the ylim kwarg, the GFP can float or disappear.

@jaeilepp 

[image]
[image]

jasmainak, 2015-12-23:
do you have a quick fix?

Eric89GXL, 2015-12-23:
I think it should be pretty to fix -- check for ylim kwarg and use that as bounds instead of the `get_ylim()` bounds instead, hopefully.

jona-sassenhagen, 2015-12-23:
Or this code
```
            if ylim is not None and t in ylim:
                if plot_type == 'butterfly':
                    ax.set_ylim(ylim[t])
                elif plot_type == 'image':
                    im.set_clim(ylim[t])
```
in viz evoked lines 350ff could be split up and put in the parts of the code where BF and image plots are dealt with, and the one for BF before the GFP part.
I think.

I can try later, unless Jakko wants to do it on the quick, but I can't promise I'll get to do it before xmas.

closed 2015-12-23 by Eric89GXL