mne-tools/mne-python #2744: CUFFT library

kambysese, 2015-12-23:
I am running current dev master on OS X 10.11.2 with CUDA toolkit 7.5, pycuda, and scikits-cuda, `mne.cuda.init_cuda()` returns `OSError: cufft library not found`. Never encountered this one, and it appears that the CUDA toolkit installation should've included this lib. Any seasoned MACOS users have any ideas how to solve this?

Eric89GXL, 2015-12-24:
Is that on init of `scikits.cuda`? If so, you could try pinging them, too. Please tag me if you do because I'd like to follow.

kambysese, 2015-12-24:
I don't see any mention of "cufft" in the file `scikits.cuda.__init__.py` Is that what you meant?

Eric89GXL, 2015-12-24:
Nah, I meant where are you getting the error in mne-python? It's possible we're catching it and the traceback isn't so useful, in which case we should improve the error message to say which library import fails (e.g., a `from scikits.cuda import fft` call or something).

Can you try this on your system and see if you get that error?
```
from skcuda import fft
```
or if you're running and older version,
```
from scikits.cuda import fft
```
If it is indeed a problem with scikits-cuda as I suspect, you will probably need to ask [over there] (https://github.com/lebedov/scikit-cuda).

Eric89GXL, 2016-02-03:
Closing as an issue with other lib. Did this ever end up working?

closed 2016-02-03 by Eric89GXL

kambysese, 2016-02-03:
Nope.