mne-tools/mne-python #2747: pop_prop equivalent (sensor or component summary)

jona-sassenhagen, 2015-12-24:
One EEGALB function I miss and find myself reproducing manually a lot is pop_prop.
These give a summary of a sensor or component that is helpful especially for identifying artefact types.
(They also work with raw files, segmenting the raw stream into 2-second epochs and showing you were gross artefacts are located in the data.)

[image]
[image]

Maybe if others also find these interesting, we could implement them.

E.g. @jaeilepp

jona-sassenhagen, 2015-12-24:
For channels rather than ICs, the sensor position is highlighted instead of showing the mixing weights.

jona-sassenhagen, 2016-04-24:
If nobody is interested in this, I'll close.

agramfort, 2016-04-25:
at least make a gist of your code so we know where to start to implement this.

jona-sassenhagen, 2016-06-27:
closed by #3275

mmagnuski, 2016-06-27:
Although the logic of OR (in the issue title) is fulfilled I think you can leave this open until channel properties are ready. :)

jona-sassenhagen, 2019-04-26:
@agramfort I would like to also allow this for channels (like EEGLAB has). It's good to quickly see e.g. what kind of artefacts a channel has (just 2 or 3, or more). Ok?