mne-tools/mne-python #2748: import mne error

ahibahi, 2015-12-24:
Hi 
I've been working using mne for past few months and am having an issue importing mne and the co libraries from yesterday.
I am not sure if it's mne issue or a common one
Please help me out here
[image]

Eric89GXL, 2015-12-24:
What version of Python are you running?

Eric89GXL, 2015-12-24:
It's weird that the problem would just have started because that code is a couple of years old (and mostly unchanged) I think. Did you upgrade packages or Python interpreter around the time things stopped working?

ahibahi, 2015-12-24:
Anaconda 2.4.0
I have not done any changes recently.
I tried upgrading and re installing MNE but still no luck

jasmainak, 2015-12-24:
are you able to import other libraries? `import numpy` or `import scipy` -- do they work?

agramfort, 2015-12-24:
I suspect your config file is corrupted.

Try removing it.

It's in $HOME/.mne/mne-python.json

ahibahi, 2015-12-24:
@jasmainak  yes I am able to install other libraries 
@agramfort  I removed json file and now am getting this error
[image]

I'll reinstall python and check

agramfort, 2015-12-24:
remove everything and run again a "pip install mne"

ahibahi, 2015-12-24:
Already tried
no luck

agramfort, 2015-12-24:
Do you install with pip?

ahibahi, 2015-12-24:
Yes 
I reinstalled python too
still getting the same error

agramfort, 2015-12-26:
in ipython do:

import mne
mne

it will point you where mne is installed. Remove the folder and reinstall.

if it breaks make sure you did not name a file with a name similar to an
mne file.

HTH

Eric89GXL, 2016-02-03:
Closing due to inactivity, @ahibahi feel free to reopen if you are still having trouble or have new insights

andrewheusser, 2016-06-15:
@ahibahi did you ever resolve this?  i have the exact same error and i've been stuck on it for weeks!

andrewheusser, 2016-06-15:
@Eric89GXL @agramfort.  I'm also getting this error.  I followed @agramfort's advice removing the config file: $HOME/.mne/mne-python.json.  The package loaded successfully, but seems to depend on that file.  Any ideas for how to regenerate it or replace it?

Eric89GXL, 2016-06-15:
We could try to be more careful when reading the file, and warn if it's corrupted here:

https://github.com/mne-tools/mne-python/blob/master/mne/utils.py#L1258

agramfort, 2016-06-15:
+1 for being more careful and avoid a crash

https://github.com/mne-tools/mne-python/blob/master/mne/utils.py#L1258

andrewheusser, 2016-06-15:
is the $HOME/.mne/mne-python.json file supposed to be generated on install? I uninstalled mne and reinstalled but I don't see the config file

Eric89GXL, 2016-06-16:
It should only be generated when saving a config value to disk, I think.

PR #3312 (2016-06-16): MRG: Safer config reading
@andrewheusser can you see if this fixes the issue for you?

Closes #2748.
closed 2016-06-16 by agramfort