mne-tools/mne-python #2753: plot_evoked_topo and callbacks

jjnurminen, 2015-12-28:
I'm trying to use mne.viz.plot_evoked_topo, but I'm having some trouble with the callback that displays a single channel. If I run my script from the Spyder IDE, it works perfectly. However if I run it from command line (or command line IPython), the single-channel plot is not opened on a mouse click.

I tried adding plt.show() to the end of _plot_topo_onpick(). If I do this, I get figures on click. So this seems to be related to the default behavior of matplotlib being different in the two contexts.

Should the callback call plt.show() always or does it have some undesirable side effects?

jasmainak, 2015-12-28:
@jjnurminen it's generally not recommended to hack the code yourself unless you have git installed and can keep track of the changes :)

It seems to me that you didn't start ipython in interactive mode. Did you try `ipython --matplotlib=qt` ? You may also try to use `plt.ion()` in the beginning of your script. That will also make your plot interactive.

jasmainak, 2015-12-28:
> Should the callback call plt.show() always or does it have some undesirable side effects?

Regarding this specific question, if this solves your problem maybe we need to make this fix in our code so that our plotting routines work regardless of the backend. But I'm not entirely aware of the issues involved. Maybe @jaeilepp or @dengemann can answer this question better.

Eric89GXL, 2015-12-28:
`plt.show` has blocking behavior in non-interactive mode, so I don't think we always want to do it. Maybe you just need `plt.draw()`? That can be added and only has the drawback of reduced performance in some cases -- but if a redraw is really supposed to occur at that time, then it probably *should* be called.

jjnurminen, 2015-12-29:
@jasmainak Adding plt.ion() does not work - it causes the main topo figure window to close immediately after it has been drawn. The --matplotlib=qt option has the same effect. Not sure why.

@Eric89GXL Adding a call to plt.draw() has no effect - the window will not open without a call to plt.show().

agramfort, 2015-12-29:
jussi start ipython with:

$ipython --matplotlib

then run the script you want with

[1] %run plot_my_script.py

this will make it run in interactive mode as within spyder.

jjnurminen, 2015-12-29:
@agramfort I'm writing a script for someone else (a medical doctor :), so it should be as simple to run as possible. Ideally it should run from the command line by just invoking the Python interpreter.

jasmainak, 2015-12-29:
@jjnurminen this is a hack but it should do the job for you:

```py
plt.ion()
plot_evoked_topo(evokeds)
import numpy as np
plt.pause(np.inf)
```
I think the viz folks can propose a more permanent solution ...

agramfort, 2015-12-29:
you can use something like :

ipython plot_read_evoked.py -i --matplotlib

it will keep the shell open when script is done. So you keep figures open.

HTH

agramfort, 2016-01-05:
@jjnurminen pb solved?

jjnurminen, 2016-01-09:
@agramfort  will check asap, have been busy with other things. Thanks.

jjnurminen, 2016-01-11:
@agramfort Your solution works ok, so I'm closing this. Thanks.

closed 2016-01-11 by jjnurminen