mne-tools/mne-python #2756: Q: LogisticRegressionCV Cs with GAT

kingjr, 2015-12-30:
@agramfort Using the `LogisticRegressionCV` to optimize the C parameters in the gat leads to quite variable C values; Do you have any idea why, and / or whether it's normal?

@jona-sassenhagen Do you generally `gridsearch` for the C value when you use the LR?

For an example:
```
import numpy as np
import mne
from mne.datasets import spm_face
from mne.decoding import GeneralizationAcrossTime
from sklearn.preprocessing import RobustScaler
from sklearn.pipeline import make_pipeline
from sklearn.linear_model import LogisticRegression

# Preprocess data
data_path = spm_face.data_path()
raw_fname = data_path + '/MEG/spm/SPM_CTF_MEG_example_faces%d_3D_raw.fif'
raw = mne.io.Raw(raw_fname % 1, preload=True)  # Take first run
picks = mne.pick_types(raw.info, meg=True, exclude='bads')
raw.filter(1, 30, method='iir')
events = mne.find_events(raw, stim_channel='UPPT001')
event_id = {"faces": 1, "scrambled": 2}
tmin, tmax = -0.1, 0.5
decim = 4  # decimate to make the example faster to run
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    picks=picks, baseline=None, preload=True,
                    reject=dict(mag=1.5e-12), decim=decim, verbose=False)

# Define decoding params
pipe = make_pipeline(RobustScaler(),
                     LogisticRegression(Cs=np.logspace(-3, 3, 5)))
gat = GeneralizationAcrossTime(clf=pipe, n_jobs=-1)

# /!\ a bit slow
gat.fit(epochs)

print([clf.steps[1][1].C_[0] for clf_ in gat.estimators_ for clf in clf_])

```

agramfort, 2015-12-31:
@kingjr well I am not surprise. Did you look at CV curves? are they flat
ie. having many values of C with roughly the same performance? does
estimating the C change a lot your results?

kingjr, 2016-01-03:
The values are not very different indeed (~+/- 4%). Changing the scoring function does reduce the variance in the selected C. Thanks!

closed 2016-01-03 by kingjr