mne-tools/mne-python #2757: pb website image

agramfort, 2015-12-31:
@jaeilepp the following image is wrong

http://martinos.org/mne/dev/auto_examples/plot_compute_mne_inverse.html#sphx-glr-auto-examples-plot-compute-mne-inverse-py

results should highlight auditory cortex. I works on my machine.

is your sample data broken?

jaeilepp, 2016-01-02:
Not anymore. The example draws it right when I run it on my computer. I have no idea how that wrong image got there. Now I tried updating sphinx and I'm having trouble building the docs. It just freezes at some point without any errors. I'll see if I can fix it tomorrow.

agramfort, 2016-01-02:
thanks

while you're at it can you fix the rendering of the author lists. See:

http://martinos.org/mne/dev/whats_new.html

the list with number of commits

thanks ** 2

Eric89GXL, 2016-02-03:
#2831 reports the following as broken:

http://martinos.org/mne/dev/auto_examples/plot_compute_mne_inverse.html#sphx-glr-auto-examples-plot-compute-mne-inverse-py

http://martinos.org/mne/stable/auto_examples/inverse/plot_label_source_activations.html#sphx-glr-auto-examples-inverse-plot-label-source-activations-py

http://martinos.org/mne/dev/auto_examples/visualization/plot_meg_eeg_fields_3d.html#sphx-glr-auto-examples-visualization-plot-meg-eeg-fields-3d-py

http://martinos.org/mne/dev/auto_examples/inverse/plot_make_inverse_operator.html#sphx-glr-auto-examples-inverse-plot-make-inverse-operator-py

closed 2016-03-07 by agramfort