mne-tools/mne-python #2759: ENH: Drop old numpy, Python support

Eric89GXL, 2015-12-31:
It looks like the scipy folks are moving toward dropping Python 2.6, 3.2, 3.3 and Numpy < 1.7.1 support. I vote we do the same. Thoughts?

agramfort, 2016-01-01:
@dengemann would it create a problem at neurospin?

@yarikoptic what would be the consequence on debian packaging?

dengemann, 2016-01-01:
Not sure, honestly. I can check next week. What is with jenkins at inria? Is it still on python 2.6?

agramfort, 2016-01-01:
yes. It's running an old ubuntu stable.

yarikoptic, 2016-01-02:
I just will not provide recent mne for those older releases of Debian and Ubuntu

Eric89GXL, 2016-01-02:
Looks like Ubuntu 14.04 and Debian stable (Jessie) are on 1.8.x, and 12.04 and oldstable (Wheezy) are on 1.6.x.

jona-sassenhagen, 2016-01-02:
At least Mac OSX El Capitan has Python 2.7 now.

agramfort, 2016-01-02:
ok thanks

@Eric89GXL I would say that next time we need to backport something we'll drop the support. Presently it works so let's keep it like this.

Eric89GXL, 2016-01-03:
Sounds good to me, I'll close for now then

closed 2016-01-03 by Eric89GXL