mne-tools/mne-python #2760: SSP with filtering + picks

jasmainak, 2016-01-01:
Discussion continued from https://github.com/mne-tools/mne-python/pull/2710/ with @choldgraf and @Eric89GXL 

Filtering a subset of channels and then applying SSP will make the filtered channels mix with the unfiltered channels. Here are a few possible solutions:

- Throw an error when you try to apply SSP when only a subset of channels have been filtered
- Throw an error
- Truncate the proj

According to @Eric89GXL:

> * There are legitimate use cases for this. Imagine that exactly one EEG channel had really bad line noise. You'd want to filter the line noise (only from that channel is fine) before applying average ref. 
> * Note that projectors are often calculated in a specific frequency band of data, too -- look at our defaults for ECG and EOG, each one filters before calculating proj. So often the filtering of raw data will differ from the filtering used to calculate projectors. The filtering during proj is used to better isolate the proper contaminating spatial component, but should clean the unfiltered data fine.
> * Projecting with fewer than the original channels used to determine the basis vector for the projector, as was done in the PSD functions, should be avoided if possible. Right now we don't have great safeguards against it but we have recently discussed adding them.

Also, it's not super clear what happens when you filter first then proj or proj then filter (with and without picks). According to @Eric89GXL :

> When are linear operations operating *on the same data*, so you can swap order without issue. But if you apply a linear (temporal) operator to an *incomplete* subset of (spatial) channels, the result will not be the same with the order switched, because the operators are not both applied to the same data.

My understanding is that these are subtle issues and we have to have clear safeguards to avoid the users making mistakes.

ps: @Eric89GXL I've never calculated proj on filtered data and applied it on unfiltered data. I'm not sure you want to do that ... but do we already do that under the hood somewhere?

jasmainak, 2016-01-01:
There are also these weird cases where you might filter say 5 channels, then apply proj, then filter the rest of the channels ... and as a naive user, you might expect the result to be the same as when you filter all channels and then apply proj.

agramfort, 2016-01-01:
to me filtering only some channels was allowed to speed up computation.
Filtering one channel only as it's more noisy is a bit weird/dangerous to
me.

my feeling is that maybe we should remove the picks param in filtering and
replace it by a ch_type param to filter only let's say grad or mag or eeg.
As proj apply to ch types independently that would make it safe.

what do you think?

dengemann, 2016-01-01:
> my feeling is that maybe we should remove the picks param in filtering and
replace it by a ch_type param to filter only let's say grad or mag or eeg.

+1 we don't need complex APIs where there is anyways little gain to expect.

jasmainak, 2016-01-01:
> to me filtering only some channels was allowed to speed up computation. Filtering one channel only as it's more noisy is a bit weird/dangerous to me. 

I agree with this ... the same filter should be applied to all channels. Actually, does the current API allow you to filter multiple times with different passbands for different channels?

> my feeling is that maybe we should remove the picks param in filtering and replace it by a ch_type param to filter only let's say grad or mag or eeg. As proj apply to ch types independently that would make it safe. what do you think?

yes agreed.

Eric89GXL, 2016-01-01:
>> to me filtering only some channels was allowed to speed up computation.
Filtering one channel only as it's more noisy is a bit weird/dangerous to
me.

You'd have to be careful with applying it and knowing the consequences. It
could violate some typical assumptions about the data (to varying degrees
depending on the content of the signals), which I assume is what you mean
about danger. I can imagine cases where this tradeoff might be more
appealing than the alternatives (leave chanel noisy, mark it bad, or filter
all channels), but given the concerns, this is perhaps too small a corner
case to drive the API.

> I agree with this ... the same filter should be applied to all channels.
Actually, does the current API allow you to filter multiple times with
different passbands for different channels?

Yes, you can do it in principle. Note that this will also be true if we
give a ch_type argument as proposed below (MEG in one band, EEG another).

>> my feeling is that maybe we should remove the picks param in filtering
and replace it by a ch_type param to filter only let's say grad or mag or
eeg. As proj apply to ch types independently that would make it safe. what
do you think?
>
> yes agreed.

We can try it and see if it breaks anyone's code. If it does we can revert,
or really the workaround of operating on the data attribute manually isn't
that bad. So I can live with it.

...but then again, why do we want to offer a channel type argument instead
of forcing filtering of all data channels to be done the same way? If we
are worried about linear operators staying linear, then processing channel
types differently can still violate this assumption for other operators
(e.g., grad processed differently than mag before SSS, inverse operator,
etc.). It's not all that different from how using picks can violate
assumptions, except that the proj linear operator is safe at least with
this change.

In practice I'm not really worried about ch_type even though in principle
is very similar to picks -- we can't prevent all possible errors, and I
imagine errors related to this will hopefully be pretty rare, and probably
rarer than with picks. The expected legitimate use cases (e.g. sensor space
analyses or data cleaning) are probably worth having the ch_type (or picks)
argument. But hopefully we can also improve documentation to help users and
devs think about these issues a bit. We might also mention in the docstring
that we recommend to filter all data channels that will be used together
later in the same way to be safe.

agramfort, 2016-01-01:
I agree with eric.

@jasmainak let's debrief IRL on monday

jasmainak, 2016-01-02:
debriefing IRL sounds good.

Eric89GXL, 2016-02-03:
So I think the conclusion from this was that we should use a `ch_type` instead of `picks` param. Maybe `ch_types` would actually be better because then it could be a list, e.g. `['meg', 'eeg']`. That sounds doable.

The somewhat related yet unresolved issue of doing a subselection of channels when doing projection reared its ugly head in #2834.

I propose we make it so that you can only apply projectors if all channels from the given projection are available. In real use cases, users should set bad channels *before* computing and applying projectors. The one use case I expect this to screw up is the acquisition projectors. Such a change would force people to remove projectors added during acquisition if they have a single bad channel. But maybe this annoyance is worth the gained safety of not having these partial projection operators.

Eric89GXL, 2016-02-03:
If this is too drastic a change, we could instead detect the decrease in projection vector magnitude due to exclusion of channels, and throw a warning or error when it goes to < 0.95 or some other reasonable threshold (since our vectors are normalized to 1).

Eric89GXL, 2016-02-03:
For example, the failing #2834 had original projection vector magnitudes of `[1.00000000756, 1.0000000536, 0.999999953912]` (excluding nothing) go to `[0.0577592503054, 0.0117516470802, 0.252452924576]` by the picks it made using `picks = picks[2:9:3]`.

jasmainak, 2016-02-03:
> I propose we make it so that you can only apply projectors if all channels from the given projection are available. In real use cases, users should set bad channels before computing and applying [...]

Maybe we could add a method `add_bad_channels` that does it automatically?

Eric89GXL, 2016-02-03:
We already have `raw.add_bad_channels`, we could have it remove existing projectors. I guess we need to decide if we want an error or a warning, though. I'll start with a warning and see how many tests it makes angry.

Eric89GXL, 2016-02-03:
or `raw.load_bad_channels` rather.

Eric89GXL, 2016-09-08:
I think this has approximately been taken care of with warnings. I think we can iterate on it now with more targeted issues as we hit them. I'll close but @jasmainak feel free to reopen if there is a clear proposal or issue we should still tackle.

closed 2016-09-08 by Eric89GXL