mne-tools/mne-python #2768: 0.11 test_other_systems test failure

yarikoptic, 2016-01-05:
while building the package locally on my dirty debian testing/unstable system:
```
======================================================================
FAIL: Test Maxwell filtering on KIT, BTI, and CTF files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.11+dfsg/mne/preprocessing/tests/test_maxwell.py", line 125, in test_other_systems
    _assert_n_free(raw_sss, 28, 32)  # bad origin == brutal reg
  File "/home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.11+dfsg/mne/preprocessing/tests/test_maxwell.py", line 95, in _assert_n_free
    'nfree fail: %s <= %s <= %s' % (lower, n_free, upper))
AssertionError: nfree fail: 28 <= 33 <= 32
```

Eric89GXL, 2016-01-05:
What versions of Python, scipy, and numpy are you running?

yarikoptic, 2016-01-05:
python 2.7.10
numpy 1:1.9.2-5 (disregard 1: -- debian epoch)
scipy 1:1.9.2-5

Eric89GXL, 2016-01-06:
Well that one has me puzzled. In practice it's probably not a big deal that it differs, but in principle I'm bothered by it because it means the computations come out differently on your system than on Travis and others. I'm not sure what would cause this difference, though. I'll mark it for 0.12 so I don't forget to look into it. Is this time sensitive?

yarikoptic, 2016-01-06:
not really, but while I can (if I still can) reproduce I could provide more information.  Since it built fine in clean environments, it doesn't hold me so not otherwise time sensitive.

yarikoptic, 2016-01-06:
not sure if of any help but "badly conditioned" made me share it -- that log_file caught in the test
```
(Pdb) p log_file
'Maxwell filtering raw data\n    No bad MEG channels\n    Processing 0 gradiometers and 157 magnetometers (of which 157 are actually KIT gradiometers)\n(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin\n    Automatic origin fit: head of radius 90.5 mm\n    Using origin 156.6, -28.4, 102.4 mm in the head frame\n        Using 80/80 inside and 12/15 outside harmonic components\n(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin\nMatrix is badly conditioned: 519567056 >= 1000\n'
```
may be has any hits

Eric89GXL, 2016-01-06:
I think that comes from a different call to the function unfortunately, because it's for the no-regularization version, and your error arises on the regularized version

yarikoptic, 2016-01-06:
"my" failure is from within test_other_systems, here is more context:
```
$> nosetests -s -v --pdb /home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.11+dfsg/mne/preprocessing/tests/test_maxwell.py 
Test Maxwell filtering on KIT, BTI, and CTF files ... Extracting SQD Parameters from /home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.11+dfsg/mne/preprocessing/tests/../../io/kit/tests/data/test.sqd...
Creating Raw.info structure...
Setting channel info structure...
Creating Info structure...
Ready.
Maxwell filtering raw data
    No bad MEG channels
    Processing 0 gradiometers and 160 magnetometers (of which 157 are actually KIT gradiometers)
(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin
    Automatic origin fit: head of radius 90.5 mm
    Using origin 156.6, -28.4, 102.4 mm in the head frame
Maxwell filtering raw data
    No bad MEG channels
    Processing 0 gradiometers and 157 magnetometers (of which 157 are actually KIT gradiometers)
    Using origin 0.0, 0.0, 40.0 mm in the head frame
    Computing regularization
        Resulting information: 270.9 bits/sample (98.2% of peak 275.8)
        Using 65/80 inside and 12/15 outside harmonic components
    Processing data in chunks of 2.0 sec
[done]
> /usr/lib/python2.7/unittest/case.py(422)assertTrue()
-> raise self.failureException(msg)
(Pdb) up
> /home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.11+dfsg/mne/preprocessing/tests/test_maxwell.py(95)_assert_n_free()
-> 'nfree fail: %s <= %s <= %s' % (lower, n_free, upper))
(Pdb) up
> /home/yoh/deb/gits/pkg-exppsy/build-area/python-mne-0.11+dfsg/mne/preprocessing/tests/test_maxwell.py(125)test_other_systems()
-> _assert_n_free(raw_sss, 28, 32)  # bad origin == brutal reg
(Pdb) p log_file
'Maxwell filtering raw data\n    No bad MEG channels\n    Processing 0 gradiometers and 157 magnetometers (of which 157 are actually KIT gradiometers)\n(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin\n    Automatic origin fit: head of radius 90.5 mm\n    Using origin 156.6, -28.4, 102.4 mm in the head frame\n        Using 80/80 inside and 12/15 outside harmonic components\n(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin\nMatrix is badly conditioned: 519567056 >= 1000\n'
```

Eric89GXL, 2016-01-06:
Yeah that makes sense -- unfortunately the `log_file` contents are not terribly useful -- what you have there is from the `assert_raises` call, not the call that starts `raw_sss = maxwell_filter(raw_kit, origin='auto', ...`, because that one uses `verbose='warning'`, but the `AssertionError` is raised by looking at some properties of the `raw_sss` object that gets created.

You could try doing `verbose='debug'` instead of `verbose='warning'` on that call and it might be a little bit more informative, but I suspect it might just tell us that `Using 33/80 inside and 12/15 outside harmonic components`, which would make sense from the `AssertionError` we already get. But maybe there will be something useful in there.

yarikoptic, 2016-01-06:
```
(Pdb) p log_file
'Maxwell filtering raw data\n    No bad MEG channels\n    Processing 0 gradiometers and 157 magnetometers (of which 157 are actually KIT gradiometers)\n(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin\n    Automatic origin fit: head of radius 90.5 mm\n    Using origin 156.6, -28.4, 102.4 mm in the head frame\n        Using 80/80 inside and 12/15 outside harmonic components\nMaxwell filtering raw data\n    No bad MEG channels\n    Processing 0 gradiometers and 157 magnetometers (of which 157 are actually KIT gradiometers)\n(X, Y) fit (156.6, -28.4) more than 20 mm from head frame origin\n    Automatic origin fit: head of radius 90.5 mm\n    Using origin 156.6, -28.4, 102.4 mm in the head frame\n    Computing regularization\n            Condition 8.813/0.000 = 13754439876104.8, Removing in component 41: l=6, m=+0\n            Condition 8.756/0.000 = 7423736647042.5, Removing in component 29: l=5, m=+0\n            Condition 8.699/0.000 = 4685458864407.9, Removing in component 47: l=6, m=+6\n            Condition 8.641/0.000 = 4594402166877.5, Removing in component 26: l=5, m=-3\n            Condition 8.583/0.000 = 3584316399136.6, Removing in component 36: l=6, m=-5\n            Condition 8.524/0.000 = 3283908860921.4, Removing in component 32: l=5, m=+3\n            Condition 8.466/0.000 = 2681365028525.9, Removing in component 27: l=5, m=-2\n            Condition 8.406/0.000 = 2511093949620.4, Removing in component 24: l=5, m=-5\n            Condition 8.347/0.000 = 2109742225262.3, Removing in component 33: l=5, m=+4\n            Condition 8.287/0.000 = 1995562641620.9, Removing in component 43: l=6, m=+2\n            Condition 8.226/0.000 = 1959077590124.7, Removing in component 60: l=7, m=+5\n            Condition 8.165/0.000 = 735996300176.8, Removing in component 61: l=7, m=+6\n            Condition 8.104/0.000 = 508426939755.1, Removing in component 19: l=4, m=+0\n            Condition 8.042/0.000 = 489514932583.3, Removing in component 40: l=6, m=-1\n            Condition 7.979/0.000 = 422579349699.9, Removing in component 31: l=5, m=+2\n            Condition 7.916/0.000 = 413979305335.1, Removing in component 73: l=8, m=+2\n            Condition 7.853/0.000 = 226187110205.9, Removing in component 13: l=3, m=+2\n            Condition 7.789/0.000 = 223574519755.2, Removing in component 39: l=6, m=-2\n            Condition 7.725/0.000 = 213876493433.0, Removing in component 23: l=4, m=+4\n            Condition 7.724/0.000 = 179503521023.7, Removing in component 46: l=6, m=+5\n            Condition 7.659/0.000 = 110638256550.9, Removing in component 59: l=7, m=+4\n            Condition 7.594/0.000 = 106230634083.4, Removing in component 49: l=7, m=-6\n            Condition 7.528/0.000 = 90189310753.2, Removing in component 37: l=6, m=-4\n            Condition 7.461/0.000 = 55423511154.6, Removing in component 35: l=6, m=-6\n            Condition 7.394/0.000 = 51328371730.1, Removing in component 11: l=3, m=+0\n            Condition 7.326/0.000 = 50816845892.3, Removing in component 55: l=7, m=+0\n            Condition 7.257/0.000 = 34592731586.9, Removing in component 57: l=7, m=+2\n            Condition 7.188/0.000 = 34137348125.6, Removing in component 56: l=7, m=+1\n            Condition 7.118/0.000 = 28821990058.4, Removing in component 5: l=2, m=+0\n            Condition 7.048/0.000 = 28535980661.8, Removing in component 71: l=8, m=+0\n            Condition 6.977/0.000 = 24169414000.2, Removing in component 25: l=5, m=-4\n            Condition 6.905/0.000 = 15767265412.0, Removing in component 16: l=4, m=-3\n            Condition 6.832/0.000 = 9133120892.0, Removing in component 8: l=3, m=-3\n            Condition 6.759/0.000 = 8880302600.5, Removing in component 34: l=5, m=+5\n            Condition 6.685/0.000 = 8533011111.6, Removing in component 17: l=4, m=-2\n            Condition 6.610/0.000 = 8345021381.6, Removing in component 54: l=7, m=-1\n            Condition 6.533/0.000 = 7898851549.6, Removing in component 1: l=1, m=+0\n            Condition 6.457/0.000 = 7223149169.5, Removing in component 20: l=4, m=+1\n            Condition 6.379/0.000 = 5298261200.2, Removing in component 44: l=6, m=+3\n            Condition 6.300/0.000 = 5060517578.4, Removing in component 15: l=4, m=-4\n            Condition 6.221/0.000 = 4733116663.4, Removing in component 7: l=2, m=+2\n            Condition 6.141/0.000 = 3839983866.4, Removing in component 28: l=5, m=-1\n            Condition 6.059/0.000 = 3783125575.9, Removing in component 74: l=8, m=+3\n            Condition 5.976/0.000 = 2283187646.5, Removing in component 12: l=3, m=+1\n            Condition 5.892/0.000 = 1322239042.2, Removing in component 2: l=1, m=+1\n            Condition 5.833/0.000 = 1308434374.1, Removing in component 22: l=4, m=+3\n            Condition 5.747/0.000 = 1230031428.6, Removing in component 70: l=8, m=-1\n        Resulting information: 219.8 bits/sample (100.0% of peak 219.8)\n        Using 33/80 inside and 12/15 outside harmonic components\n    Decomposition matrix condition: 519567056.0\nMatrix is badly conditioned: 519567056 >= 1000\n    Processing data in chunks of 2.0 sec\n[done]\n'
```

Eric89GXL, 2016-01-06:
Interesting, for the head origin fit you get:
```
```
And I get different x,y coord and slightly different radius:
```
(X, Y) fit (155.3, -27.9) more than 20 mm from head frame origin
    Automatic origin fit: head of radius 89.2 mm
```
I've run into this before, slight differences in fits due to different versions of `scipy`. I wonder if they fixed a minor bug somewhere that changes it, or if maybe we need to fix some parameters.

For now I might just skip the origin fit altogether. @agramfort I assume it's worth backporting a fix to 0.11?

agramfort, 2016-01-06:
+1 for the backport

Eric89GXL, 2016-01-08:
Pushed a commit to `master` and `maint/0.11`:

https://github.com/mne-tools/mne-python/commit/a58b1470e2a04f227ddd3c764dc5d3452313a0d7

closed 2016-01-08 by Eric89GXL

yarikoptic, 2016-01-08:
;-)