mne-tools/mne-python #2771: checksum of downloaded dataset doesn't match somehow expected one

yarikoptic, 2016-01-06:
```
$> python -m duecredit examples/connectivity/plot_mne_inverse_psi_visual.py

=====================================================================
Compute Phase Slope Index (PSI) in source space for a visual stimulus
=====================================================================

This example demonstrates how the Phase Slope Index (PSI) [1] can be computed
in source space based on single trial dSPM source estimates. In addition,
the example shows advanced usage of the connectivity estimation routines
by first extracting a label time course for each epoch and then combining
the label time course with the single trial source estimates to compute the
connectivity.

The result clearly shows how the activity in the visual label precedes more
widespread activity (a postivive PSI means the label time course is leading).

References
----------
[1] Nolte et al. "Robustly Estimating the Flow Direction of Information in
Complex Physical Systems", Physical Review Letters, vol. 100, no. 23,
pp. 1-4, Jun. 2008.

Checking for sample data in "~/mne_data"...
Trying to create '~/mne_data' in home directory
Downloading or reinstalling data archive MNE-sample-data-processed.tar.gz at location /home/yoh/mne_data
Downloading data from https://s3.amazonaws.com/mne-python/datasets/MNE-sample-data-processed.tar.gz (1.35 GB)

[........................................] 100.00000 \ downloading   
Verifying download hash.
Error while fetching file https://s3.amazonaws.com/mne-python/datasets/MNE-sample-data-processed.tar.gz. Dataset fetching aborted.
Error: Hash mismatch for downloaded file /home/yoh/mne_data/MNE-sample-data-processed.tar.gz.part, expected f73186795af820428e5e8e779ce5bfcf but got ccf5cbc41a3727ed02821330a07abb13
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/lib/python2.7/dist-packages/duecredit/__main__.py", line 94, in <module>
    main()
  File "/usr/lib/python2.7/dist-packages/duecredit/__main__.py", line 85, in main
    runctx(code, globs, globs)
  File "/usr/lib/python2.7/dist-packages/duecredit/__main__.py", line 34, in runctx
    exec(cmd, globals, locals)
  File "examples/connectivity/plot_mne_inverse_psi_visual.py", line 37, in <module>
    data_path = sample.data_path()
  File "<string>", line 2, in data_path
  File "/usr/lib/python2.7/dist-packages/mne/utils.py", line 551, in verbose
    return function(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/mne/datasets/sample/sample.py", line 22, in data_path
    download=download)
  File "/usr/lib/python2.7/dist-packages/mne/datasets/utils.py", line 247, in _data_path
    hash_=hash_)
  File "<string>", line 2, in _fetch_file
  File "/usr/lib/python2.7/dist-packages/mne/utils.py", line 551, in verbose
    return function(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/mne/utils.py", line 1465, in _fetch_file
    % (temp_file_name, hash_, md5))
RuntimeError: Hash mismatch for downloaded file /home/yoh/mne_data/MNE-sample-data-processed.tar.gz.part, expected f73186795af820428e5e8e779ce5bfcf but got ccf5cbc41a3727ed02821330a07abb13

DueCredit Report:
- MEG and EEG data analysis with MNE-Python., MNE software for processing MEG and EEG data. / mne (v 0.10) [1, 2]
- Scientific tools library / numpy (v 1.9.2) [3]
- Scientific tools library / scipy (v 0.16.1) [4]

3 packages cited
0 modules cited
0 functions cited

References
----------

[1] Gramfort, A. et al., 2014. MNE software for processing MEG and EEG data. NeuroImage, 86, pp.446–460.
[2] Gramfort, A., 2013. MEG and EEG data analysis with MNE-Python. Frontiers in Neuroscience, 7.
[3] Van Der Walt, S., Colbert, S.C. & Varoquaux, G., 2011. The NumPy array: a structure for efficient numerical computation. Computing in Science & Engineering, 13(2), pp.22–30.
[4] Jones, E. et al., 2001. SciPy: Open source scientific tools for Python.
```

I have double-verified by independent download that checksum of that file is as not-expected ;):

```
$> wget https://s3.amazonaws.com/mne-python/datasets/MNE-sample-data-processed.tar.gz 
--2016-01-05 21:11:46--  https://s3.amazonaws.com/mne-python/datasets/MNE-sample-data-processed.tar.gz
Resolving s3.amazonaws.com (s3.amazonaws.com)... 54.231.114.164
Connecting to s3.amazonaws.com (s3.amazonaws.com)|54.231.114.164|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1450850396 (1.4G) [application/x-gzip]
Saving to: ‘MNE-sample-data-processed.tar.gz’

MNE-sample-data-processed.tar.gz           100%[========================================================================================>]   1.35G  73.7MB/s   in 22s    

2016-01-05 21:12:08 (62.0 MB/s) - ‘MNE-sample-data-processed.tar.gz’ saved [1450850396/1450850396]
$> md5sum MNE-sample-data-processed.tar.gz 
ccf5cbc41a3727ed02821330a07abb13  MNE-sample-data-processed.tar.gz
```

Eric89GXL, 2016-01-06:
I suspect you're on an older release of mne-python. `master` currently looks for the proper checksum (`ccf5cbc41a3727ed02821330a07abb13`):

https://github.com/mne-tools/mne-python/blob/master/mne/datasets/utils.py#L198

I think `f73186795af820428e5e8e779ce5bfcf` is an older iteration of the sample dataset. @agramfort any idea what the best solution is here? Do we need to do some versioning, or maybe patch back to older versions of mne-python to look for the newer version of the sample dataset?

yarikoptic, 2016-01-06:
now I don't exactly know what has happened since 0.10.1 is the only version which I had installed but duecredit above says 0.10 (but might have come from previous run)... anyways -- if you have "allowance" to enable versioning for that bucket just do it and then you could store versionId and request the "expected" one

agramfort, 2016-01-06:
> I think f73186795af820428e5e8e779ce5bfcf is an older iteration of the
> sample dataset. @agramfort <https://github.com/agramfort> any idea what
> the best solution is here? Do we need to do some [...]

crap... yarik are you getting this error with latest release?

eric we should then start to have all versions of the sample data and put
it in file name eg:

MNE-....-v0.10.tar.gz

the pain is that there is one more thing to do to release ...

wdyt?

yarik you can fix the issue by putting manually the folder MNE-sample-data
in the mne/examples

yarikoptic, 2016-01-06:
heh -- rechecked with clean installation and just a copy of the example script without running it from within git repo (possibly at older point?) -- works fine!  so sorry for the noise.  and yeap -- that is how I fixed it locally ... FWIW visualization there didn't work due to some obscure crashes from tvtk from mayavi2 I have installed (4.4.3-2)

yarikoptic, 2016-01-06:
and on another system (mayavi2 4.3.1-3.1, matplotlib  1.4.2-3.1) with
```
Traceback (most recent call last):
  File "plot_mne_inverse_psi_visual.py", line 112, in <module>
    clim=dict(kind='percent', pos_lims=(95, 97.5, 100)))
TypeError: plot() got an unexpected keyword argument 'clim'
```

closed 2016-01-06 by yarikoptic

Eric89GXL, 2016-01-06:
That suggests that the example you're running is from a more recent version of `mne` than the one you have installed. We added the `clim` argument to `stc.plot()` a couple releases ago.

yarikoptic, 2016-01-06:
damn -- sorry guys... I have installed python-mne package and by default it pulled the one available in debian not neurodebian and it was older 0.8.6... sorry about the noise. with 0.11 it ran fine and just spit me those nasty  upon exit which I guess we could ignore (so just FYI):

```
colormap: fmin=-2.33e-01 fmid=0.00e+00 fmax=2.33e-01 transparent=0
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
PyAssertionError: C++ assertion "GetEventHandler() == this" failed at ../src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
wx._core.PyAssertionError: C++ assertion "GetEventHandler() == this" failed at ../src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed
```

Eric89GXL, 2016-01-06:
I also have problems with Mayavi2 on Ubuntu. I've seen that before, but have no idea how to fix it.