mne-tools/mne-python #2772: EHN: coregistration instance without file

kingjr, 2016-01-06:
When using the coregistration, the `inst` param can only be a file. It'd be great if it accepted actual instances too (@christianbrodbeck ?)

e.g.
```
from mne.gui import coregistration
coregistration(inst=raw)
```

christianbrodbeck, 2016-01-07:
Hm, I think that would require some redesign of the model because the relation from filename to object is built into the traits model currently

kingjr, 2016-10-21:
Any foreseeable progress on this?

Eric89GXL, 2017-02-16:
In theory this should be doable, all you need is a good `info['dig']`. So we could make it so that the trait is actually `Instance(Info)` or better eventually `Instance(Digitization)`, and you'd pass `inst.info['dig']` to it. I'll mark it as part of the digitization stuff and hopefully tackle it soon.

larsoner, 2018-04-04:
Copied this over to #3934

closed 2018-04-04 by larsoner