mne-tools/mne-python #2773: BUG: Info object not correctly read from HDF5

wmvanvliet, 2016-01-07:
When reading from HDF5, the `Info` object is not properly reconstructed:

    import mne
    import tempfile
    tmp = tempfile.NamedTemporaryFile(delete=False)
    info_a = mne.create_info(ch_names=['a', 'b', 'c'], sfreq=1000., ch_types=None)
    mne.externals.h5io.write_hdf5(tmp.name, info_a, overwrite=True)
    info_b = mne.externals.h5io.read_hdf5(tmp.name)
    info_b._check_consistency()  # Error
    assert type(info_a) == type(info_b)  # Fail

The `Info` object gets reconstructed as a regular `dict`, which means any code relying on the `_check_consistency` function to work will fail.

Eric89GXL, 2016-01-07:
This is expected behavior. `h5io` isn't designed to read specialized (e.g., mne-python) objects. I'd make the downstream functions smart enough to do the casting as `Info(...)`.

Eric89GXL, 2016-01-07:
In other words, you should be able to do `info_b = Info(info_b)` in your code to fix it

Eric89GXL, 2016-01-07:
(or something like that)

wmvanvliet, 2016-01-07:
also fine.

agramfort, 2016-01-07:
please close then

closed 2016-01-08 by wmvanvliet