mne-tools/mne-python #2775: error installing mne

mmagnuski, 2016-01-08:
I just tried running `pip install -U mne` which after many screens of text culminated in the following error:
```
(...)
    byte-compiling c:\Miniconda3\Lib\site-packages\mne\viz\__init__.py to __init
__.cpython-34.pyc
    byte-compiling c:\Miniconda3\Lib\site-packages\mne\__init__.py to __init__.c
python-34.pyc
    running install_egg_info
    running egg_info
    running build_src
    build_src
    writing mne.egg-info\PKG-INFO
    writing top-level names to mne.egg-info\top_level.txt
    writing dependency_links to mne.egg-info\dependency_links.txt
    warning: manifest_maker: standard file '-c' not found

    reading manifest file 'mne.egg-info\SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no previously-included files matching '*' found under directory 'ex
amples\MNE-sample-data'
    warning: no previously-included files matching '*' found under directory 'ex
amples\MNE-testing-data'
    warning: no previously-included files matching '*' found under directory 'ex
amples\MNE-spm-face'
    warning: no previously-included files matching '*' found under directory 'ex
amples\MNE-somato-data'
    writing manifest file 'mne.egg-info\SOURCES.txt'
    Copying mne.egg-info to c:\Miniconda3\Lib\site-packages\mne-0.11.0-py3.4.egg
-info
    running install_scripts
    copying build\scripts.win-amd64-3.4\mne -> c:\Miniconda3\Scripts
    running install_clib
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Ola\AppData\Local\Temp\pip-build-ctedd7ip\mne\setup.py", li
ne 119, in <module>
        scripts=['bin/mne'])
      File "c:\Miniconda3\lib\site-packages\numpy\distutils\core.py", line 169,
in setup
        return old_setup(**new_attr)
      File "c:\Miniconda3\lib\distutils\core.py", line 148, in setup
        dist.run_commands()
      File "c:\Miniconda3\lib\distutils\dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "c:\Miniconda3\lib\distutils\dist.py", line 974, in run_command
        cmd_obj.run()
      File "c:\Miniconda3\lib\site-packages\numpy\distutils\command\install.py",
 line 62, in run
        r = self.setuptools_run()
      File "c:\Miniconda3\lib\site-packages\numpy\distutils\command\install.py",
 line 36, in setuptools_run
        return distutils_install.run(self)
      File "c:\Miniconda3\lib\distutils\command\install.py", line 551, in run
        self.run_command(cmd_name)
      File "c:\Miniconda3\lib\distutils\cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "c:\Miniconda3\lib\distutils\dist.py", line 974, in run_command
        cmd_obj.run()
      File "c:\Miniconda3\lib\site-packages\numpy\distutils\command\install_clib
.py", line 26, in run
        compiler = new_compiler(compiler=None)
      File "c:\Miniconda3\lib\site-packages\numpy\distutils\ccompiler.py", line
596, in new_compiler
        compiler = klass(None, dry_run, force)
      File "c:\Miniconda3\lib\site-packages\numpy\distutils\mingw32ccompiler.py"
, line 58, in __init__
        dry_run, force)
      File "c:\Miniconda3\lib\distutils\cygwinccompiler.py", line 126, in __init
__
        if self.ld_version >= "2.10.90":
    TypeError: unorderable types: NoneType() >= str()

    ----------------------------------------
  Rolling back uninstall of mne
←[31mCommand "c:\Miniconda3\python.exe -c "import setuptools, tokenize;__file__=
'C:\\Users\\Ola\\AppData\\Local\\Temp\\pip-build-ctedd7ip\\mne\\setup.py';exec(c
ompile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), _
_file__, 'exec'))" install --record C:\Users\Ola\AppData\Local\Temp\pip-v_d34207
-record\install-record.txt --single-version-externally-managed --compile" failed
 with error code 1 in C:\Users\Ola\AppData\Local\Temp\pip-build-ctedd7ip\mne←[0m
```

All my other packages are installed with conda (including numpy 1.10.1). My python is 3.4.3 and I'm on windows. I would be thankful for your guidance.

agramfort, 2016-01-08:
not sure problem comes from mne.

can you install any other package with pip on your system?

mmagnuski, 2016-01-08:
Ok, I got it from another package too. `pip` is downloading numpy, scipy and other packages I have already conda-installed and then screams about lapack, atlas, openblas etc. being unavailable. Conda is just so much easier :tired_face:

mmagnuski, 2016-01-08:
BTW - is it possible to install mne through conda? You don't seem to be available on the official channel.

closed 2016-01-08 by mmagnuski

agramfort, 2016-01-08:
why the hell is it downloading numpy? it's not written in the setup.py as a
requirement exactly for this reason.

our user should use pip.

mmagnuski, 2016-01-08:
Sorry, I was referring to errors I got from the other package. In the case of mne - I am not sure what is happenning, the errors suggest it is related to `numpy\distutils` and some compilation step, but it doesn't tell me much.

mmagnuski, 2016-01-09:
I discovered I had another python distribution I was not aware about (that python was lurking in the shadows) - after uninstalling, mne installed smoothly. But it may have also been because previously I was trying to `pip` install` from git bash (I'm on windows), although I doubt it could affect installation.
All in all - sorry for the problem, and especially sorry for nagging about pip. And thanks for mne, great package!