mne-tools/mne-python #2778: `add_response_shift` conflicts with `event_id` in `read_raw_brainvision`

teonlamont, 2016-01-08:
This was brought up on the listserv by Boris and is being hammered out now. One problem is that there is a conflict in `mne.io.read_raw_brainvision` when `add_response_shift` is not `None` (default is 0) and `event_id` is a `dict`. 

We should set a constraint that add_response_shift must be `None` when event_id is a `dict` otherwise there will be an error.

agramfort, 2016-01-09:
sorry but I don't see add_response_shift in the code base with git grep.

can you clarify the issue?

teonlamont, 2016-01-09:
sorry, it's `response_trig_shift`, just edited above. what happens is that `response_trig_shift` looks `Response` type of marker message. It is expecting an integer to shift a given amount. If the user `Response` message is an annotation (str), there will be a conflict.

agramfort, 2016-01-11:
what do you propose?

bburle, 2016-01-11:
To follow on Teon's comment, it is actually inconsistent to use both `event_id` and `response_trig_shift`: `response_trig_shift` is useful to disambiguate Stimulus and Response events that would have the same integer code (e.g. 'S 1' and 'R 1'). The use of  `event_id` removes any ambiguity. So, as Teon proposed, one should prevent the simultaneous use of both `event_id` and `response_trig_shift`, maybe, indeed by forcing  `response_trig_shift` to be `None` when `event_id` is defined. Issuing a warning would be useful so that the user is informed of the inconsistence. 
If this choice is made, the documentation should be updated accordingly. 
(hey, that's my first contribution to github ;-)

agramfort, 2016-01-11:
+1 for forcing response_trig_shift to None if event_id is defined. It
should raise a ValueError

bburle, 2016-01-11:
I can try to do that (although probably not today...), if you agree to check a beginner's first steps ;-)

teonlamont, 2016-01-11:
+1 would be great to help out with your first contrib. afterwards we send a follow up email to the listserv for archive purpose on how the problem was addressed

jona-sassenhagen, 2016-01-11:
I don't understand how it must be None. Couldn't it be that you have Rint and Sint type triggers for which you want to apply response shift, but also break_recording and fmri_onset triggers?

bburle, 2016-01-11:
My feeling is that  if one uses `event_id` for some events, it is better to use it for all. In case you have `break_recording` or `fmri_onset`, given them a explicit code would make the link explicit. In your example, one could have:
`event_id = {'S 1': 1,'R 1': 11, 'break_recording': 100,'fmri_onset': 200}`
In such case, the use of `response_trig_shift` is useless, and actually replaced by the explicit mapping in `event_id`.

jona-sassenhagen, 2016-01-11:
>My feeling is that if one uses event_id for some events, it is better to use it for all

My typical scenario is: I have 100 or 200 or so regular triggers, and 4 or so for which I need `event_id`. This was actually the scenario for which the functionality was conceived ...

I personally never use response_shift, but I don't think we should assume we'll never have a situation where somebody uses a nontrivial number of S and R triggers in the fMRI or such things.

jona-sassenhagen, 2016-01-11:
In the eeglab trigger reading PR, I do it like this:

```
for t, latency in zip(types, latencies):
    event_code = event_id[t] if t in event_id else event_id_func(t)
    events.append([int(latency), 1, event_code])
```

Wouldn't that be a rather straight-forward solution?

teonlamont, 2016-01-11:
> I don't understand how response shift must be None if event_id is given. Couldn't it be that you have Rint and Sint type triggers for which you want to apply response shift, but also break_recording and fmri_onset triggers?

the problem arises with marker events like this:
```
Mk16=Response,Left_Resp,3180,1,0
```
vs.
```
Mk10=Response,R255,6000,1,0
```

You're right, it could possibly be a great imposition if you have multiple triggers to have to specify them in the `event_id`. Are you often faced with both at the same time? The source of the issue is how `response_trig_shift` is applied, it assumes that any marker with `Response` will be Rint. One possible fix would be something like this: 
```python
try:
    trigger = int(string[1:])
   trigger = trigger + response_trig_shift
except ValueError:
    pass
```

jona-sassenhagen, 2016-01-11:
What about my suggestion above? E.g. something like

```
triggers_ = [event_id[t_] if t_ in event_id
             else t_[1:] + (response_trig_shit if t_[0] == "R" else 0)
             for t_ in triggers]
```

Or with `try` to catch ValueErrors and raise a warning that an event will be dropped.

bburle, 2016-01-11:
Let's try to summarize the situation: If all triggers are `Sint` or `Rint`, extracting `int` and adding `response_trig_shit` if the triggers's type is `Response` works well. For abitrary triggers names (*e.g.*, `LeftResp`), one needs `event_id` to translate such names into integers. In case of `event_id` and no `response_trig_shit`, this works well. The problem arises when triggers are arbitrary (as above) **and** `response_trig_shit` is not `None`. In this case, one tries to add `response_trig_shit` to a `None` object and the code fails. As it stands, even if one defines an `event_id`, setting `response_trig_shit` to something else than `None` will fail with arbitrary names, since the evaluation of `event_id` occurs after adding `response_trig_shit`. One simple solution would be to evaluate `event_id' before attempting to apply `response_trig_shit`. This is pretty similar to the proposition of @jona-sassenhagen, but the current code is a bit more complex than simply evaluating the first letter of the event `description`. I think code like that works (on line 200 of brainvision.py):
`        
if mdesc in event_id:
            trigger = event_id[mdesc]
        else:
            try:
                trigger = int(re.findall('[A-Za-z]*\s*?(\d+)',mdesc)[0])
            except IndexError:
                trigger = None

        if mtype.lower().startswith('response'):
            if response_trig_shift is not None:
                trigger += response_trig_shift
            else:
                trigger = None
#        if mdesc in event_id:
#            trigger = event_id[mdesc]
`

jona-sassenhagen, 2016-01-11:
@bburle , note your spelling error :) (I do that one all the time.)

To have your code appear as code on github, wrap it in three backticks on start and end.

Something like this?
```
if mdesc in event_id:
    trigger = event_id[mdesc]
else:
    try:
        trigger = int(re.findall('[A-Za-z]\s?(\d+)',mdesc)[0])
    except IndexError:
        trigger = None
    if mtype.lower().startswith('response'):
        if response_trig_shift is not None:
            trigger += response_trig_shift
    else:
        trigger = None
```

I think that should work. It could be coded a bit more pythonic, but I think it would cover the expectable usage cases in a comfortable manner.

I think we should consider raising warnings or possibly errors for dropped events though. Possibly, a high priority warning like "the following events were dropped: ..., use event_id to include them".

teonlamont, 2016-01-12:
ok, seems like we are all on board and understand what's going on.

+1 warning and +.5 noting events dropped (some might be dropped intentionally).

agramfort, 2016-01-16:
just need a PR now :)

jona-sassenhagen, 2016-02-09:
@bburle are you still on this?

bburle, 2016-02-10:
I've been overloaded those weeks, and will still be for a couple of weeks more. i'll take care of that early march is that's not too late...

Sorry for that,

teonbrooks, 2016-04-06:
@bburle, are you able to get around to solving this?

Eric89GXL, 2016-04-20:
@jona-sassenhagen do you have time to attempt a fix in the next week? If not I'll push to 0.13

jona-sassenhagen, 2016-04-20:
I can (and should) attempt a fix (unless @bburle says he can do it), but we have deadlines upcoming, so no promises ... Please move to .13. I will PR later.

Eric89GXL, 2016-04-20:
Will do, if you get it in for 0.12 great, if not, no problem from my end

jona-sassenhagen, 2016-04-20:
Can you assign me Eric? (Or @teonbrooks )? :)

Eric89GXL, 2016-04-20:
Sure, I have to add you to the org, accept the subscription (you should have an email?) and I can assign it to you

Eric89GXL, 2016-04-20:
@jona-sassenhagen done

PR #3162 (2016-04-27): MRG Make event_id compatible with response_trig_shift in BV event parser
~~WIP, probably needs a test.~~
Long term, could possibly be refactored to share more code with EEGLAB.

@bburle, @teonbrooks

closes #2778
commit 786de5fb (2016-04-27): Make event_id compatible with response_trig_shift in BV parser

WIP, probably needs a test.
Long term, could possibly be refactored to share more code with EEGLAB.

@bburle, @teonbrooks

closes #2778
closed 2016-04-27 by Eric89GXL