mne-tools/mne-python #2782: BUG..?: linear_regression_raw doesn't warn for rank deficiency

jona-sassenhagen, 2016-01-11:
Passing predictors that will lead to a rank deficient matrix to linear_regression_raw should throw an error or at least a big fat warning. Currently, it silently computes a stupid solution.
This is in my experience a *very* common scenario.

An extremely easy way may be checking the rank of the resulting predictor matrix.

(It would also be nice to estimate the degree of multicollinearity in the model.)

agramfort, 2016-01-11:
how would you do this?

jona-sassenhagen, 2016-01-11:
I'm not sure - the problem is, how to do it on the cheap? `np.matrix_rank` uses an SVD which I assume takes about as long as the actual regression.

Another option I could explore might be doing it with the much smaller input matrices, but that would only approximate it.

agramfort, 2016-01-11:
you can solve the system with an SVD so you know before solving the system
if you have a conditioning pb.

jona-sassenhagen, 2016-01-11:
I'll start trying things out once I've brought a few of my other PRs home ... I'd also extend the docstring a bit.

jona-sassenhagen, 2017-03-28:
As discussed with @Eric89GXL , I will catch the linalg error the Cholesky solver gives for a perfectly collinear pair of predictors and replace it with a more helpful warning telling people to fix their model.

Eric89GXL, 2017-04-18:
@jona-sassenhagen did you fix this in some recent PR?

jona-sassenhagen, 2017-04-18:
Nope. We discussed it at the sprint a bit, but for now I'd wait for https://github.com/mne-tools/mne-python/pull/3563

NadipilliAditya, 2025-04-07:
Hi! I’m Aditya, a student preparing for GSoC 2025 and exploring open-source contributions. This issue looks interesting, and I’d love to work on it as my first contribution. Could you please assign it to me or let me know if it’s available? Thanks!

larsoner, 2025-04-08:
It's still available!