mne-tools/mne-python #2789: Python bug when use of gat.predict

romquentin, 2016-01-14:
Hello, 
I am using the function gat.predict for decoding analyses with MEG data. When I run the function with n_jobs=-1, my python crashes. But it works with n_jobs=1. Attached the screenshot of the bug and the problem details. I am running under OS X El Capitan (10.11.2) and python 2.7. Thanks for your help! 

[error python details.txt](https://github.com/mne-tools/mne-python/files/91022/error.python.details.txt)

[image]

kingjr, 2016-01-15:
@dengeman, @alexgramfort

I tried to debug it with him, but it seems to come from the parallel.
There's no python error thrown. Not sure what to do from there...

dengemann, 2016-01-15:
I think this is a problem with joblib and numpy. What kind of numpy setup
do you have? And JR, what was your impression, After our vectorization to
me any benefit of parallel processing was lost. Not even when running
n_jobs = number of physical cores. I have Anaconda with MKL optimization,
and this makes sense, our new code will trigger more intelligent
multiprocessing at a lower level. Maybe we need to poll a bit for whom it
actually works / and for whom it brings any benefit.

agramfort, 2016-01-15:
use anaconda with mkl on mac os. Ask for the academic license

it's related to a bug in the apple accelerate framework.

kingjr, 2016-01-15:
> I think this is a problem with joblib and numpy. What kind of numpy setup do you have? And JR, what was your impression, After our vectorization to me any benefit of parallel processing was lost. [...]

I haven't checked, I generally don't use MKL as I had not installed it on my aws image... Is it worth it IYO?

In any case we'll need to keep joblib 1) for those who don't have it 2) for `window_size > 1` as we don't vectorize then.

@romquentin Let us know how it goes with anaconda & mkl.

romquentin, 2016-01-15:
Thanks. It works with anaconda & mkl ! The numpy version was 1.10.1 and
turn to 1.10.2 when I installed mkl.

kingjr, 2016-01-15:
Great, you may close the issue.

closed 2016-01-15 by romquentin