mne-tools/mne-python #2792: EHN: raw decimate?

kingjr, 2016-01-15:
I often play with data on my small laptop, and fancy the need to have a decimation in the raw object (ideally at loading time) to alleviate my memory. Is there any reason why we shouldn't have this? Could it then create issues with the events, or the subsequent epoching?

jona-sassenhagen, 2016-01-15:
I was just running into this yesterday on the train. My poor Macbook crashed because I was trying to load in *two* files at once. So +1

Eric89GXL, 2016-01-15:
Resampling or decimating raw before epoching creates sub-sample jitter, so it's better to decimate after creating epochs. (There was a long discussion of this issue about a year ago I can dig up if you're really curious.) There is already the `raw.resample()` method, but you're right it won't save you memory because it requires preloading. However, just decimating (taking every Nth sample) without doing some sort of filtering causes aliasing.

All that being said, if this is a really common use case, we can probably figure out a way to resample single channels at a time without preloading all the data at once. That way only the result needs to be stored in memory. It will be pretty brutally slow, though, I expect, since in some cases you'd essentually end up reading the entire file `n_channels` times (!) to do this.

Why do you need to load the raw data with `preload=True`? Can you use `preload=False` for your use case? Have you tried using memmapping?

kingjr, 2016-01-15:
Thanks for the quick reply. If it's slow (e.g. by using resample), then forget about it, as the main aim was intended for fast checkup. My use case was read raw > extract events > low-pass filter & bad chan detection > linear_regression_raw, but I guess I can just save the fitlered raw object, and use a `preload=False` with `decimate=n` in `linear_regression_raw`.

For future reference, the topic was mentioned here https://github.com/mne-tools/mne-python/issues/954

closed 2016-01-15 by kingjr

Eric89GXL, 2016-01-15:
FWIW resampling might not be slow forever: #2722, #2035. And if you're using CUDA it's almost instantaneous (for me at least).

Eric89GXL, 2016-01-15:
...but I guess the re-reading would probably still make it slow.

jona-sassenhagen, 2016-01-15:
> My use case was read raw > extract events > low-pass filter & bad chan detection > linear_regression_raw

Note you can do most of that stuff on the evokeds. If you interpolate or reject a channel anyways, it doesn't matter if you do it pre or post regression. And you can filter evokeds.
> if you're using CUDA it's almost instantaneous (for me at least).
> 
> 
No CUDA on lightweight Macbooks.

kingjr, 2016-01-15:
>  If you interpolate or reject a channel anyways, it doesn't matter if you do it pre or post regression

I guess it does if I go for a regularized solver? My intuition is that the filtering should in principle reduce some of the unexplainable variance, and thus improve the regression.

jona-sassenhagen, 2016-01-15:
Yes, if you stray from OLS, you lose the benefits of linearity :)

The regression fit is not so much our worry here - we want unbiased coefficients.

Filtering should make a difference it it's the difference between keeping or removing a longer segment of data due to a localized extreme spike.