mne-tools/mne-python #2793: Problem in scaling of ICA corrected data

hsonntag, 2016-01-15:
We observed a scaling of the MEG channels by a huge factor after applying ICA to raw data.
I prepared a small script (very simple only to show the problem) to demonstrate this effect for one of our measurement files (neuromag vectorview) where the scaling factor is about 10E10.
I would like to provide the dependency files on our bigmail server, would that be ok?
In the script, the first ICA component is excluded but the effect also shows up when nothing is excluded.
Maybe you know about this issue already when you see the code, otherwise I will upload the dependencies.
 
[test_ica.txt](https://github.com/mne-tools/mne-python/files/92182/test_ica.txt)

dengemann, 2016-01-15:
Hi @hsonntag this reminds me of an issue we saw 1.5 years ago, and after the subsequent fix I never saw it again (our ICA is heavily used in my network). Please share the data, I'm happy to take a look at this. I'm just seeing that you're using the noise covariance as a whitener, as this is an option that we haven't used for quite some time. Maybe a problem is lurking there.

hsonntag, 2016-01-18:
Thank you Denis,

I uploaded the file eg28a/eg28a1_ss.fif in:
https://bigmail.cbs.mpg.de/i/4bba73d90eea2183d452bcab2e29c95f.fif

and the file eg28a/eg28ab_ss.fif in:
https://bigmail.cbs.mpg.de/i/f4658234ba3e1b7ba2acbaf6046dbe2e.fif

The measurement file is rather big, I am sorry for that, I just wanted to provide the original data where the problem occurred.

dengemann, 2016-01-24:
Hi Hermann and Jing,

I started downloading the data and will have a look soon.

dengemann, 2016-01-25:
Hello Hermann,

I see a few bad channels that are not taken care of. Any ideas on this?
Also this is not a good thing as you apparently SSSed the data. SSS is
giving you crappy outputs if you don't declare or sanitize bad channels.
I'm now running things with one of my scripts to see what we get.

Denis

dengemann, 2016-01-25:
I used the following snippet to compute this [result-report](https://www.dropbox.com/s/dfbp75tht62b0gz/hermann-ica-report.html?dl=0) while respecting your filter choices.

```Python
import mne
from meeg_preprocessing import compute_ica
mne.set_log_level('INFO')

raw_fname = '4bba73d90eea2183d452bcab2e29c95f.fif'
empty_fname = 'f4658234ba3e1b7ba2acbaf6046dbe2e.fif'  # Empty room measurement
target_fname = 'eg28a1_ss_ica.fif'

raw_fname = '4bba73d90eea2183d452bcab2e29c95f.fif'
empty_fname = 'f4658234ba3e1b7ba2acbaf6046dbe2e.fif'  # Empty room measurement
target_fname = 'eg28a1_ss_ica.fif'

raw = mne.io.Raw(raw_fname, preload=True, verbose=True)

raw.filter(0.3, 120, l_trans_bandwidth=0.299, n_jobs=4)

reject_meg = dict(grad=4000e-12, mag=4e-12)
ica, reports = compute_ica(
    raw, 'Hermann-test', n_components='rank', reject=reject_meg, decim=4)

reports['html'].save('hermann-ica-report.html')
```

To me the ICA artefacts part looks good. But you see of course weird offsets because of the low high pass cutoff frequency and the low frequency garbage in your data. Also bad channels.

On my end after applying the ICA things look good. It's difficult to see what goes wrong on your end.

[image]

Let me know if you want to hang out on skype or gchat the next days.

hsonntag, 2016-01-26:
Hello Denis,

we will have a look at the bad channels in our preprocessing and at frequency cutoffs.
However, our problem is a scaling factor of about 10E10 after applying ICA with noise covariance as a whitener. I cannot imagine that this huge factor is a result of some preprocessing issues.
As you already mentioned, maybe the problem is related to the whitener.
We want to use the covariance matrix for whitening mainly because of the two different channel types in our MEG.
In your report you did not use a covariance matrix in your ICA computation.
If I do not use a covariance matrix I get a similar result as you.

Yes, let's talk on gchat about it soon.

dengemann, 2016-01-26:
Ok but the ICA rescales the channels blockwise to unit variance. As you see without cov we get nice results. No problem here because of different scales. Also note that SSS practically equalizes the covariance of the mags and grads as these are reprojected from the same low rank covariance. The naive whitening is often driven by one of them and you need to be careful with the rank of your cov when whitening with SSSd mags and grads. I think the ICA code does not yet take into account these complexities that are meanwhile considered for cov estimation on epochs and source localization. I will take a look at I think I roughly feel what's the reason for your observation. But my bet is that you will not win anything. But yoy have another potentially complex step in your pipeline that can cause trouble if you get it wrong.

hsonntag, 2016-01-28:
Ok, maybe we just avoid using a noise covariance matrix in the ICA.
But the scaling issue is rather independent from SSS, I tried the same thing with raw data and got more or less the same scaling issue. Apparently this is caused by providing a noise covariance matrix to the ICA routines because the scaling issue does not occur if i do not use the covariance matrix.

dengemann, 2016-01-28:
Hi Hermann, i checked the code and tried different things.
It is neither the filter, nor the regularization / scaling of the cov. In fact the ICA code takes care of that. I was wrong when saying it might be a problem with not using the latest whitening code. Also the thing does not deoend on the ica algorithm. I get the same when using fastica. It occurs already when not rejecting any  component.
I need to investigate deeper. 
Maybe it helps if can tell you at least that the defaults without cov will produce a reasonable whitening of mags and grads by rescaling and subsequent PCA. 
I saw hundreds of good solutions like that with data from many sitss.

Keeping you posted.

mmagnuski, 2016-05-07:
I have recently had this problem a few times with mne-python dev branch - working with epoched EEG data, no noise covariance. I am trying to reproduce the problem but it is not easy - currently I've only had this problem when marking components for rejection through the gui.
BTW - the scale seemed to be different each time I had this problem. In one of the notebooks I see scale of 1e33 for example.
I'll let you know once I have, at least a probabilistically reproducible example.

agramfort, 2017-07-22:
is this fixed?

hsonntag, 2017-07-26:
I did not test it recently. Did someone work on this issue in the meanwhile?

mmagnuski, 2017-07-26:
I had this issue before and couldn't replicate later when I was not using `plot_topo_image` and the rescaling of original data in `plot_topo_image` was fixed (#3827) so I suppose this one can be closed. @hsonntag - do you remember if you were plotting component time x trial images for example when you found the bug?

hsonntag, 2017-07-27:
@mmagnuski - The scaling issue occurred without any plotting, I found it by saving the data and visualizing from command line using mne_browse_raw.

However, I just ran some tests using 3791f9c945f5f411e87a19dbf05295bec891c79b and did not find any scaling problem anymore.

closed 2017-07-27 by hsonntag