mne-tools/mne-python #2794: equalize_event_counts in conflict with HED

teonlamont, 2016-01-17:
cc: @jona-sassenhagen
I'm getting an error when I am using this above method with HED. The error message is not really informative:
```python
ValueError: Don't mix hierarchical and regular event_ids like in ''.
```
the event ids used are all HED tags so I don't know what the conflict lies. I debugged it to this section:
https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L1596. After this section, I am left with empty lists `event_ids`.

jona-sassenhagen, 2016-01-17:
Can you post the event_ids and the tags you're using?

teonlamont, 2016-01-18:
```python
In [183]: event_id
Out[183]: 
{'fixation': 128,
 'nonword/prime': 9,
 'nonword/target': 10,
 'word/prime/primed': 5,
 'word/prime/unprimed': 1,
 'word/target/primed': 6,
 'word/target/unprimed': 2}
In [184]: epochs.equalize_event_counts(['word', 'nonword'])
```

jona-sassenhagen, 2016-01-18:
Hm ... I'll see. I'm already wishing I hadn't used nested comprehensions to do that. It seemed super smart when I wrote it, but now it just makes me feel stupid.

Have you tried, as a temporary work-around,
```
es_ = epochs["word"], epochs["nonword"]
mne.epochs.equalize_epoch_counts(es_)
epochs_equalized = mne.events.concatenate_epochs(es_)
```

Also, why are your epochs so uneven!?

jona-sassenhagen, 2016-01-18:
I'll do this, but I may be away from a reliable network connection in the next few days, I can't promise when I'll get it done.

teonlamont, 2016-01-19:
yeah, I was trying to unpack that list comprehension and couldn't clearly follow the logic + jetlag. Maybe we should unpack it into a few lines ;) 

> Also, why are your epochs so uneven!?

lol, because the number of nonwords and words in a single run of the experiment is contingent on behavior ;) also, there aren't equal presentations built in to the design by design.

> I'll do this, but I may be away from a reliable network connection in the next few days, I can't promise when I'll get it done.

no worries. your workaround worked (with minor fix). whenever you have a chance to look at this, that would be great. I can share the file with one, if you need them

jona-sassenhagen, 2016-01-19:
Can you indeed share the file? (If you can just downsample a lot, you can possibly upload it here.) I fail at replicating it here in the abstract.

teonlamont, 2016-01-20:
here's a link to the epochs [file](https://www.dropbox.com/s/822cccwi3s779nn/test-epo.fif?dl=0).

teonlamont, 2016-01-24:
thanks @jona-sassenhagen, that solved my problem! sorry I didn't have a chance to review your pr in time.

closed 2016-01-24 by teonlamont