mne-tools/mne-python #2799: local path in HTML?

kingjr, 2016-01-20:
@Eric89GXL 

Is [this](http://martinos.org/mne/dev/manual/c_reference.html#mne-redo-file) intended?

```
mne_redo_file

Usage: /home/larsoner/custombuilds/mne/current/bin/mne_redo_file file-to-redo
mne_redo_file_nocwd

Usage: /home/larsoner/custombuilds/mne/current/bin/mne_redo_file_nocwd file-to-redo
```

dengemann, 2016-01-20:
We had this before, I think there was some sphinx option to avoid this.

Eric89GXL, 2016-01-20:
Hah! No that's not intended. I don't remember fixing it previously, but it would be great if we could find it. Perhaps @agramfort remembers more

agramfort, 2016-01-20:
I have no idea... :( :(

Eric89GXL, 2016-02-04:
I had actually put in the full path at some point. Fixed here:

https://github.com/mne-tools/mne-python/commit/47cc50a80ae6c6920d9a216678bc137587e09f36

closed 2016-02-04 by Eric89GXL

kingjr, 2016-02-18:
It's still online though (dev + stable). But I'm not sure when / how the docs are compiled, so my comment may not be appropriate.

agramfort, 2016-02-18:
@jaeilepp any chance you can update the dev doc?