mne-tools/mne-python #2804: EHN: GAT 'mean-prediction' for circular data?

kingjr, 2016-01-20:
I have been decoding circular `y` (e.g. the angle of a Gabor patch) and [built a meta regressor to fit a linear regressor to the `sin(y)` and a linear regressor to `cos(y)`, and combine the corresponding predictions to output an angle (potentially accompanied by a radius)](https://github.com/kingjr/jr-tools/blob/master/jr/gat/classifiers.py#L197). 

Nothing crazy there.

However, in the `GeneralizationAcrossTime`, deviates from the sklearn API, by running the cross-val internally. This simplified some usecases, such as [generalization across conditions](http://martinos.org/mne/dev/auto_examples/decoding/plot_decoding_time_generalization_conditions.html#example-decoding-plot-decoding-time-generalization-conditions-py). 

In this scenario, we set `predict_mode='mean-prediction'`, which basically consists in 
1) making a prediction for any new trial `X_test[trial, :]` for each cross-val estimators(e.g. `predictions = [clf.predict(X_test[trial, :]) for clf in gat.estimators_[train_time][range(n_folds)]]`) and 
2) averaging across predictions.

But averaging across predictions isn't trivial: if it is a continuous output (e.g. a distance from the hyperplan), we can just do `np.mean(predictions, axis=0)`). However, if it is a categorical output, we take the mode `scipy.stats.mode(predictions)`.

Now, coming back on my circular data usecase, I would tyoically need to have a mean in the phase space, i.e. `prediction=artan2(mean(sin(predictions)), mean(cos(predictions))) * sqrt(cos(predictions)**2 + sin(predictions)**2)`, which is currently not implemented.

Considering that the decoding of Gabor patches is going to be a very common case (I already know 4 teams doing this), what shall we do? 
- Leave this specific usecase aside?
- Detect circular data and automatically applies the averaging in phase space
- Implement a `predict_mode` that takes any function.

I think I prefer the latter. LMKWYT

PS: I'll make some data publicly available soon. In the mean time, this kind of usecase with toy data [here](https://github.com/kingjr/jr-tools/blob/master/jr/gat/tests/test_clf.py).

Eric89GXL, 2016-01-20:
First option is sad, the second is a bit too magical, so I'm +1 on the third. I imagine there could be other applications for which you'd need a custom function, too.

dengemann, 2016-01-20:
I think providing a custom function would be the cleanest with the lowest
maintenance effort.

jona-sassenhagen, 2016-01-24:
I'll probably use a function like this soon.

jona-sassenhagen, 2016-01-24:
BTW @kingjr on your suggestion I started using RidgeCV and it works pretty well for my data, especially after a bit of dimensionality reduction.

kingjr, 2016-08-06:
I'm closing this, because this issue does not fit in the sklearn API. It is a computational layer above the CV, and thus above sklearn pipeline and its estimators/transformer objects.

closed 2016-08-06 by kingjr